evcc-io/evcc #14811: evcc not starting when OCPP is used

BorisBrock, 2024-07-10:
### Describe the bug

I installed evcc via the Home Assistant add on. The evcc version is 0.128.2. Then I added my devices to the configuration (Shelly Plug and a Fronius Smart Meter). So far everything is working as expected, and evcc runs fine.

The I added a OCPP wallbox like this:
```
chargers:
  - name: shelly_garage
    type: template
    template: shelly
    host: 192.168.178.79 # shellyplug-s-A75322
    channel: 0
    standbypower: 15
    integrateddevice: true
    icon: bike
  - name: heidelberg_ec
    type: template
    template: ocpp
    timeout: 2m # optional 
```

Now evcc does not start any more. Here's the outputs from the Home Assistant add on:
```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:08:38 evcc 0.128.2
[main  ] INFO 2024/07/10 08:08:38 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:08:38 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:08:38 connecting evcc-1393549785 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:08:38 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:11:39 evcc 0.128.2
[main  ] INFO 2024/07/10 08:11:39 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:11:39 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:11:39 connecting evcc-325288652 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:11:39 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:14:39 evcc 0.128.2
[main  ] INFO 2024/07/10 08:14:39 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:14:39 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:14:39 connecting evcc-1579390343 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:14:39 listening at :7070
...
```

I don't see any useful info in there, but it seems to keep restarting over and over. Once I remove the OCCP charger from the config, everything is fine again.

How can I fix this? Thanks!

### Steps to reproduce

1. Install evcc add on for Home Assistant
2. Add OCPP charger
3. Restart add on

### Configuration details

```yaml
Config:

network:
  schema: http
  host: 192.168.178.30
  port: 7070

interval: 10s # control cycle interval

log: info
levels:
  core: info
  lp-1: info
  lp-2: info

meters:
  - name: grid
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.178.84
  - name: pv
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.178.84

chargers:
  - name: shelly_garage
    type: template
    template: shelly
    host: 192.168.178.79 # shellyplug-s-A75322
    channel: 0
    standbypower: 15
    integrateddevice: true
    icon: bike
  - name: heidelberg_ec
    type: template
    template: ocpp
    timeout: 2m # optional 
    
vehicles:
  - name: kalkhoff
    title: Boris' E-Bike
    type: template
    template: offline
    capacity: 0.625 # Akku-Kapazit  in kWh

site:
  title: Home
  meters:
    grid: grid
    pv: pv

loadpoints:
  - title: Shellyplug S Garage
    charger: shelly_garage
    vehicle: kalkhoff
    mode: off
    guardduration: 5m

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.32 # EUR/kWh bei EnBW
  feedin:
    type: fixed
    price: 0.075 # EUR/kWh nach osterpaket 2022

mqtt:
  broker: 192.168.178.30:1883
  topic: evcc
```
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:08:38 evcc 0.128.2
[main  ] INFO 2024/07/10 08:08:38 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:08:38 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:08:38 connecting evcc-1393549785 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:08:38 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:11:39 evcc 0.128.2
[main  ] INFO 2024/07/10 08:11:39 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:11:39 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:11:39 connecting evcc-325288652 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:11:39 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/10 08:14:39 evcc 0.128.2
[main  ] INFO 2024/07/10 08:14:39 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 08:14:39 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/10 08:14:39 connecting evcc-1579390343 at tcp://192.168.178.30:1883
[main  ] INFO 2024/07/10 08:14:39 listening at :7070
...
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.2

andig, 2024-07-10:
timeout: 2m # optional 

What happens after 2 minutes?

BorisBrock, 2024-07-10:
Here's the output, when I set the log level to "debug":

> Using config file: /config/evcc.yaml
> starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
> [main  ] INFO 2024/07/10 09:13:32 evcc 0.128.2
> [main  ] INFO 2024/07/10 09:13:32 using config file: /config/evcc.yaml
> [db    ] INFO 2024/07/10 09:13:32 using sqlite database: /data/evcc.db
> [mqtt  ] INFO 2024/07/10 09:13:32 connecting evcc-771682612 at tcp://192.168.178.30:1883
> [mqtt  ] DEBUG 2024/07/10 09:13:32 tcp://192.168.178.30:1883 connected
> [main  ] INFO 2024/07/10 09:13:32 listening at :7070
> [ocpp-1] DEBUG 2024/07/10 09:13:32 waiting for chargepoint: 5m0s
> Using config file: /config/evcc.yaml
> starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
> [main  ] INFO 2024/07/10 09:16:32 evcc 0.128.2
> [main  ] INFO 2024/07/10 09:16:32 using config file: /config/evcc.yaml
> [db    ] INFO 2024/07/10 09:16:32 using sqlite database: /data/evcc.db
> [mqtt  ] INFO 2024/07/10 09:16:32 connecting evcc-1051987594 at tcp://192.168.178.30:1883
> [mqtt  ] DEBUG 2024/07/10 09:16:32 tcp://192.168.178.30:1883 connected
> [main  ] INFO 2024/07/10 09:16:32 listening at :7070
> [ocpp-1] DEBUG 2024/07/10 09:16:32 waiting for chargepoint: 5m0s

The evcc webserver never comes up, it seems to restart frequently.

premultiply, 2024-07-10:
Heidelberg EC does not support OCPP.

Please use the native Modbus RTU connection and the provided template.

closed 2024-07-10 by premultiply

BorisBrock, 2024-07-10:
> Heidelberg EC does not support OCPP.
> 
> Please use the native Modbus RTU connection and the provided template.

I know. This is just a name I used. Currently I'm only trying to initiate the OCPP server built into evcc.

premultiply, 2024-07-10:
> [ocpp-1] DEBUG 2024/07/10 09:16:32 waiting for chargepoint: 5m0s

Waiting for the specified OCPP charger to connect.

evcc does not start with missing devices.

BorisBrock, 2024-07-10:
You're right. If I start the charger and evcc at the same time it seems to work.

I think this behavior is a bit unintuitive. I would expect the server (evcc) to be started first. Once that's up and running, I can start the clients and let them connect to the server. But maybe my mental model of clients and servers is a bit off ;-)

What happens if the charger goes offline while evcc is running? E.g. the network connection is interrupted?
Will evcc go down in flames, or will the other chargers still be usable?

premultiply, 2024-07-10:
All chargers need to be reachable when starting up evcc.

Proper OCPP chargers keep reconnecting to the CS.

evcc waits up to 5 minutes for the OCPP connections.