evcc-io/evcc #14817: So, nach ein paar Tests habe ich neue Erkenntnisse.

steve0564, 2024-07-10:
So, nach ein paar Tests habe ich neue Erkenntnisse.
Die Geisterladungen werden von evcc gestartet.
Jedes Mal, wenn die Wallbox go-e kurz nicht mehr erreichbar ist, was mit der Meldung:

> charger out of sync: expected disabled, got enabled

im Log auftaucht, wird eine PV-Ladung gestartet, so als ob genügend Solarleistung zur Verfügung stehen würde und die Hausbatterieparameter entsprechend eingestellt wären.
Da dies manchmal nachts passiert, wird die Ladung nach dem Standarddelay von 3 Minuten wieder beendet. Ebenso, wenn die Hausbatterieparameter eigentlich eine Ladung nicht zulassen würden.
Ich versuch mich mal mit einem Issue.

_Originally posted by @steve0564 in https://github.com/evcc-io/evcc/discussions/14752#discussioncomment-10008722_

andig, 2024-07-10:
Bitte vollständiges Log der laufenden evcc Instanz.

StefanSchoof, 2024-07-10:
Die Meldung sagt das evcc eigentlich auch einen disabled charger erwartet. Also vermutlich liegt das Problem davor. Etwas startet den Ladevorgang und evcc zeigt sich mit der Log Meldung verwundert. (Also mehr die Feuerwehr Leute die immer auftauchen wenn es brennt, als der Brandstifter)

steve0564, 2024-07-10:
> Bitte vollständiges Log der laufenden evcc Instanz.

Welches Loglevel? Trace? Debug?

mfuchs1984, 2024-07-10:
Fallback Modus der Wallbox aktiv?

steve0564, 2024-07-10:
Fallback Modus?

steve0564, 2024-07-10:
So, hier das Debuglog, in dem der Fall zu beobachten sein sollte.
[evcc.log](https://github.com/user-attachments/files/16163221/evcc.log)

mfuchs1984, 2024-07-10:
> Fallback Modus?

https://go-e.com/de-de/magazin/go-e-charger-app-einstellungen
"Für den Fall, dass die Netzwerkverbindung einmal mehr als 2 Minuten unterbrochen sein sollte, kannst du den Fallback-Modus aktivieren, damit der go-e Charger mit einer gewissen maximalen Strommenge weiter lädt."

steve0564, 2024-07-10:
Jupp ist aktiviert, tut aber in diesem Fall, nix zur Sache, denk ich. Es hatte ja wochenlang vorher funktioniert und das mit Fallback!

premultiply, 2024-07-10:
Router und andere Netzwerkkomponenten updaten sich oft automatisch und starten dann neu, ferner fallen WiFi-Netzwerke durch unterschiedlichste Störungen und Firmware-Bugs zu beliebigen Zeitpunkten aus.
Daher gibt es immer wieder die Empfehlung für feste Infrastruktur KEINE WiFi-Verbindungen zu nutzen um wenigstens diese große Fehlerquelle zu vermeiden.
Gegen weitere Firmware-Bugs der Geräte hilft das natürlich auch nicht.

| charger out of sync: expected disabled, got enabled

Dies bedeutet dass evcc festgestellt hat, dass die Box **eigenmächtig** eine Ladung gestartet hat **ohne** dass dies von evcc erlaubt worden wäre.

Ist also eindeutig der Box zuzuschreiben.

Falls es zusätzlich vorab ein Kommunikationsproblem gegeben haben sollte, dann sollte man dies auch im evcc-Log sehen können.
Sind ansonsten keinerlei Fehlermeldungen erkennbar handelt es sich hier rein um ein internes Firmware-Problem der Box.

steve0564, 2024-07-10:
Vielen Dank für die Klarstellung!
D. h. die Box startet eigenmächtig, evcc unterbindet das sofort wieder, da die Parameter ja nicht stimmen....

closed 2024-07-10 by steve0564