evcc-io/evcc #14820: BMW Mini only detected as Guest Car and not charging automatically

Dervorino, 2024-07-10:
### Describe the bug

BMW Mini only detected as Guest Car and not charging automatically. Or is it normal that it takes 15 min or more until it is detected?

Configuration is as described in the documentation.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: my_car
    type: template
    template: mini
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    user: XXX@gmail.com # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: XXX # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: XXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 10 # Akkukapazität in kWh (optional)
```

### Log details

```text
[lp-1  ] DEBUG 2024/07/10 16:14:24 charge power: 0W
[lp-1  ] DEBUG 2024/07/10 16:14:24 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/10 16:14:24 pv power: 39W
[site  ] DEBUG 2024/07/10 16:14:24 grid meter: 758W
[site  ] DEBUG 2024/07/10 16:14:24 grid powers: [82 441 236]W
[site  ] DEBUG 2024/07/10 16:14:24 grid currents: [1.44 2.1 1.81]A
[site  ] DEBUG 2024/07/10 16:14:24 site power: 68W
[lp-1  ] DEBUG 2024/07/10 16:14:24 charge total import: 588.655kWh
[lp-1  ] DEBUG 2024/07/10 16:14:24 charger status: B
[lp-1  ] DEBUG 2024/07/10 16:14:24 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/10 16:14:24 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/07/10 16:14:24 pv charge current: 0A = 0A + -0.297A (68W @ 1p)
```

### What type of operating system are you running?

HomeAssistant Add-on

andig, 2024-07-10:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-07-10 by andig

Dervorino, 2024-07-10:
I think I might have found the issue. I have not configured the vehicle in the loadpoint. 

Now I have included it and I think it works.

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    vehicle: my_car
    # meter: charge # charge meter