evcc-io/evcc #14827: Fast mode (with peak shaving configured) doesn't start at auto-switching phases

Unwillingly473, 2024-07-10:
### Describe the bug

Peak shaving is working correctly when I manually select 1 phase in "fast" mode.
In case that it's configured at "auto-switching", the charger doesn't start.

I'm seeing no issues with auto-switching in case I'm using the "solar" and "minpv" mode.
It seems to be the combination with the fast mode and peak shaving.

I've attached the debug log which shows that the charger doesn't start when changing to "fast" mode with auto-switching.
Once I configure it for "1 phase", charging starts

### Steps to reproduce

1. Configure "auto-switching" as phases
2. Select "Fast" as mode
==> Charging doesn't start
3. Configure "1 phase" as phases
==> Charging does start

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 15s

sponsortoken: ***

# database configuration for persisting charge sessions and settings
database:
   type: sqlite
   dsn: /root/evcc/evcc.db

# Plant ID required for telemetry (ID generated in the GUI if chosen to enable telemetry
plant: ***

site:
  - title: Home # display name for UI
    circuit: Automaat-Hoofdaansluiting
    meters:
      grid: P1Meter_grid # grid meter reference
      pv: HuaweiDongle_pv
    residualPower: 10 # additional household usage margin

circuits:
  - name: Automaat-Hoofdaansluiting
    #maxCurrent: 40
    maxPower: 3500 #3.5 kW
    meter: P1Meter_grid
    parent:
  - name: Automaat-LaadpaalVoortuin
    #maxCurrent: 40
    maxPower: 22000 #22.0 kW
    meter: 
    parent: Automaat-Hoofdaansluiting

loadpoints:
  - title: Voortuin # display name for UI
    charger: Alfen # charger reference
    vehicle: vehicle # reference to standard vehicle
    mode: minpv # charge mode (off, now, minpv, pv). Default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available
    circuit: Automaat-LaadpaalVoortuin

meters:
  - name: P1Meter_grid
    type: template
    template: homewizard
[... 122 lines cut ...]
[circuit-Automaat-LaadpaalVoortuin] DEBUG 2024/07/11 00:08:56 power: 3025.6W
[circuit-Automaat-LaadpaalVoortuin] DEBUG 2024/07/11 00:08:56 current: 13.2A
[circuit-Automaat-Hoofdaansluiting] DEBUG 2024/07/11 00:08:56 power: 3368W
[circuit-Automaat-Hoofdaansluiting] DEBUG 2024/07/11 00:08:56 current: 0A
[site ] DEBUG 2024/07/11 00:08:57 pv power: 0W
[site ] DEBUG 2024/07/11 00:08:58 grid meter: 3330W
[site ] DEBUG 2024/07/11 00:08:58 site power: 3340W
[lp-1 ] DEBUG 2024/07/11 00:08:58 !! session: chargeRater.chargedEnergy=2.6 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/11 00:08:58 charge total import: 2168.841kWh
[lp-1 ] DEBUG 2024/07/11 00:08:58 charger status: C
[lp-1 ] DEBUG 2024/07/11 00:08:58 soc estimated: 28.04% (vehicle: 28.00%)
[lp-1 ] DEBUG 2024/07/11 00:08:58 vehicle soc: 28%
[lp-1 ] DEBUG 2024/07/11 00:08:58 vehicle range: 158km
[lp-1 ] DEBUG 2024/07/11 00:08:58 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/11 00:08:58 charger disable
[lp-1 ] DEBUG 2024/07/11 00:08:58 wake-up timer: stop
```

### What type of operating system are you running?

Docker container

### Version

0.128.2

premultiply, 2024-07-11:
Yes, fast mode is always 3p on switchable charger in automatic mode.

andig, 2024-07-11:
I don't see a good solution for this. At the point the limit is applied this is a 3p device. Since we don't have timers there we can't do this automatically. Since "Fast" mode is equivalent to 3p that's kind of expected. For time being works as implemented. Unfortunate but wontfix, we'll put it in the backlog.

closed 2024-07-11 by andig

andig, 2024-11-23:
Siehe https://github.com/evcc-io/evcc/discussions/15779#discussioncomment-11357550: 

> Nur dass das tatsächlich nicht nur beim Schnelladen sondern auch im PV Modus ein Problem ist. Wenn ich das richtig verstehe kann das nur bei einem Leistungslimit passieren. Wir brauchen also einen Mechanismus, der- egal ob PV oder Fast- ein Leistungslimit abfragt und- falls MinPower3p unterschritten wird auf 1p geht.

pergolafabio, 2025-01-13:
is this being worked on? i have the same kind of question, soon i will have a new meter (belgium)

i want to start at 1/3fase charging with a max load balancing for example 6kw , so as soon my other household load is below 1.8 , its ok, when my other household is higher then 1.8 (max 4.2  3 fase charging) , i want to switch back to 1 fase charging...