evcc-io/evcc #14830: Syntax error in entrypoint.sh

cbshopper, 2024-07-11:
### Describe the bug

 if [ "$1" == '"evcc"' ] || expr "$1" : '-*' > /dev/null; then
        exec evcc "$@"
    else
        exec "$@"
    fi

### Steps to reproduce

1. build docker image
2. try to start 
3. error messages like "/app/entrypoint.sh" will appear
...

### Configuration details

```yaml
solution: change "==" operator at line 26 to "="
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.128.0

andig, 2024-07-11:
I don't see any errors starting Docker?

cbshopper, 2024-07-11:
the problem seems to be that we don't use bash, we use sh. I tried to start the entrypoint.sh with sh i get this message:
entrypoint.sh.org: 26: [: unexpected operator
when i use the "=" operator it works!

StefanSchoof, 2024-07-11:
Running

```
> docker run -it --rm alpine:3.19 sh -c '[ "evcc" == "evcc" ]; echo $?'
0
```

works on my machine.

thiloms, 2024-07-11:
Both of you are right...

The evcc Docker Image is based on Alpine 3.19 and this uses a busybox implementation of sh shell and it seems that this sh shell has not an issue with == so it's working for the original evcc Docker container.

But if you use ubuntu or any other Docker image (or real OS) with an standard sh implementation (not busybox) you will get reported error:

```
> docker run -it --rm ubuntu:24.04 sh -c '[ "evcc" == "evcc" ]; echo $?'
sh: 1: [: evcc: unexpected operator
2
```
As the entrypoint also works with a single '=" (IMHO this is the right syntax for standard sh shell) it should be fixed...

andig, 2024-07-11:
Thank you. It does help to understand the conext (i.e. self-built docker image) ;)

StefanSchoof, 2024-07-11:
I am curious: Why do you use not the default alpine image for your docker setup?

cbshopper, 2024-07-11:
I used the Dockerfile without modifications - in Windows 11 Linux subshell with vscode.
FROM --platform=$BUILDPLATFORM node:18-alpine as node
FROM --platform=$BUILDPLATFORM golang:1.22-alpine as builder
# STEP 3 build a small image including module support
FROM alpine:3.19

closed 2024-07-11 by andig

andig, 2024-07-28:
I‘m thinking about reverting this, having looked at a few other images.

> IMHO this is the right syntax for standard sh shell

Is there any reference for this claim?

thiloms, 2024-08-22:
@andig 

> I‘m thinking about reverting this, having looked at a few other images.
> 
> > IMHO this is the right syntax for standard sh shell
> 
> Is there any reference for this claim?

Sorry for the late response.

It doesn't get right if others doing the same mistake ;-)

As an example look at https://stackoverflow.com/questions/20449543/shell-equality-operators-eq or https://stackoverflow.com/questions/49978928/conditionals-in-posix-standards.

'==' is NON-POSIX syntax (like bash) see https://thoughtbot.com/blog/the-unix-shells-humble-if#non-posix-concerns
'=' is POSIX syntax (standard sh shell is based on POSIX, see https://drewdevault.com/2018/02/05/Introduction-to-POSIX-shell.html and https://pubs.opengroup.org/onlinepubs/9699919799/utilities/V3_chap02.html)

As I wrote above: Alpine Docker images uses a busybox implementation of sh which also accepts '=='.

Anyway... A **stupid** alternative would be: "You can add bash to the alpine image, revert and use '==' as compare expression and change the shebang from #!/bin/sh  to #!/bin/bash in the entrypoint.sh file. That would blow up the image for nothing (at least 1 Byte for the addtional '=' and a few MB for adding bash) but the syntax would be correct then..." **lol**

Hope the references support you to keep the change with a single '='.