evcc-io/evcc #14839: Fahrzeug beginnt immer wieder zu laden und bricht dann wieder ab

winfotiker, 2024-07-11:
### Describe the bug

Das Fahrzeug (ID5) ist an der Elli Pro Wallbox angestöpselt. Die Sonne scheint nicht (hinreichend). Das Fahrzeug steht auf PV. immer wieder, innerhalb von wenigen Minuten, wird trotzdem ein Ladevorgang getartet und dann mangels Energie gleich wieder beendet.
installiert ist die Latest Verion. Problem tritt erst seit dem letzten Updaten von EVCC auf. 

### Steps to reproduce

1. Ladewilliges Auto ( ID5) im PV Modus anschließen
2.Sonne scheint nicht 
3. Auto beginnt trotzdem immer weider kurz zu laden und läuft dann auf einen Fehler-
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  cache: error

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3OTAyNzY0MDAsImlhdCI6MTY5NTY2ODQwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IndpbmZvdGlrZXIiLCJzcmMiOiJnaCJ9.CTotp1McZyyOGh5O0tOOE7G7QN6vLyyNbUH51WrPRkI

# unique installation id
plant: e4d9764606db15846e18aff901d5d5c4a8c5796afa9fad57ea63f3879687a1a8

interval: 10s # control cycle interval

meters:
- type: template
#  template: e3dc 
  usage: grid  
  host: 192.168.178.20  
#  port: 502  
  template: e3dc-rscp
  port: 5033
  name: grid1
  user: xxxxde
  password: ****
  key: ****
- type: template
#  template: e3dc
  usage: pv 
  host: 192.168.178.20  
#  port: 502  
  name: pv2
  template: e3dc-rscp
  port: 5033
  user: xxxxx
  password: *****
  key: *****
- type: template
#  template: e3dc 
  usage: battery  
  host: 192.168.178.20  
#  port: 502  
  name: battery3
  template: e3dc-rscp
  port: 5033
  user: xxxxx
  password: *****
  key: xxxxx

chargers:
- type: template
  template: ellipro 
[... 465 lines cut ...]
[eebus ] DEBUG 2024/07/11 19:47:13 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:Measurement to Measurement reply 2623 1003 MeasurementListData
[eebus ] DEBUG 2024/07/11 19:47:13 Recv: d:_i:47859_Elli-Wallbox-2109A184ND:Measurement to Measurement reply 98309 44243 MeasurementListData
[eebus ] DEBUG 2024/07/11 19:47:13 Recv: d:_i:47859_Elli-Wallbox-2109A184ND:Measurement to Measurement reply 98314 44244 MeasurementListData
[eebus ] DEBUG 2024/07/11 19:47:13 Recv: d:_i:47859_Elli-Wallbox-2109A184ND:Measurement to Measurement reply 98316 44245 MeasurementListData
[eebus ] DEBUG 2024/07/11 19:47:13 Recv: d:_i:47859_Elli-Wallbox-2109A184ND:Measurement to Measurement reply 98318 44246 MeasurementListData
[eebus ] DEBUG 2024/07/11 19:47:17 Send: d:_i:47859_Elli-Wallbox-2109A181H6 notify 1005 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/11 19:47:17 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:DeviceConfiguration to DeviceConfiguration notify 2627 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2024/07/11 19:47:17 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:DeviceConfiguration to DeviceConfiguration notify 2628 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2024/07/11 19:47:17 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2629 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2630 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2631 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:DeviceConfiguration to DeviceConfiguration notify 2632 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:DeviceConfiguration to DeviceConfiguration notify 2633 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2634 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2635 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/11 19:47:18 Recv: d:_i:47859_Elli-Wallbox-2109A181H6:ElectricalConnection to ElectricalConnection notify 2636 ElectricalConnectionPermittedValueSetListData
```

### What type of operating system are you running?

Linux

### Version

0.128.2

winfotiker, 2024-07-11:
Der ID ist aktuell über das UI konfiguriert mit gleichen Einstellungen. Keine Ahnung wie ich das exportieren könnte

Ramada2020, 2024-07-11:
Bei meinem Q4 etron und der elli Pro passiert seit 128.2 das Gleiche. Laden beginnt und wird gleich wieder beendet da in PV kein Strom da ist. Auch beim erreichten Zielladen das gleiche. Nur bei Aus erfolgt keine Ladung

andig, 2024-07-11:
Davon ist in deinem Logfile nichts zu sehen- nur "verbunden" ist zu erkennen:

```
[lp-2  ] DEBUG 2024/07/11 19:45:43 charger status: B
[lp-1  ] DEBUG 2024/07/11 19:45:53 charger status: B
[lp-2  ] DEBUG 2024/07/11 19:46:03 charger status: B
[lp-1  ] DEBUG 2024/07/11 19:46:13 charger status: B
[lp-2  ] DEBUG 2024/07/11 19:46:23 charger status: B
[lp-1  ] DEBUG 2024/07/11 19:46:33 charger status: B
[lp-2  ] DEBUG 2024/07/11 19:46:43 charger status: B
[lp-1  ] DEBUG 2024/07/11 19:46:53 charger status: B
[lp-2  ] DEBUG 2024/07/11 19:47:03 charger status: B
[lp-1  ] DEBUG 2024/07/11 19:47:13 charger status: B
```

Es bräuchte ein Log in dem wir das irgendwie erkennen können. Bitte `eebus: trace`.

DerAndereAndi, 2024-07-11:
Schau mal auf die ganzen

```
charge power:
```

log statements. Die Wallbox meldet keinen Verbrauch, bzw. nur 30W.

spikey-de, 2024-07-11:
Ich kann die Probleme bestätigen - E.ON / Vestel Box mit Tesla Y. An der Heidelberg mit Modus Gateway funktioniert alles.

DerAndereAndi, 2024-07-11:
In dem gezeigten Log ist gar keine Laderegelung von evcc zu sehen. Es wäre hilfreich ein TRACE level EEBUS log zu erhalten, welches vor dem Einstecken des Fahrzeugs beginnt.

winfotiker, 2024-07-12:
Sorry, mit dem Trace kann ich nicht dienen. Nach ein bisschen rumprobieren konnte ich aber anscheinend mein Problem lösen. Habe die Fahrzeug Config in der Oberfläche aktualisiert und gespeichert. Dann evcc (läuft im Docker auf meiner Synologie) neu gestartet und das Fahrzeug neu verbunden. Seit dem ist wieder Ruhe. Wenn sich das wieder ändert reiche ich den Trace nach.

winfotiker, 2024-07-12:
Ich lasse den Trace über Nacht mitlaufen und mache den ID jetzt nicht voll. Mal gucken ob das wieder auftritt

DerAndereAndi, 2024-07-12:
In der UI auf das Menü-Icon drücken, dann "Log", dort "Debug" auswählen und auf "Trace" ändern und auf den Download Button drücken. Sollte hoffentlich genügen.

Aber nur mal so generell: Wenn es jetzt tut, kann es kein Fehler in evcc sein. Die Elli hat sehr viele Software Bugs und schickt in einigen Szenarien einfach keine korrekten Daten. Und da sind sicher noch ein paar weitere Bugs drin, welche z.b. das gesehen Verhalten erzeugen können.

Ramada2020, 2024-07-12:
hi, hier 2 kurze logs in Trace für eebus. Hoffe ihr erkennt was. Q4 Etron wird kurz geladen, in evcc wird nichts angezeigt, geladen wird auch nichts. Ladekontollleuchte am etron ist grün. 
[evcc-20240712-172210-debug.log](https://github.com/user-attachments/files/16196186/evcc-20240712-172210-debug.log)
[evcc-20240712-165801-debug.log](https://github.com/user-attachments/files/16196187/evcc-20240712-165801-debug.log)

winfotiker, 2024-07-12:
> Aber nur mal so generell: Wenn es jetzt tut, kann es kein Fehler in evcc sein. Die Elli hat sehr viele Software Bugs und schickt in einigen Szenarien einfach keine korrekten Daten. Und da sind sicher noch ein paar weitere Bugs drin, welche z.b. das gesehen Verhalten erzeugen können.

Vor dem Release ist der Fehler aber überhaupt nicht aufgetreten... und nen Tesla hab ich im Threat auch gesehen :) 
hier mal der Log. Ich stelle einen neuen ein wenn das Verhalten wieder auftritt
[evcc-20240712-181112-trace.log](https://github.com/user-attachments/files/16196741/evcc-20240712-181112-trace.log)

Ramada2020, 2024-07-13:
Guten Morgen, hier mal ein log von der Version 0.127.3 (wo das Problem nicht auftritt) und der aktuellen Version : Hier tritt am Ende vom log das Problem auf. 

[evcc-20240713-101136-trace_0.128.2.log](https://github.com/user-attachments/files/16201368/evcc-20240713-101136-trace_0.128.2.log)
[evcc-20240713-092326-trace_0.127.3.log](https://github.com/user-attachments/files/16201369/evcc-20240713-092326-trace_0.127.3.log)

DerAndereAndi, 2024-07-13:
@Ramada2020 zu deinen Logs. Da deine Logs nur EEBUS beinhalten, kann man keine Aussagen zu evcc machen. Also warum ob die Kommandos geschickt werden.

12.07.:
17:31:35 Es wird ein Ladelimit von 0A geschickt -> Auto soll nicht laden
17:37:33 Es wird ein Ladelimit von 0A geschickt -> Auto soll nicht laden

In dem neueren Log vom 13.7.:
Version 0.128.2: Ich sehe keine einzige Laderegelung in dem Log. Weder eine Freigabe noch eine Ladesperre. Ich kann hier also keine Aussage treffen
Version 0.127.3:  09.19:51: Es wird ein Ladelimit von 0A geschickt -> Auto soll nicht laden

Ich kann hier nichts ungewöhnlich oder falsches von EEBUS Seite erkennen. In allen Fällen soll das Auto nicht laden und genau wird dann geschickt. An keiner Stelle sehe ich eine Ladefreigabe welches dem Auto erlauben sollte zu laden.

@winfotiker auch in deinem Log ist weder ein Ladestop noch eine Ladefreigabe zu sehen. Es werden auch nie Messwerte geschickt welche zeigen würden dass das EV lädt.

An alle: Was genau soll denn nun der Fehler sein? Ich kann hier keinen erkennen.

andig, 2024-07-13:
Wie immer: bitte vollständige Logs mit Kommentaren, zu welchen Zeitpunkten welches Fehlverhalten gesehen wird. 

Danke @DerAndereAndi dass Du Dich an dem Suchspiel beteiligst...

winfotiker, 2024-07-13:
Hey, hier das Log
 Aktuell tritt der Fehler auf. Also um 18:40 Rum.
[evcc-20240713-184328-trace.log.txt](https://github.com/user-attachments/files/16202978/evcc-20240713-184328-trace.log.txt)
[image]

winfotiker, 2024-07-13:
[evcc-20240713-192158-trace.log.txt](https://github.com/user-attachments/files/16203036/evcc-20240713-192158-trace.log.txt)
So?

DerAndereAndi, 2024-07-13:
Ja, in dem Log wird das Laden gestoppt, genau dann wenn evcc es will:

```
[lp-2  ] DEBUG 2024/07/13 18:56:42 pv disable in 0s
[lp-2  ] DEBUG 2024/07/13 18:56:42 pv disable timer elapsed
[eebus ] DEBUG 2024/07/13 18:56:42 Send: d:_i:47859_Elli-Wallbox-2109A181H6 write 3686 LoadControlLimitListData
[lp-2  ] DEBUG 2024/07/13 18:56:42 charger disable
[lp-2  ] DEBUG 2024/07/13 18:56:42 wake-up timer: stop
[eebus ] TRACE 2024/07/13 18:56:42 Send: c04cc6e65dec6a3c308f178017bef67693856a78 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC-3064386362653464"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2109A181H6"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":3686},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
```

Ich sehe in diesem Log kein Fehlverhalten.

spikey-de, 2024-07-13:
Ich kann es euch leider wegen Urlaub nicht nachstellen. Aber es war gestern so dass ich keine Chance hatte, die Vestel mit dem Tesla zum Laden zu bekommen. 

Einstecken, Modus schnell, er fährt hoch und bricht nach 5 Sekunden ab. An der Box dann hart lastmangement ausgeschaltet, über die App gestartet, lädt mit 11kw.

andig, 2024-07-13:
Das nutzt alles nix ohne Logfile. Bitte keine Textaufgaben mit denen wir nichts anfangen können. Danke!

winfotiker, 2024-07-13:
> Ja, in dem Log wird das Laden gestoppt, genau dann wenn evcc es will:
>
> ```
> [lp-2  ] DEBUG 2024/07/13 18:56:42 pv disable in 0s
> [lp-2  ] DEBUG 2024/07/13 18:56:42 pv disable timer elapsed
> [eebus ] [...]

Was soll ich dazu sagen. Tritt erst seit dem letzten Release auf. Und die Karre erschleicht sich durchaus Strom. Ist jetzt schon auf 78% 
Mal gucken wie ich das im Docker zurück gedreht bekomme.
Hier nochmal das aktuelle Log.
[evcc-20240713-201614-trace.log.txt](https://github.com/user-attachments/files/16203192/evcc-20240713-201614-trace.log.txt)

DerAndereAndi, 2024-07-13:
@winfotiker in dem letzten Log meldet die Box keinen Stromverbrauch. Das bedeutet evcc hat keine Chance zu erkennen das ein Auto lädt. Das ist auch wieder etwas komplett anders als Thema hier im Ticket. Ich kann nicht beeinflussen was die Box macht, welche Daten sie schickt und welche nicht. Die macht das z.b. wenn das Auto eingesteckt ist aber die Netzwerk-Verbindung unterbrochen wurde. Entweder bei WLAN oder mit einem evcc update, etc. pp

Von EEBUS Seite sehe ich hier nichts, wirklich nichts das zu korrigieren wäre.

Die Box ist Schrott. Sorry. Und wenn "das" mit der vorherigen Version besser sein soll, ist das nicht nachvollziehbar auf die Code zurückzuführen. Sondern dass z.b. dass die genannten (oder ähnlichen) Umstände IN DER BOX dann nicht auftreten. Hier ist kein kausaler Zusammenhang zu erkennen. Es erscheint eher zufällig so zu sein.

StefanSchoof, 2024-07-13:
Ich habe das Problem bei mir auch mehrfach mit dem neuen EEEBUS Stack und der Elli Box beobachtet. Ich nutzte alten Stack seit es ihn gibt. Ich glaube statistisch gesehen, nes fast ausgeschlossen ist das der Fehler unabhängig von dem neuen Stack ist. Ich befürchte, dass eine Änderung im neuen Stack einen unbekannten Bug in der Box auslöst.

andig, 2024-07-13:
Das kann natürlich sein. Den neuen Stack werden wir nicht zurück drehen da uns das von der Weiterentwicklung abkoppelt. Bleibt die Frage, ob es noch lohnt, weitere Logs zu machen und weiter zu suchen?

DerAndereAndi, 2024-07-13:
@StefanSchoof was ist "das Problem" ? Hier wurden im Verlauf verschiedenste "Probleme" genannt. Nur eines war nachvollziehbar: Das Auto lädt und die Wallbox meldet keine Messwerte.

Alles andere war bisher nachvollziehbar: evcc sagt das Laden soll beendet werden, das Laden wird beendet, evcc sagt das Laden soll fortgesetzt werden, das Laden wird fortgesetzt.

Es gibt bisher kein Log das zeigt, ob bei diesem laden stoppen/starten Szenario wirklich ein Problem gibt. Und weiter ob das wirklich im EEBUS Stack ist. Der Unterschied von 0.127.3 und 0.128.2 sind 69 Commits. Ein paar EEBUS, aber nicht alle.

Also nochmal: Bitte stellt ein komplettes Log mit alles Daten im TRACE Log zur Verfügung, welches VOR dem Anstecken des Fahrzeugs beginnt. Die UI hat unter Umständen nicht die ganze benötigte Historie, in der Dokumentation ist beschrieben wir man an das Log kommt. Bis dahin sehe ich hier von EEBUS Seite wirklich wirklich nichts zu tun.

StefanSchoof, 2024-07-13:
Das Problem das ich hatte zeigt der Screenshot von https://github.com/evcc-io/evcc/issues/14839#issuecomment-2226993398 gut. Es soll nichts laden, aber die Box fängt immer kurz an und hört sofort wieder auf. Wenn man beim Auto steht hört man auch das Klacken des Schütz. Ich habe da keine Meldungen in evcc gesehen. Es kam mir so vor als würde die Box ein Eigenleben haben. Ein Starten über Min+PV war möglich. Das ganze trat nicht immer auf.

Ich kann versuchen eine Log zu ziehen, aber da das Auto gerade voll ist und in den nächsten Tagen auch nicht benötigt wird, wird das bei mir vermutlich erst Ende der nächsten Woche was. (Daher hatte ich mich bisher auch noch nicht gemeldet, da ich hoffe das jemand vor mir in der Lage ist ein sinnvolles Log zu erstellen)

andig, 2024-07-13:
Bitte Keine Diskussion über Screenshots. Das ist Zeitverschwendung.

winfotiker, 2024-07-13:
> Ich habe das Problem bei mir auch mehrfach mit dem neuen EEEBUS Stack und der Elli Box beobachtet. Ich nutzte alten Stack seit es ihn gibt und habe den Fehler dort nicht einmal gehabt. Ich glaube statistisch gesehen, nes fast ausgeschlossen ist das der Fehler unabhängig von dem neuen Stack ist. Ich befürchte, dass eine Änderung im neuen Stack einen unbekannten Bug in der Box auslöst.

Stimme dem zu. Wenn es zu dem Verhalten kommt, meldet die Box das sie nicht mit dem Energiemanagement System verbunden sei. Ein Neustart vom EVCC behebt das Problem dann. (hatte gerade ein wenig Zeit zum rumspielen)

Sorry wenn euch das nervt. Aber das ist nun mal die Box welche ich habe. 
Log liefere ich nach wenn das Problem das nächste Mal auftritt.

Ramada2020, 2024-07-14:
Guten Morgen,
ich habe heute ein Komplettlog von diesem Geisterladen erstellt. Log anbei. Kurze Erklärung:
-	Musste neu starten weil der Audi beim ersten Start nicht erkennt wurde
-	Beim 2. Startversuch wurde das Auto schnell erkannt. Zwischen 8:54 und 8:56 kam es zum Geisterladen – Version 1. Auto startet ganz kurz das Laden. Wallbox zeigt Laden an. Ladeleuchte am Auto nur kurz grün dann wieder vorbei (Screenshot von Evcc 8:55)
-	Um 8:59 laut EVCC Screenshot kommt es zur zweiten Version des „Geisterladens“ obwohl kein Überschuss vorhanden ist startet EVCC das Laden für 3 min.
Vielleicht ist es auch gar kein EEBUS Fehler sondern eine Fehlinterpretation des Verfügbaren Ladestroms – aber da bin ich Laie
Ich wäre Euch dankbar wenn ihr euch das Log mal anschaut – eine andere Wallbox als die ElliPro zu kaufen ist schwierig…
Log:
[evcc.log](https://github.com/user-attachments/files/16215903/evcc.log)

1. Geisterladen
[image]
2.Version
[image]
[image]

DerAndereAndi, 2024-07-14:
Danke für das Log, anbei meine Analyse:

8:53:17 evcc wird beendet
8:53:27 evcc wird neu gestartet
8:53:29 die Wallbox baut eine Verbindung zu evcc auf und evcc bricht sie ab: da ist die evcc Konfiguration noch nicht komplett eingelesen und die Wallbox noch nicht registriert. Das lässt sich vielleicht verbessern.
8:53:29 die Wallbox baut sofort eine neue Verbindung zu evcc auf und evcc bricht sie wieder ab
8:53:30 evcc hat die Konfiguration komplett eingelesen/verarbeitet und baut selbst eine Verbindung zur Wallbox auf
8:53:31 die Wallbox meldet dass kein EV angesteckt ist
8:53:53 die Wallbox meldet dass ein EV angesteckt ist
8:53:55 evcc erkennt ein angestecktes EV an lp-1, Zeitversatz aufgrund des intervalls in dem geregelt wird
8:53:56 die Wallbox meldet dass das EV nicht lädt
8:53:56 die Wallbox meldet dass das EV lädt
8:54:00 evcc stoppt den Ladevorgang

Soweit ist der Ablauf völlig normal. In EEBUS wird das Auto IMMER mit einer Ladefreigabe starten und evcc wird dies dann stoppen, außer in "MIN+PV" bzw. "Schnell". Das war auch zuvor so. Zu beachten ist, dass der Ladestopp innerhalb des Intervals erfolgt, und bei mehreren (n) angeschlossenen Wallboxen die Regelung einer Wallbox in n*Intervall erfolgt! Das bedeutet dass das Auto auch schon mal für eine etwas längere Zeit lädt, bis die Wallbox zur Regelung wieder an der Reihe ist und einen Ladestop erhält.

8:54:27 Die Wallbox meldet eine Ladeleistung von 12W, diese wird NICHT als aktiver Ladevorgang interpretiert -> Status B, die Leistung muss NICHT vom Fahrzeug stammen, kann auch der Eigenverbrauch sein. Das ist absolut normal und war auch schon zuvor genau so. Das ist KEINE "Geisterladung"
8:54:27 Der PV Timer wird inaktiv 
8:54:52 Die Wallbox meldet Aktualisierungen der Verbindung -> Das ist ungewöhnlich und sollte nicht sein, denn es hat sich nichts geändert. Auch die Aktualisierungen sind identisch zu den erwarteten Werten
8:54:53 Die Wallbox meldet 87W, weiter Status B

Es könnte sein, dass die Wallbox hier intern eine Ladefreigabe erteilt hat, obwohl keine da ist. Und dann das Laden sofort wieder stoppte. Von evcc wurde KEIN Ladestop geschickt!

8:55:13 Die Wallbox meldet 12W, Status B
8:55:33 Die Wallbox meldet 12W, Status B
8:55:53 Die Wallbox meldet 12W, Status B
8:56:13 Die Wallbox meldet 12W, Status B
8:56:33 Die Wallbox meldet 12W, Status B
8:56:53 Die Wallbox meldet 12W, Status B
8:57:13 Die Wallbox meldet 12W, Status B
.....

...
8:59:33 Die Wallbox meldet 12W, Status B
8:59:45 Die Wallbox meldet Aktualisierungen der Verbindung -> Das ist ungewöhnlich und sollte nicht sein, denn es hat sich nichts geändert. Auch die Aktualisierungen sind identisch zu den erwarteten Werten
8:59:53 Die Wallbox meldet schon wieder Aktualisierungen der Verbindung -> Das ist ungewöhnlich und sollte nicht sein
8:59:53 Die Wallbox meldet 2737W, Status C -> Hier ist ein Problem. Die Wallbox hat immer noch eine Ladesperre und informierte auch über keine Änderung dass diese evtl. selbst aufgehoben wurde
8:59.54 Der PV Disable Timer startet, evcc schickt ein Ladelimit von 6A

Beobachtung:
1. Warum auch immer meldet die Wallbox unerwartete Daten-Aktualisierungen, ohne dass eine Änderung existiert
2. Zu diesen Zeitpunkten gibt es manchmal (!!) höhere Messwerte, teilweise hören die einfach auf, teilweise nicht
3. evcc startet bei unerwartetem Status C einen Disable-Timer und lässt die Wallbox dann in diesem Fall 3 Minuten weiter laden
4. Zu keiner Zeit sieht es für evcc so aus, als wäre die Verbindung zur Wallbox abgebrochen! Die Timeouts werden geschickt, Messwerte empfangen.

Konsequenz:
Es ist unbekannt was die Box jetzt hier dazu führt intern teilweise das Laden freizugeben. Was in diesem Code bisher neu ist, dass aktiv nach neuen Messwerte gefragt wird und damit mehr "leere" Kommunikation stattfindet. Den entsprechenden Code habe ich vor ein paar Tagen im Stack entfernt, da er keinen Benefit gezeigt hat. Inhaltlich ist der Code nicht falsch und darf keine Auswirkungen auf das Verhalten der Wallbox haben. Es sind nur Leseabfragen. Ich werde diese Änderung nun per PR mal in evcc anbieten. Evtl. führt das zum entsprechenden Verhalten in der Box.

Das ganze Verhalten ist so oder so nur ein weiteres Beispiel für die sehr sehr schlechte Software-Qualität der Wallbox. Ich kann und werde nicht für alle Ewigkeit darauf Rücksicht nehmen können! Das kostet mich seit Jahren so viel Zeit, diesen Problemen hinterher zu programmieren und ich persönlich habe auch keine Gegenleistung. Ich verwende die Box nicht und plane es auch nicht zu tun. Die Box kann ungesteuert gut seinen Dienst verrichten, aber für gesteuerten Betrieb ist sie eine Zumutung. Selbst ohne dieses "neue" Verhalten gibt es genug Problemfelder die von Außen nicht gelöst werden können.

Nur mal zur Einordnung: Nur dieses Ticket alleine kostet mich bereits 3h Arbeit.

andig, 2024-07-14:
Neues Nightly kann in 15min getestet werden.

> Nur mal zur Einordnung: Nur dieses Ticket alleine kostet mich bereits 3h Arbeit.

Als evcc wünschen wir uns natürlich bestmöglichen Support für eine breite Anzahl von Geräten. Man muss aber leider auch sagen, dass es Geräte und Hersteller gibt, die das Leben unglaublich schwer machen. Ihren Anspruch, eine (implizit) nutzbare EEbus Implementierung zu liefern erfüllt die Box leider nicht. Am Ende wäre da der Hersteller in der Pflicht- hier schlägt dann völlig inexistente Support zu.

Ramada2020, 2024-07-14:
Hi, habe das Nightly 30 min getestet - Geisterladen tritt nicht mehr auf, Vielen Dank. Beobachte morgen weiter (heute ist die Sonne weg).
Anbei der log - es kamen ein paar mehr Fehler wegen charger out of sync  beim anstecken: - aber damit kann man leben.

[Evcc_nightly.log](https://github.com/user-attachments/files/16227784/Evcc_nightly.log)

andig, 2024-07-14:
Wie Andi schon schrieb- das kurze anladen ist per Design und nicht zu verhindern. Sonst klingt das doch jetzt ganz gut 👍🏻.

rupenik, 2024-07-18:
I'm still seeing this issue with nightly build (commit dca3612, that should include "Improve EEBUS support for Elli Gen 1" fix). When EV is connected, but not charging (charger status B), evcc log shows that charge power is frequently fluctuating from normal charger self consumption (< 20W) to around 340W or even more.

During those higher consumptions, my car's app also briefly shows ongoing charging and then changes back to not charging state. When charge power in the log is over 340W (more close to 1000W), evcc switches to charger status C and logs charger out-of-sync message.

```
evcc  | [lp-1  ] WARN 2024/07/18 06:20:31 charger out of sync: expected disabled, got enabled <-- expected, because I restarted evcc while EV was connected
evcc  | [lp-1  ] WARN 2024/07/18 06:47:47 charger out of sync: expected disabled, got enabled <-- expected, because I connected cable to EV
evcc  | [lp-1  ] WARN 2024/07/18 07:47:31 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 12:41:46 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 14:36:16 charger out of sync: expected disabled, got enabled <-- expected, because I connected cable to EV
evcc  | [lp-1  ] WARN 2024/07/18 15:17:31 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 15:27:46 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 16:47:01 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 17:00:31 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
evcc  | [lp-1  ] WARN 2024/07/18 17:19:01 charger out of sync: expected disabled, got enabled <-- not expected, EV was already connected
```

Example log messages of charge power fluctuating even though charging shouldn't be happening:
```
evcc  | [lp-1  ] DEBUG 2024/07/18 16:56:30 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:56:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:57:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:57:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:57:30 charge power: 14W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:57:45 charge power: 343W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:58:00 charge power: 39W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:58:15 charge power: 39W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:58:30 charge power: 39W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:58:45 charge power: 39W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:59:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:59:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:59:30 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 16:59:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:00:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:00:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:00:30 charge power: 1233W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:00:45 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:01:00 charge power: 340W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:01:15 charge power: 336W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:01:30 charge power: 1086W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:01:45 charge power: 340W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:02:00 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:02:15 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:02:30 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:02:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:03:00 charge power: 23W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:03:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:03:30 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:03:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:04:00 charge power: 338W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:04:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:04:30 charge power: 120W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:04:45 charge power: 340W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:05:00 charge power: 14W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:05:15 charge power: 14W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:05:30 charge power: 14W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:05:45 charge power: 14W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:06:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:06:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:06:30 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:06:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:07:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:07:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:07:30 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:07:45 charge power: 340W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:08:00 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:08:15 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:08:30 charge power: 108W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:08:45 charge power: 340W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:09:00 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:09:15 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:09:30 charge power: 9W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:09:45 charge power: 12W
evcc  | [lp-1  ] DEBUG 2024/07/18 17:10:00 charge power: 12W
```

When I downgraded to 0.127.3, charge power log messages stay at low figures (< 20W) and no ghost charging or charger out-of-sync is happening.

Log file:
[evcc_0.128.2_nightly_log_18072024.txt](https://github.com/user-attachments/files/16286583/evcc_0.128.2_nightly_log_18072024.txt)

DerAndereAndi, 2024-07-18:
Slowly I am out of ideas. The last thing I can do is send the heartbeats quicker than required, maybe they check if the heartbeat arrives within 4s, which would be a wrong interpretation of the specification, and thus put the wallbox in fallback mode and allow minimal 4,2kW charging. All values in measurements are always <6A as far as I can see.

The old stack did send the heartbeat every second, which was way over the top, just caused way more unnecessary traffic but according to the spec does not have to be that way.

I'll try.

rupenik, 2024-07-19:
I have been testing new nightly build (commit [9e75d8c7](https://github.com/evcc-io/evcc/commit/9e75d8c7)) for over 3 hours and seems that it fixed the issue! Charge power is back to being very stable and low (9W / 18W) and so far no unexpected charging has happened (two expected charger out of syncs, because of evcc nightly build restart and EV cable reconnect). 

Here's a new log file just in case you would like to see it: 
[evcc_0.128.2_nightly_log_19072024.txt](https://github.com/user-attachments/files/16302482/evcc_0.128.2_nightly_log_19072024.txt)

Thank you so much for your hard work regarding EEBUS support for Elli Gen 1 chargers @DerAndereAndi!

winfotiker, 2024-08-02:
Hab jetzt tatsächlich Kontakt zum Support von Elli. Kann mir jemand eine treffende Fehlerbeschreibung für das unklare Verhalten der Wallbox geben? Wie konkret verhält die sich nicht entsprechend ihrer Spezifikationen?
Das Thema mit dem fehlenden Zertifikat habe ich bereits adressiert.
Danke!

andig, 2024-08-02:
> Das Thema mit dem fehlenden Zertifikat habe ich bereits adressiert.

Das kenne Elli seit 3 Jahren ;)

winfotiker, 2024-08-02:
Werde die trotzdem damit nerven 😌

Auch mit dem neuen tollen Feature. Ich hab zwei von den Boxen. Ganz neu ist: wenn ich ein Fahrzeug abstöpsle, dann fängt das andere Fahrzeuge an zu laden. Evcc zeigt mir dann zwar an dass da eine Ladung mit minimum Strom stattfindet (4,1kw in meinem Fall) aber zeichnet das leider nicht als Ladevorgang auf. Dafür scheint das alte sporadische Geisterladen definitiv erledigt zu sein.
Bin aber trotzdem noch nicht bereit zwei neue Boxen zu kaufen

winfotiker, 2024-08-04:
Bin jetzt wieder auf die 0.127.3 zurück. Immer wenn ich ein Fahrzeug von einer Box trenne, fängt das andere Fahrzeug an der zweiten Box an zu laden. Ohne Rücksicht auf die aktuelle Situation und zieht mir dann den Hausakkku leer. Lässt ich dann auch nicht mehr über EVCC steuern. 
Ich versuch jetzt den Fehler bei Elli zu adressieren. Was mir wirklich helfen würde ist eine kurze Beschreibung was sich für die Elli Boxen in der EEBUS-Anbindung nach dem Release 0.127.3 geändert hat.

andig, 2024-08-04:
@winfotiker das wird nichts nutzen. Helfen kann ein Logfile mit aktuellem Stand- soweit ich das sehe gibts das nicht?

winfotiker, 2024-08-04:
ok. Rolle Rückwärts und zweimal um die Autos getanzt 
[evcc-20240804-194514-trace.log](https://github.com/user-attachments/files/16488456/evcc-20240804-194514-trace.log)

DerAndereAndi, 2024-08-05:
Hier mal eine erste Analyse des Logs:

```

19:40:33: LP-1 und LP-2 mit "charge power: 0W", kein EV angeschlossen
19:40:33: LP-1 und LP-2 mit "charger status: A"
19:41:42: LP-1: EV über EEBUS als angeschlossen gemeldet

Intervall LP-1 ist an der Reihe:
19:41:44: LP-1: "charge power: not available", es wurden bisher keine Messwerte von der Wallbox gemeldet
19:41:44: LP-1: "car connected", evcc weiss nun dass ein EV angeschlossen ist
19:41:44: LP-1: "charger status: B"
19:41:47: LP-1: evcc schickt 0A Ladelimit, wird bestätigt

Intervall LP-2 ist an der Reihe:
19:41:54: LP-1: "charge power: not available", die Wallbox meldet weiterhin keine Messwerte
19:41:54: LP-2: "charge power: 4140W"
19:41:54: LP-2: "charger status: B"

Intervall LP-1 ist an der Reihe:
19:42:03: LP-1: "charge power: not available", die Wallbox meldet weiterhin keine Messwerte
19:42:03: LP-2: "charge power: 0W"
19:42:03: LP-1: "charger status: B"
19:42:07: LP-2: EV über EEBUS als angeschlossen gemeldet

Intervall LP-2 ist an der Reihe:
19:42:13: LP-2: Messwerte werden gemeldet
19:42:14: LP-1: "charge power: not available", die Wallbox meldet weiterhin keine Messwerte
19:42:14: LP-2: "charge power: 4140W"
19:42:14: LP-2: "charger status: C"
19:42:14: LP-2:"car connected",  evcc weiss nun dass ein EV angeschlossen ist
19:42:18: LP-2: ID.5 als angesteckt erkannt
19:42:18: LP-2: evcc schickt 0A Ladelimit, wird bestätigt

Intervall LP-1 ist an der Reihe:
19:42:24: LP-1: "charge power: not available"
19:42:24: LP-2: "charge power: 0W"
19:42:24: LP-1: "charger status: B"

Intervall LP-2 ist an der Reihe:
19:42:34: LP-1: "charge power: not available"
19:42:34: LP-2: "charge power: 0W"
19:42:34: LP-2: "charger status: B"
19:42:39: LP-2: EV über EEBUS als nicht mehr angeschlossen gemeldet

Intervall LP-1 ist an der Reihe:
19:42:44: LP-1: "charge power: not available"
19:42:44: LP-2: "charge power: 0W"

Intervall LP-2 ist an der Reihe:
19:42:53: LP-2: EV über EEBUS als angeschlossen gemeldet
19:42:53: LP-1: "charge power: 1382W", zum ersten Mal meldet LP-1 Messwerte
19:42:54: LP-2: "charge power: not available"
19:42:54: LP-2: "charger status: B"

Intervall LP-1 ist an der Reihe:
19:43:03: LP-1: "charge power: 1405W"
19:43:03: LP-2: "charge power: not available"
19:43:03: LP-1: "charger status: C"
19:43:06: LP-1: Passat wird als angesteckt erkannt
19:43:06: LP-1: "set charge mode: pv"
19:43:06: LP-1: evcc schickt 6A Ladelimit, wird bestätigt

Intervall LP-2 ist an der Reihe:
19:43:13: LP-1: "charge power: 1405W"
19:43:13: LP-2: "charge power: 4140W", zum ersten Mal meldet LP-2 Messwerte (seit dem erneuten Anstecken)
19:43:13: LP-2: "charge status: C"

Intervall LP-1 ist an der Reihe:
19:43:23: LP-1: "charge power: 1405W"
19:43:23: LP-2: "charge power: 4140W"
19:43:23: LP-1: "charger status: C"

Intervall LP-2 ist an der Reihe:
19:43:33: LP-1: "charge power: 1405W"
19:43:33: LP-2: "charge power: 4140W"
19:43:33: LP-2: evcc schickt 6A Ladelimit, wird bestätigt
19:43:33: LP-2: "charge status: C"

.....
```

@winfotiker Es wäre sehr hilfreich zu erfahren, ob der Passat an LP-1 am Anfang schon geladen hat, oder wirklich erst ab 19:42:53 ?
Und was ist ist deiner Meinung nach an dem Verhalten wann genau falsch? "Rolle Rückwärts und zweimal um die Autos getanzt" ist leider keine nachvollziehbare Beschreibung eines Fehlverhaltens.

Das einzige was sein könnte (auch in anderen Bug Reports zu sehen ist): evcc hat eine Änderung im Code wann/wie die Ladelimits geschickt werden, und das hier auch einen Effekt hat. Auf EEBUS Seite ist das Verhalten identisch zu alten Versionen.

Die Analyse hat jetzt 1h gedauert, und das ist nicht das erste Mal wegen der miserablen EEBUS Implementierung in dieser Wallbox. Dass die Wallbox keine Messwerte schickt, ist nicht von evcc oder EEBUS beeinflussbar! Die einzige Lösung hierfür ist, wie in der Dokumentation geschrieben, externe Messgeräte für jede Wallbox zu verwenden und in evcc zu konfigurieren.

Zu Elli und EEBUS: Der Stack läuft fehlerfrei mit allen anderen Wallboxen, Elli hat in dieser Wallbox eine sehr fehlerhafte EEBUS Implementierung. Es gibt wenige die mehr über EEBUS wissen als ich. Es auch bringt gar nichts Elli zu schreiben was sich in meinem Stack geändert hat, das sind tausende Zeilen Code. Diese Wallbox wird auch kein Software Update mehr bekommen. Vor allem vermute ich hier sehr stark, dass die Verhaltensänderung gar nicht an EEBUS liegt!

Bitte bitte bitte, hört auf selbst Ursachen zu definieren, ihr habt doch keine Ahnung was im Code alles geändert wurde. Das macht die Analyse nur noch viel schwieriger und Zeitaufwändiger.

winfotiker, 2024-08-05:
Hi,

ich habe für das Log beide Fahrzeuge Abends angeschlossen als nicht mehr
genug Sonne aufs Dach kam. Beide Fahrzeuge bzw LPs stehen auf PV. Beide
Fahrzeuge haben zunächst nicht geladen. Bis hier ist alles korrekt.

Der Passat (lp1) fing erst an zu laden als ich den ID von der Wallbox (lp2)
getrennt habe.
Das ist ein Fehlverhalten. Es gab nicht genug PV Strom um das Laden zu
starten.

Ich habe dann im Anschluss versucht über evcc das Laden manuell zu stoppen.
Das hat nicht funktioniert. Fehlverhalten.

Auch konnte ich lp1 nicht auf Schnellladen umstellen um wenigstens nicht
den Hausakku zu belasten. Die Box reagiert ab  dem Moment gar nicht mehr
auf Änderungen in evcc.

Zum Thema Fehler melden: Hier gilt für mich immer der gleiche Grundsatz.
Wenn die Implementierung vom eebus in der Elli Box fehlerhaft ist, will ich
denen das melden. Dazu wäre eine konkrete Beschreibung, in wie weit die
Implementierung fehlerhaft ist bzw woran ich das festmachen kann sehr
hilfreich. Im schlechtesten Fall könnte das sogar als Grund für eine
Wandlung hinreichend sein. Im besten Fall provoziert es doch nich ein
Update.
Danke und Gruß

DerAndereAndi, 2024-08-05:
Elli hat seit 2 Jahren eine genaue Fehlerliste von mir und war mit mir im Austausch. Sehe es mir nach, ich werde mich hier nicht immer wieder mit den Fehlerbeschreibungen wiederholen. Ich verbrenne genug Zeit ohne dass ich diese Box verwende.

Verstehe ich das richtig, dass um 19:42:39 das Auto nicht abgesteckt wurde?

winfotiker, 2024-08-05:
Nee. Um 19:42 hab ich den ID abgesteckt.
Der Fehler ist, dass der Passat auf LP1 dann spontan und ohne PV Strom
einfach anfängt zu laden. Das darf nicht sein.

DerAndereAndi, 2024-08-05:
Die CT-Spulen der Wallboxen sind nicht angeschlossen, korrekt?

winfotiker, 2024-08-05:
Korrekt

andig, 2024-08-06:
Ist drin- danke @DerAndereAndi

Sparlader, 2024-08-07:
Bei mir war auch das Problem elli wallbox und e-golf brachte mich zum verzweifeln:
  Laden begonnen, zurückgeschaltet auf min strom 1,4 bzw 2,8 kW bei 2 phasen, BEV nicht erkannt bzw warte auf fahrzeug, weiter geladen aber als Hausstrom angezeigt usw...
nach zurücksetzen wallbox auf Werkseinstellung (Achtung SKI neu) und löschen MQTT im UI klappt es plötzlich wieder.Warum oder Grund, keine Ahnung!!!

DerAndereAndi, 2024-08-07:
@Sparlader das konkrete Problem hier trat nur auf wenn 2 EEBUS Wallboxen angebunden waren, und an beiden ein EV hängt, und dann eines davon abgesteckt wurde.

closed 2024-08-09 by andig