evcc-io/evcc #14845: Add Wattsonic, Sunway, Solinteng, A-Tronix and St-ems

andig, 2024-07-12:
Finally i got a configuration for the Wattsonic Gen 2 inverter that reads pv/grid/battery power and the battery SOC

The same inverter (at least Gen3) is distributed as Sunway, Solinteng, A-Tronix and St-ems)

```
#############################################################################
#
# Meters - read from Wattsonic using Modbus RTu connected by /dev/ttyUSB0
#
# https://docs.evcc.io/docs/reference/modbus#manuelle-konfiguration
#
# All register share the same modbus bus configuration
#      source: modbus
#      id: 247
#      device: /dev/ttyUSB0
#      baudrate: 9600
#      comset: "8N1"
#      rtu: true
#      timeout: 10s         # 
#      scale: 1.0           # some registers require a scale, double value using ., negative values are supported
#      register:
#        type: holding      # coil, input, holding
#        address: >id<      # check documentation of you inverter
#        decode: >type<     # check documentation of you inverter [u]int16|32|64 , float32|64,
#
#
#############################################################################

meters:
  - name: meter-battery
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      register:
        type: holding
        decode: int32
        address: 40258    # Total_Backup_P
    soc:
      source: modbus
      id: 247
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      scale: 0.01
      timeout: 10s
      register:                         
        address: 43000    # SOC
        type: holding
        decode: int16   
  - name: meter-pv
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      register:                         
        address: 11028    # PV Input Total Power
        type: holding
        decode: int32
  - name: meter-grid
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      scale: -1.0
      register:
        address: 11000  # Total Power on Meter
        type: holding   
        decode: int32

```

Thanks to @andig for his patients with my misunderstanding of reading negative values from modbus.

_Originally posted by @negesti in https://github.com/evcc-io/evcc/discussions/14477#discussioncomment-10027717_

negesti, 2024-07-30:
i had to add a delay to the registers. Wattsonic uses baudrate 9600 and I had serial timeouts reading the registers at once (always the last register that is read fails with a timeout). 

After checking the sourcecode,  i notices that modbus supports a `delay` property (not listed in the docs). My current config is: 

```
#############################################################################
#
# Meters - read from Wattsonic using Modbus RTu connected by /dev/ttyACM0
#
# https://docs.evcc.io/docs/reference/modbus#manuelle-konfiguration
#
# All register share the same modbus bus configuration
#      source: modbus
#      id: 247
#      device: /dev/ttyACM0
#      baudrate: 9600
#      comset: "8N1"
#      rtu: true
#      timeout: 10s        
#      delay: 1s            # dont read all registers at once 
#      scale: 1.0           # some registers require a scale, double value using ., negative values are supported
#      register:
#        type: holding      # coil, input, holding
#        address: >id<      # check documentation of you inverter
#        decode: >type<     # check documentation of you inverter [u]int16|32|64 , float32|64,
#
#
#############################################################################

meters:
  - name: meter-grid
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      scale: -1.0
      delay: 3s
      register:
        address: 11000  # Total Power on Meter
        type: holding   
        decode: int32
  - name: meter-pv
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      delay: 2s
      register:
        address: 11028    # PV Input Total Power
        type: holding
        decode: int32
  - name: meter-battery
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      delay: 1s
      register:
        address: 40258    # Total_Backup_P
        type: holding
        decode: int32
    soc:
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      scale: 0.01
      timeout: 10s
      delay: 1s
      register:                         
        address: 43000    # SOC
        type: holding
        decode: int16
```

negesti, 2024-08-05:
Noticed, that the delay accepts ms and updated the config to read currents and powers from grid meter

```

meters:
  - name: meter-grid                               # Grid meter, power and L1-3 current, delay 0-1s
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      scale: -1.0
      delay: 100ms
      register:
        address: 11000  # Total Power on Meter
        type: holding   
        decode: int32
    currents:
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 200ms
        scale: 0.1
        register:
          address: 11010  # Grid Phase A Current
          type: holding
          decode: uint16
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 300ms
        scale: 0.1
        register:
          address: 11012  # Grid Phase B Current
          type: holding
          decode: uint16
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 400ms
        scale: 0.1
        register:
          address: 11014  # Grid Phase C Current
          type: holding
          decode: uint16
    powers:
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 500ms
        scale: -1
        register:
          address: 10994  # Phase A Power on Meter
          type: holding
          decode: int32
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 600ms
        scale: -1
        register:
          address: 10996  # Phase B Power on Meter
          type: holding
          decode: int32
      - source: modbus
        id: 247
        device: /dev/ttyACM0
        baudrate: 9600
        comset: "8N1"
        rtu: true
        timeout: 10s
        delay: 700ms
        scale: -1
        register:
          address: 10998  # Phase C Power on Meter
          type: holding
          decode: int32

  - name: meter-pv                                     # PV Metery, only power,  delay 1000ms
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      delay: 1000ms
      register:
        address: 11028    # PV Input Total Power
        type: holding
        decode: int32

  - name: meter-battery                               # Battery Meter,  power and soc, delay 2000ms
    type: custom
    power: # power (W)
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      timeout: 10s
      delay: 2000ms
      register:
        address: 40258    # Total_Backup_P
        type: holding
        decode: int32
    soc:
      source: modbus
      id: 247
      device: /dev/ttyACM0
      baudrate: 9600
      comset: "8N1"
      rtu: true
      scale: 0.01
      timeout: 10s
      delay: 2100ms
      register:                         
        address: 43000    # SOC
        type: holding
        decode: int16
```

closed 2024-09-14 by andig