evcc-io/evcc #14883: Addd device Schneider Electric EVlink Wallbox G4 Smart

Incom0de, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
It seems that Schneider EVlink Smart is not implemented. Got error _charger status: invalid status: 117_ when trying to add the G4 with EVlink pro configuration.

**Describe the solution you'd like**
Would be great when someone can implement the EVlink Smart devices

**Describe alternatives you've considered**
Maybe it is possible to add EVlink Wallbox G4 smart with the current Schneider Electric implementation, then it would be great if someone can post a working configuration.

**Additional context**
Documentation I got from Schneider Electric Support:
[EVSE_ModbusTable_V3.9.xlsx](https://github.com/user-attachments/files/16234444/EVSE_ModbusTable_V3.9.xlsx)
[EV_ParkingBoard_Modbus_Protocol-2.4.docx](https://github.com/user-attachments/files/16234445/EV_ParkingBoard_Modbus_Protocol-2.4.docx)

andig, 2024-07-29:
@Incom0de could you kindly test the PR?

Incom0de, 2024-07-30:
Sure, I would love to test it. Unfortunately, I'm not that deep into software development. How can I test the PR?