evcc-io/evcc #14885: Invalid vehicle SoC at charger: 65535.0%

marcokaiser, 2024-07-15:
### Describe the bug

After I attached my vehicle to the wallbox, it identifies the vehicle correctly. In the UI, it does not display a SoC though, and I get an error message in the logs:

`[lp-1  ] ERROR 2024/07/15 14:08:37 vehicle soc: invalid soc: 65535.0`

If I run `evcc vehicle`, I get the correct SoC value:

```/app # evcc vehicle
[main  ] INFO 2024/07/15 15:21:21 evcc 0.128.2
[main  ] INFO 2024/07/15 15:21:21 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/15 15:21:22 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/07/15 15:21:22 connecting evcc-1784660011 at tcp://mqtt.intern.kaiser.io:1883
car_ev6
-------
Soc:           91%
Capacity:      77.0kWh
Charge status: B
Range:         400km
Odometer:      23912km
Finish time:   2024-07-15 16:06:00 +0200 CEST
Position:      53.556147,9.460567
Limit Soc:     100%
Identifiers:   [04:E7:7E:71:99:CE 04E77E7199CE]
```

The charger reports the wrong SoC though:

```/app # evcc charger
[main  ] INFO 2024/07/15 15:15:03 evcc 0.128.2
[main  ] INFO 2024/07/15 15:15:03 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/15 15:15:03 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/07/15 15:15:03 connecting evcc-1771356870 at tcp://mqtt.intern.kaiser.io:1883
Power:          4213W
Energy:         4458.7kWh
Current L1..L3: 6.09A 5.76A 6.09A
Voltage L1..L3: 236V 237V 236V
Soc:            65535%
Charge status:  C
Enabled:        true
Identifier:     XXXXXXXXXXX
```

Running 0.128.2, issue appeared after updating to that version, I believe. Updating car status from Home Assistant or the Kia app works fine.

### Steps to reproduce

1. connect car (EV6)
2. vehicle gets correctly identified
3. no SoC in UI, error in logs every couple seconds

### Configuration details

```yaml
chargers:
  - name: wallbox
    type: template
    template: bender
    host: xxx # IP-Adresse oder Hostname
    port: 502 # Port # Optional

loadpoints:
  - title: Carport # display name for UI
    charger: wallbox # Wallbox Referenz
    mode: off # charge mode (off, now, minpv, pv)

vehicles:
  - name: car_ev6
    type: template
    template: kia
    user: xxx
    password: "xxx"
    vin: xxx
    title: Kia EV6
    capacity: 77
    identifiers:
      - xxx
      - xxx
  - name: car_minise
    type: template
    template: mini
    user: xxx
    password: "xxx"
    title: Mini Cooper SE
    capacity: 29
    identifiers:
      - XXX
      - XXX

site:
  - title: MySite
    meters:
      grid: grid1
      pv:
        - pv1
      battery:
        - battery1
```

### Log details

```text
Nothing in the logs except the error message many times:

[lp-1  ] ERROR 2024/07/15 14:08:37 vehicle soc: invalid soc: 65535.0

Let me know if I can provide other specific log or trace outputs
```

### What type of operating system are you running?

Docker container

### Version

0.128.2

VolkerK62, 2024-07-15:
liegt vermutlich an https://github.com/evcc-io/evcc/pull/14360

mfuchs1984, 2024-07-15:
Im Hinblick auf die Implementierung wäre es wohl gut, alles außerhalb 0-100 zu verwerfen und den vehicle SOC (sofern verfügbar) zu verwenden? Die Modbus spec. sagt "EV Battery State (% 0-100)". Ob die 65535 von der Wallbox oder vom Fahrzeug kommen, ist wohl schwer herauszufinden.

marcokaiser, 2024-07-15:
> Hast du ISO 15118 an der Wallbox aktiviert? Unterstützt der EV6 das schon vollständig? Wenn du keinen Anwendungsfall dafür hast, kannst du es an der Wallbox deaktivieren. Habe selber eine Bender Wallbox (Spelsberg Smart Pro) und nutze sie mit AutoCharge aber ohne ISO 15118 mit einem BMW i4 ohne das Problem.

Okay, ISO 15118 aus und Autocharge an hat geholfen - danke für den Tip! Und jetzt sehe ich auch den Identifier (MAC-Adresse) des Autos im Charger. Nice.

mfuchs1984, 2024-07-15:
Super! (Dennoch denke ich, das ungültige werte im SOC Register hier abgefangen werden sollten.)

andig, 2024-07-15:
@premultiply 65535 ist MaxUint16

andig, 2024-07-15:
@marcokaiser bitte mal

    evcc charger --log trace,db:error

marcokaiser, 2024-07-15:
@andig here you go:

```
app # evcc charger --log trace,db:error
[main  ] INFO 2024/07/15 18:14:12 evcc 0.128.2
[main  ] INFO 2024/07/15 18:14:12 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/07/15 18:14:12 connecting evcc-169003527 at tcp://mqtt.xxx.yyy:1883
[mqtt  ] DEBUG 2024/07/15 18:14:12 tcp://mqtt.xxx.yyy:1883 connected
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 01 00 00 00 06 ff 03 00 8e 00 0a
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 01 00 00 00 17 ff 03 14 20 54 45 43 48 4e 49 56 4f 4c 54 31 31 30 30 53 4d 41 52 54
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 02 00 00 00 06 ff 03 00 dc 00 02
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 03 00 00 00 06 ff 03 00 de 00 02
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 eb
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 04 00 00 00 06 ff 03 02 da 00 01
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 04 00 00 00 05 ff 03 02 ff ff
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 05 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 05 00 00 00 05 ff 03 02 00 00
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 06 00 00 00 06 ff 03 02 d0 00 0a
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 06 00 00 00 17 ff 03 14 20 20 20 20 20 20 20 20 30 34 45 37 37 45 37 31 39 39 43 45
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 07 00 00 00 06 ff 03 00 dc 00 02
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 00 00
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 08 00 00 00 06 ff 03 00 da 00 02
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 08 00 00 00 07 ff 03 04 00 44 21 0d
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 09 00 00 00 06 ff 03 00 d4 00 06
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 09 00 00 00 0f ff 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 0a 00 00 00 06 ff 03 00 de 00 06
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 0a 00 00 00 0f ff 03 0c 00 00 00 eb 00 00 00 ee 00 00 00 e5
[bender] TRACE 2024/07/15 18:14:12 modbus: send 00 0b 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:14:12 modbus: recv 00 0b 00 00 00 05 ff 03 02 00 00
[bender] TRACE 2024/07/15 18:14:15 modbus: send 00 0c 00 00 00 06 ff 03 00 7a 00 01
[bender] TRACE 2024/07/15 18:14:15 modbus: recv 00 0c 00 00 00 05 ff 03 02 00 02
[bender] TRACE 2024/07/15 18:14:15 modbus: send 00 0d 00 00 00 06 ff 03 03 e8 00 01
[bender] TRACE 2024/07/15 18:14:15 modbus: recv 00 0d 00 00 00 05 ff 03 02 00 00
[bender] TRACE 2024/07/15 18:14:15 modbus: send 00 0e 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:14:15 modbus: recv 00 0e 00 00 00 05 ff 03 02 00 00
[bender] TRACE 2024/07/15 18:14:15 modbus: send 00 0f 00 00 00 06 ff 03 02 d0 00 0a
[bender] TRACE 2024/07/15 18:14:15 modbus: recv 00 0f 00 00 00 17 ff 03 14 20 20 20 20 20 20 20 20 30 34 45 37 37 45 37 31 39 39 43 45
Power:          0W
Energy:         4464.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 235V 238V 229V
.Soc:           not available
Charge status:  B
Enabled:        false
Identifier:     XXXXXXXXX
```

andig, 2024-07-15:
Wäre natürlich gut, wenn das Log den Fehler zeigen würde ;)

mfuchs1984, 2024-07-15:
Das Log ist mit deaktivierten ISO 15118 erstellt worden, oder?

marcokaiser, 2024-07-15:
@andig tut es doch, oder? siehe die Abfrage nach 0x2da, da kommt 0xff 0xff zurück. 

Jetzt ist ISO 15118 aus, ja - vorher stand da das, was ich im Originalpost drin hatte:

```
Power:          4213W
Energy:         4458.7kWh
Current L1..L3: 6.09A 5.76A 6.09A
Voltage L1..L3: 236V 237V 236V
Soc:            65535%
Charge status:  C
Enabled:        true
Identifier:     XXXXXXXXXXX
```

Ich kann jetzt nochmal ISO 15118 anmachen und die WB neu starten - aber es wird genau das rauskommen, nur die SoC-Zeile ist anders. Wenn sein muss, mach ich das natürlich nochmal... ich hatte es vorhin gemacht, die Shell auf den EVCC-Container ist aber inzwischen zu und ich habs nicht mehr in der History.

marcokaiser, 2024-07-15:
Hier mit ISO 15118 (mit Plug&Charge) aktiviert: 

```
app # evcc charger --log trace,db:error
[main  ] INFO 2024/07/15 18:38:45 evcc 0.128.2
[main  ] INFO 2024/07/15 18:38:45 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/07/15 18:38:46 connecting evcc-1991046196 at tcp://mqtt.xxx.yyy:1883
[mqtt  ] DEBUG 2024/07/15 18:38:46 tcp://mqtt.xxx.yyy:1883 connected
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 01 00 00 00 06 ff 03 00 8e 00 0a
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 01 00 00 00 17 ff 03 14 20 54 45 43 48 4e 49 56 4f 4c 54 31 31 30 30 53 4d 41 52 54
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 02 00 00 00 06 ff 03 00 dc 00 02
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 03 00 00 00 06 ff 03 00 de 00 02
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 e9
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 04 00 00 00 06 ff 03 02 da 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 04 00 00 00 05 ff 03 02 ff ff
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 05 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 05 00 00 00 05 ff 03 02 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 06 00 00 00 06 ff 03 02 e5 00 06
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 06 00 00 00 0f ff 03 0c 30 30 30 30 30 30 30 30 30 30 30 30
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 07 00 00 00 06 ff 03 00 dc 00 02
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 00 00
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 08 00 00 00 06 ff 03 00 da 00 02
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 08 00 00 00 07 ff 03 04 00 44 21 0d
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 09 00 00 00 06 ff 03 00 d4 00 06
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 09 00 00 00 0f ff 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0a 00 00 00 06 ff 03 00 de 00 06
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0a 00 00 00 0f ff 03 0c 00 00 00 e9 00 00 00 ea 00 00 00 e8
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0b 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0b 00 00 00 05 ff 03 02 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0c 00 00 00 06 ff 03 02 da 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0c 00 00 00 05 ff 03 02 ff ff
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0d 00 00 00 06 ff 03 00 7a 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0d 00 00 00 05 ff 03 02 00 02
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0e 00 00 00 06 ff 03 03 e8 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0e 00 00 00 05 ff 03 02 00 00
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 0f 00 00 00 06 ff 03 02 e4 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 0f 00 00 00 05 ff 03 02 00 01
[bender] TRACE 2024/07/15 18:38:46 modbus: send 00 10 00 00 00 06 ff 03 02 e5 00 06
[bender] TRACE 2024/07/15 18:38:46 modbus: recv 00 10 00 00 00 0f ff 03 0c 30 30 30 30 30 30 30 30 30 30 30 30
Power:          0W
Energy:         4464.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 233V 234V 232V
Soc:            65535%
Charge status:  B
Enabled:        false
Identifier:     000000000000
```

closed 2024-07-22 by andig

mfuchs1984, 2024-09-17:
Nein, er liest mit AutoCharge gar keinen SoC aus.

PR #14972 (2024-07-22): Bender: Fix SoC range if NaN
Fixes #14885