evcc-io/evcc #14886: custom meter does not support usage

jomach, 2024-07-15:
### Describe the bug

I think there is a small bug on a meter from type custom. I get the error: `has invalid keys: usage` I cannot add the usage pv or grid for this meter.

### Steps to reproduce

add a meter from type custom

### Configuration details

```yaml
- type: custom
  usage: pv
  name: Huawei-SUN2000
  power:
    source: http
    uri: http://homeassistant.local:8123/api/states/sensor.inverter_active_power
    auth:
      type: bearer
      password: <some_token>
    jq: .state
  energy:
    source: http
    uri: http://homeassistant.local:8123/api/states/sensor.inverter_total_yield
    auth:
      type: bearer
      password: <some token>
    jq: .state
```

### Log details

```text
[main  ] INFO 2024/07/15 15:37:05 listening at :7070
[main  ] FATAL 2024/07/15 15:37:05 cannot create meter 'Huawei-SUN2000': cannot create meter type 'custom': decoding failed due to the following error(s):

'' has invalid keys: usage
[main  ] FATAL 2024/07/15 15:37:05 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.2

VolkerK62, 2024-07-15:
`usage` is only used within templates to adjust the correct scale or to use the correct register
For what purpose a meter is used, is configured within the `site` (or `loadpoint`)

jomach, 2024-07-15:
@VolkerK62 thanks for the explanation. On the docs examples it is always referenced on the meters.

closed 2024-07-15 by jomach

VolkerK62, 2024-07-15:
but only with templates, not at the custom meter https://docs.evcc.io/docs/devices/meters#manuell