evcc-io/evcc #14890: Charger: Delta AC MAX Smart - Modbus TCP - error: modbus: exception '2' (illegal data address), function '4'

madmat17, 2024-07-15:
### Describe the bug

Running evcc with Delta AC MAX Smart (via Modbus TCP) throws an error as follows:

`Carport: charger status: modbus: exception '2' (illegal data address), function '4'`

It seems as if evcc is trying to read and write non-existing registers:
```
[lp-1 ] ERROR 2024/07/15 21:16:35 charge rater: read tcp 127.0.0.1:53432->127.0.0.1:5022: i/o timeout
[delta ] TRACE 2024/07/15 21:16:35 modbus: send 00 25 00 00 00 06 01 04 03 e9 00 01
[proxy-5022] TRACE 2024/07/15 21:16:35 read input: id 1 addr 1001 qty 1
[proxy-5022] TRACE 2024/07/15 21:16:41 read input: modbus: exception '2' (illegal data address), function '4'
[proxy-5022] TRACE 2024/07/15 21:16:44 read input: modbus: exception '2' (illegal data address), function '4'
[delta ] TRACE 2024/07/15 21:16:44 modbus: recv 00 25 00 00 00 03 01 84 02
[lp-1 ] ERROR 2024/07/15 21:16:44 charger status: modbus: exception '2' (illegal data address), function '4'
[site ] DEBUG 2024/07/15 21:16:44 ----
[delta ] TRACE 2024/07/15 21:16:44 modbus: send 00 26 00 00 00 06 01 04 03 ed 00 02
[proxy-5022] TRACE 2024/07/15 21:16:44 read input: id 1 addr 1005 qty 2
[proxy-5022] TRACE 2024/07/15 21:16:47 read input: 00000000
[delta ] TRACE 2024/07/15 21:16:47 modbus: recv 00 26 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/15 21:16:47 charge power: 0W
[modbus] TRACE 2024/07/15 21:16:47 modbus: send 00 6f 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/07/15 21:16:50 modbus: recv 00 6f 00 00 00 07 01 03 04 00 00 00 00
[modbus] TRACE 2024/07/15 21:16:50 modbus: send 00 70 00 00 00 06 01 03 7d 6a 00 02
[modbus] TRACE 2024/07/15 21:16:51 modbus: recv 00 70 00 00 00 07 01 03 04 00 0c bd d8
[site ] DEBUG 2024/07/15 21:16:51 pv power: 0W
[modbus] TRACE 2024/07/15 21:16:51 modbus: send 00 71 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/07/15 21:16:53 modbus: recv 00 71 00 00 00 07 01 03 04 ff ff fe 3d
[modbus] TRACE 2024/07/15 21:16:53 modbus: send 00 72 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/15 21:16:55 modbus: recv 00 72 00 00 00 07 01 03 04 00 03 79 f4
[modbus] TRACE 2024/07/15 21:16:55 modbus: send 00 73 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/15 21:16:56 modbus: recv 00 73 00 00 00 05 01 03 02 00 96
[site ] DEBUG 2024/07/15 21:16:56 battery soc: 15%
[site ] DEBUG 2024/07/15 21:16:56 battery power: 451W
[modbus] TRACE 2024/07/15 21:16:56 modbus: send 00 74 00 00 00 06 01 03 90 f9 00 02
[modbus] TRACE 2024/07/15 21:16:58 modbus: recv 00 74 00 00 00 07 01 03 04 00 00 00 0e
[site ] DEBUG 2024/07/15 21:16:58 grid meter: -14W
[modbus] TRACE 2024/07/15 21:16:58 modbus: send 00 75 00 00 00 06 01 03 90 f3 00 02
[modbus] TRACE 2024/07/15 21:16:59 modbus: recv 00 75 00 00 00 07 01 03 04 00 00 00 4e
[modbus] TRACE 2024/07/15 21:16:59 modbus: send 00 76 00 00 00 06 01 03 90 f5 00 02
[modbus] TRACE 2024/07/15 21:16:59 modbus: recv 00 76 00 00 00 07 01 03 04 00 00 00 3f
[modbus] TRACE 2024/07/15 21:16:59 modbus: send 00 77 00 00 00 06 01 03 90 f7 00 02
[modbus] TRACE 2024/07/15 21:17:00 modbus: recv 00 77 00 00 00 07 01 03 04 ff ff ff 89
[site ] DEBUG 2024/07/15 21:17:00 grid currents: [-0.78 -0.63 1.19]A
[modbus] TRACE 2024/07/15 21:17:00 modbus: send 00 78 00 00 00 06 01 03 91 01 00 02
[modbus] TRACE 2024/07/15 21:17:01 modbus: recv 00 78 00 00 00 07 01 03 04 00 05 c6 94
[site ] DEBUG 2024/07/15 21:17:01 site power: 587W
[delta ] TRACE 2024/07/15 21:17:01 modbus: send 00 27 00 00 00 06 01 04 03 fb 00 02
[proxy-5022] TRACE 2024/07/15 21:17:01 read input: id 1 addr 1019 qty 2
[lp-1 ] ERROR 2024/07/15 21:17:11 charge rater: read tcp 127.0.0.1:55746->127.0.0.1:5022: i/o timeout
[delta ] TRACE 2024/07/15 21:17:11 modbus: send 00 28 00 00 00 06 01 04 03 e9 00 01
[proxy-5022] TRACE 2024/07/15 21:17:11 read input: id 1 addr 1001 qty 1
[proxy-5022] TRACE 2024/07/15 21:17:17 read input: modbus: exception '2' (illegal data address), function '4'
[proxy-5022] DEBUG 2024/07/15 21:17:17 failed to write response: write tcp 127.0.0.1:5022->127.0.0.1:55746: i/o timeout
```

From what I can tell from within the [delta.go](https://github.com/evcc-io/evcc/blob/master/charger/delta.go) file it looks as if quite a few registers are being read that are not present for the Delta AC MAX chargers.
The following modbus registers can be read, others don't - commented in Homeassistant `configuration.yaml` code:

```
modbus:
  - name: "Delta AC MAX Smart"
    type: tcp
    host: 127.0.0.1
    port: 5022 #port of EVCC modbus proxy
    sensors:
      - name: "Delta AC MAX State"
        unique_id: delta_ac_max_state
        device_address: 10
        data_type: uint16
[... 509 lines cut ...]
[modbus] TRACE 2024/07/15 21:16:59 modbus: recv 00 75 00 00 00 07 01 03 04 00 00 00 4e
[modbus] TRACE 2024/07/15 21:16:59 modbus: send 00 76 00 00 00 06 01 03 90 f5 00 02
[modbus] TRACE 2024/07/15 21:16:59 modbus: recv 00 76 00 00 00 07 01 03 04 00 00 00 3f
[modbus] TRACE 2024/07/15 21:16:59 modbus: send 00 77 00 00 00 06 01 03 90 f7 00 02
[modbus] TRACE 2024/07/15 21:17:00 modbus: recv 00 77 00 00 00 07 01 03 04 ff ff ff 89
[site ] DEBUG 2024/07/15 21:17:00 grid currents: [-0.78 -0.63 1.19]A
[modbus] TRACE 2024/07/15 21:17:00 modbus: send 00 78 00 00 00 06 01 03 91 01 00 02
[modbus] TRACE 2024/07/15 21:17:01 modbus: recv 00 78 00 00 00 07 01 03 04 00 05 c6 94
[site ] DEBUG 2024/07/15 21:17:01 site power: 587W
[delta ] TRACE 2024/07/15 21:17:01 modbus: send 00 27 00 00 00 06 01 04 03 fb 00 02
[proxy-5022] TRACE 2024/07/15 21:17:01 read input: id 1 addr 1019 qty 2
[lp-1 ] ERROR 2024/07/15 21:17:11 charge rater: read tcp 127.0.0.1:55746->127.0.0.1:5022: i/o timeout
[delta ] TRACE 2024/07/15 21:17:11 modbus: send 00 28 00 00 00 06 01 04 03 e9 00 01
[proxy-5022] TRACE 2024/07/15 21:17:11 read input: id 1 addr 1001 qty 1
[proxy-5022] TRACE 2024/07/15 21:17:17 read input: modbus: exception '2' (illegal data address), function '4'
[proxy-5022] DEBUG 2024/07/15 21:17:17 failed to write response: write tcp 127.0.0.1:5022->127.0.0.1:55746: i/o timeout
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.2

andig, 2024-07-15:
Can you please show

    evcc charger —log trace,db:error

madmat17, 2024-07-16:
Hi andig,

I'd love to - but I can't SSH into the containter (SSH on port 22222 is kept deactivated due to security concerns - only the limited port 22 is open).
That'd be the full trace from the GUI:

```
[main  ] INFO 2024/07/16 06:35:07 evcc 0.128.2
[main  ] INFO 2024/07/16 06:35:07 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/16 06:35:08 using sqlite database: /data/evcc.db
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[main  ] INFO 2024/07/16 06:35:08 listening at :7070
[proxy-5022] DEBUG 2024/07/16 06:35:08 modbus proxy for deltaac.lan:502 listening at :5022
[db    ] TRACE 2024/07/16 06:35:08 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/07/16 06:35:08 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[delta ] TRACE 2024/07/16 06:35:08 modbus: send 00 01 00 00 00 06 01 03 00 ca 00 01
[proxy-5022] TRACE 2024/07/16 06:35:08 read holdings: id 1 addr 202 qty 1
[proxy-5022] TRACE 2024/07/16 06:35:09 read holding: 0258
[delta ] TRACE 2024/07/16 06:35:09 modbus: recv 00 01 00 00 00 05 01 03 02 02 58
[db    ] TRACE 2024/07/16 06:35:09 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[bmw   ] TRACE 2024/07/16 06:35:09 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw   ] TRACE 2024/07/16 06:35:09 client_id=********-****-****-****-************&code_challenge=********&code_challenge_method=S256&grant_type=authorization_code&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=***********&username=***
--
{"redirect_to" : "redirect_uri=com.bmw.connected://oauth?client_id=********-****-****-****-************&response_type=code&scope=openid profile email offline_access smacc vehicle_data perseus dlm svds cesim vsapi remote_services fupo authenticate_user&state=*****&authorization=*****.******..*"}
[bmw   ] TRACE 2024/07/16 06:35:09 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw   ] TRACE 2024/07/16 06:35:09 authorization=*****..%2A&client_id=********-****-****-****-************&code_challenge=*****&code_challenge_method=S256&nonce=login_nonce&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=*****
[bmw   ] TRACE 2024/07/16 06:35:09 POST https://customer.bmwgroup.com/gcdm/oauth/token
[bmw   ] TRACE 2024/07/16 06:35:09 code=*****&code_verifier=*****&grant_type=authorization_code&redirect_uri=com.bmw.connected%3A%2F%2Foauth
--
{"gcid":"****","token_type":"Bearer","access_token":"****","refresh_token":"****","scope":"smacc vehicle_data perseus dlm svds openid profile vsapi remote_services authenticate_user cesim offline_access email fupo","expires_in":3599,"id_token":"****"}
[bmw   ] TRACE 2024/07/16 06:35:09 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles?apptimezone=120&appDateTime=1721104509924
[bmw   ] TRACE 2024/07/16 06:35:10 [{"vin":"****","mappingInfo":{"isAssociated":true,"isLmmEnabled":true,"mappingStatus":"CONFIRMED","isPrimaryUser":true},"appVehicleType":"CONNECTED","attributes":{"lastFetched":"2024-07-16T04:35:10.036Z","model":"i4 eDrive35","year":2024,"color":4280233344,"brand":"BMW","driveTrain":"ELECTRIC","headUnitType":"MGU","headUnitRaw":"MGU_02_L","hmiVersion":"ID8","softwareVersionCurrent":{"puStep":{"month":3,"year":24},"iStep":540,"seriesCluster":"S18A"},"softwareVersionExFactory":{"puStep":{ [... cut ...]
[site  ] TRACE 2024/07/16 06:38:38 ignoring excess DC charging due to grid consumption: 274W > 50W
[site  ] DEBUG 2024/07/16 06:38:38 site power: 424W
[delta ] TRACE 2024/07/16 06:38:38 modbus: send 00 17 00 00 00 06 01 04 03 fb 00 02
[proxy-5022] TRACE 2024/07/16 06:38:38 read input: id 1 addr 1019 qty 2
[lp-1  ] ERROR 2024/07/16 06:38:48 charge rater: read tcp 127.0.0.1:45512->127.0.0.1:5022: i/o timeout
[delta ] TRACE 2024/07/16 06:38:48 modbus: send 00 18 00 00 00 06 01 04 03 e9 00 01
[proxy-5022] TRACE 2024/07/16 06:38:48 read input: id 1 addr 1001 qty 1

```

LingLing-de, 2024-07-21:
Hi, have the same error on my Delta AC MAX Smart, if you need additional logs let me know.

I'm on: 
MPU Firmware version :  01.26.34.00
MCU Firmware version :  01.00.36.00
evcc v0.128.3 stable

madmat17, 2024-07-22:
@andig, I think(!) that I know how it came that modbus-tcp isn't working as expected.
The implementation of the Wallbox (in the `delta.go` file) was seemingly based on the prework done in German speaking forum [https://www.photovoltaikforum.com/thread/209699-delta-ac-max-basic-wallbox-modus-rtu-rs485-wie/](https://www.photovoltaikforum.com/thread/209699-delta-ac-max-basic-wallbox-modus-rtu-rs485-wie/) where a **WaveShare ModBus RS485 PoE TCP/RTU** module was used to read/write the registers via modbus RTU RS485. The native modbus TCP implementation of that wallbox however, doesn't present all regsiters (even though Delta's modbus documentation doesn't distinguish between modbus TCP and RS485).
I just found that the same thing is being discussed [here](https://github.com/evcc-io/evcc/discussions/13031) as well.

JohannesKlug forked his own version, but reported that he couldn't set the charging power (https://github.com/evcc-io/evcc/discussions/13031#discussioncomment-9099149)

OCPP doesn't seem to work flawlessly on my end (the GUI of evcc reports "Warte auf Fahrzeug" / "Waiting for vehicle" - but doesn't initiate charging.

Therefore, I wanted to set a custom charger in the `evcc.yaml` - but struggling massively, as the wallbox doesn't seem to offer an "enabled" register. Would a shell script (which I don't know yet how to get it running on Homeassistant yet) the only workaround?

Greeting from Austria,
Mat

premultiply, 2024-07-22:
> OCPP doesn't seem to work flawlessly on my end (the GUI of evcc reports "Warte auf Fahrzeug" / "Waiting for vehicle" - but doesn't initiate charging.

Did you try using swiping your rfid card or setting the charger to "free vending" mode?

Please include an ocpp trace log.

andig, 2024-07-22:
@premultiply: this issue is about Modbus. Let's please leave it at that. This is NOT about OCPP.

madmat17, 2024-07-22:
> @premultiply: this issue is about Modbus. Let's please leave it at that. This is NOT about OCPP.

Exactly!

@premultiply: Thanks for the hint, but OCPP was only mentioned to underline why I'm trying to get modbus TCP up and running. ;-)

premultiply, 2024-07-22:
@madmat17 I am still interessted in the OCPP issues regarding Delta chargers. Please open another issue for that.

andig, 2024-07-22:
Please test PR

madmat17, 2024-07-22:
@andig Will do as soon as possible. Right now I'm too stupid (or tired) to fork and set the fix/delta branch as main branch for getting it loaded into Homeassistant. I probably will set up a spare RPi with the fix during the weekend.

@premultiply I certainly can do so, but cannot provide any logs as I won't tinker with OCPP in coming days/weeks, nor have I backed up the OCPP logs. Still worth opening an issue?

madmat17, 2024-07-22:
I got the [fix/delta branch](https://github.com/evcc-io/evcc/tree/fix/delta) installed - cup of coffee helped...
Situation improved ( `charge rater` error is gone), but there's still something odd with getting the wallboxe's status (trace log below).

The `deltaRegEvseState` should read the input register at address 1000.
But I'm not sure that this is the case when looking at the `delta.go` code - but I'm a bloody n00p in coding.

```
Line 43:   deltaRegEvseState                 = 0 
(...)
Line 83: func NewDelta(uri, device, comset string, baudrate int, proto modbus.Protocol, slaveID uint8, connector uint16) 
Line 84: (api.Charger, error) {
	conn, err := modbus.NewConnection(uri, device, comset, baudrate, proto, slaveID)
(...)
Line 102: wb.base = connector * 1000
(...)
Line 131: func (wb *Delta) Status() (api.ChargeStatus, error) {
Line 132: 	b, err := wb.conn.ReadInputRegisters(wb.base+deltaRegEvseState, 1)
Line 133: 	if err != nil {
Line 134: 		return api.StatusNone, err
Line 135: 	}
```

Where does the program get the integer "connector" from? It should be all fine as long as its value is "1".
On the other hand side the log is reporting a "charge power" reading (or is this coming from some sort of fallback?

The EVSE state register on address 1000 (uint16) is returning "1" (= available) when I read it directly via modbus TCP - and also changing to e.g. 4 when charging. Thus, it's definitely present.

'deltaRegEvseChargerState' doesn't seem to be presented by the Delta AC MAX Smart via modbus TCP (at least I don't get any readings from my direct modbus connection.

**Trace log:**
```
[site ] DEBUG 2024/07/22 16:07:56 ----
[delta ] TRACE 2024/07/22 16:07:56 modbus: send 00 6d 00 00 00 06 0a 04 03 ed 00 02
[delta ] TRACE 2024/07/22 16:07:56 modbus: recv 00 6d 00 00 00 07 0a 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/22 16:07:56 charge power: 0W
[modbus] TRACE 2024/07/22 16:07:56 modbus: send 01 55 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/07/22 16:07:57 modbus: recv 01 55 00 00 00 07 01 03 04 00 00 08 e5
[modbus] TRACE 2024/07/22 16:07:57 modbus: send 01 56 00 00 00 06 01 03 7d 6a 00 02
[modbus] TRACE 2024/07/22 16:07:57 modbus: recv 01 56 00 00 00 07 01 03 04 00 0c ef d7
[site ] DEBUG 2024/07/22 16:07:57 pv power: 2277W
[modbus] TRACE 2024/07/22 16:07:57 modbus: send 01 57 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/07/22 16:07:57 modbus: recv 01 57 00 00 00 07 01 03 04 00 00 00 16
[modbus] TRACE 2024/07/22 16:07:57 modbus: send 01 58 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/22 16:07:58 modbus: recv 01 58 00 00 00 07 01 03 04 00 03 84 b2
[modbus] TRACE 2024/07/22 16:07:58 modbus: send 01 59 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/22 16:08:00 modbus: recv 01 59 00 00 00 05 01 03 02 03 e8
[site ] DEBUG 2024/07/22 16:08:00 battery soc: 100%
[site ] DEBUG 2024/07/22 16:08:00 battery power: -22W
[modbus] TRACE 2024/07/22 16:08:00 modbus: send 01 5a 00 00 00 06 01 03 90 f9 00 02
[modbus] TRACE 2024/07/22 16:08:00 modbus: recv 01 5a 00 00 00 07 01 03 04 00 00 07 91
[site ] DEBUG 2024/07/22 16:08:00 grid meter: -1937W
[modbus] TRACE 2024/07/22 16:08:00 modbus: send 01 5b 00 00 00 06 01 03 90 f3 00 02
[modbus] TRACE 2024/07/22 16:08:02 modbus: recv 01 5b 00 00 00 07 01 03 04 00 00 01 3c
[modbus] TRACE 2024/07/22 16:08:02 modbus: send 01 5c 00 00 00 06 01 03 90 f5 00 02
[modbus] TRACE 2024/07/22 16:08:05 modbus: recv 01 5c 00 00 00 07 01 03 04 00 00 01 28
[modbus] TRACE 2024/07/22 16:08:05 modbus: send 01 5d 00 00 00 06 01 03 90 f7 00 02
[modbus] TRACE 2024/07/22 16:08:06 modbus: recv 01 5d 00 00 00 07 01 03 04 00 00 00 a7
[site ] DEBUG 2024/07/22 16:08:06 grid currents: [-3.16 -2.96 -1.67]A
[modbus] TRACE 2024/07/22 16:08:06 modbus: send 01 5e 00 00 00 06 01 03 91 01 00 02
[modbus] TRACE 2024/07/22 16:08:07 modbus: recv 01 5e 00 00 00 07 01 03 04 00 05 d4 55
[site ] DEBUG 2024/07/22 16:08:07 site power: -1809W
[delta ] TRACE 2024/07/22 16:08:07 modbus: send 00 6e 00 00 00 06 0a 04 03 fb 00 02
[delta ] TRACE 2024/07/22 16:08:08 modbus: recv 00 6e 00 00 00 07 0a 04 04 00 00 00 00
[delta ] TRACE 2024/07/22 16:08:08 modbus: send 00 6f 00 00 00 06 0a 04 03 e9 00 01
[delta ] TRACE 2024/07/22 16:08:10 modbus: recv 00 6f 00 00 00 03 0a 84 02
[lp-1 ] ERROR 2024/07/22 16:08:10 charger status: modbus: exception '2' (illegal data address), function '4'
[site ] DEBUG 2024/07/22 16:08:26 ----
[delta ] TRACE 2024/07/22 16:08:26 modbus: send 00 70 00 00 00 06 0a 04 03 ed 00 02
[delta ] TRACE 2024/07/22 16:08:26 modbus: recv 00 70 00 00 00 07 0a 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/22 16:08:26 charge power: 0W
[... 42 lines cut ...]
[modbus] TRACE 2024/07/22 16:08:58 modbus: send 01 6c 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/22 16:08:59 modbus: recv 01 6c 00 00 00 07 01 03 04 00 03 84 b2
[modbus] TRACE 2024/07/22 16:08:59 modbus: send 01 6d 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/22 16:08:59 modbus: recv 01 6d 00 00 00 05 01 03 02 03 e8
[site ] DEBUG 2024/07/22 16:08:59 battery soc: 100%
[site ] DEBUG 2024/07/22 16:08:59 battery power: -24W
[modbus] TRACE 2024/07/22 16:08:59 modbus: send 01 6e 00 00 00 06 01 03 90 f9 00 02
[modbus] TRACE 2024/07/22 16:09:01 modbus: recv 01 6e 00 00 00 07 01 03 04 00 00 04 e6
[site ] DEBUG 2024/07/22 16:09:01 grid meter: -1254W
[modbus] TRACE 2024/07/22 16:09:01 modbus: send 01 6f 00 00 00 06 01 03 90 f3 00 02
[modbus] TRACE 2024/07/22 16:09:01 modbus: recv 01 6f 00 00 00 07 01 03 04 00 00 00 e0
[... cut ...]
[modbus] TRACE 2024/07/22 16:09:05 modbus: recv 01 72 00 00 00 07 01 03 04 00 05 d4 55
[site ] DEBUG 2024/07/22 16:09:05 site power: -1128W
[delta ] TRACE 2024/07/22 16:09:05 modbus: send 00 74 00 00 00 06 0a 04 03 fb 00 02
[delta ] TRACE 2024/07/22 16:09:06 modbus: recv 00 74 00 00 00 07 0a 04 04 00 00 00 00
[delta ] TRACE 2024/07/22 16:09:06 modbus: send 00 75 00 00 00 06 0a 04 03 e9 00 01
[delta ] TRACE 2024/07/22 16:09:08 modbus: recv 00 75 00 00 00 03 0a 84 02
[lp-1 ] ERROR 2024/07/22 16:09:08 charger status: modbus: exception '2' (illegal data address), function '4'
```

Klaus3d3, 2024-07-22:
Hi, i fanally got this thing working for about 1min.
I had to follow a specific reset procedure for this.
But sadly the box switches off after about 1min. Here is the tracelog.:
[modbus] TRACE 2024/07/22 17:14:04 modbus: send 06 c2 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2024/07/22 17:14:04 modbus: send 06 c2 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2024/07/22 17:14:04 modbus: send 06 c2 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2024/07/22 17:14:04 modbus: send 06 c2 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2024/07/22 17:14:04 modbus: recv 06 c2 00 00 00 07 f7 03 04 49 da 68 80
[site ] DEBUG 2024/07/22 17:14:04 site power: -2274W
[lp-1 ] ERROR 2024/07/22 17:14:07 charge rater: dial tcp 192.168.0.181:502: connect: no route to host
[lp-1 ] ERROR 2024/07/22 17:14:10 charger status: dial tcp 192.168.0.181:502: connect: no route to host
[site ] DEBUG 2024/07/22 17:14:10 ----
[mqtt ] TRACE 2024/07/22 17:14:12 recv evc/power: 'NaN'
[lp-1 ] ERROR 2024/07/22 17:14:13 charge power: dial tcp 192.168.0.181:502: connect: no route to host
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:13 modbus: send 06 c3 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c3 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c4 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c4 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c4 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c4 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c4 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c4 00 00 00 07 f7 03 04 00 00 0a c1
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c5 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c5 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c5 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c5 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c5 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c5 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c6 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c6 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c6 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c6 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c6 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c6 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c7 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c7 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c7 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c7 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c7 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c7 00 00 00 07 f7 03 04 00 01 a8 94
[site ] DEBUG 2024/07/22 17:14:14 pv power: 2753W
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c8 00 00 00 06 f7 03 89 6e 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c8 00 00 00 06 f7 03 89 6e 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c8 00 00 00 06 f7 03 89 6e 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c8 00 00 00 06 f7 03 89 6e 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c8 00 00 00 06 f7 03 89 6e 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c8 00 00 00 07 f7 03 04 00 00 00 3b
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c9 00 00 00 06 f7 03 89 89 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c9 00 00 00 06 f7 03 89 89 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c9 00 00 00 06 f7 03 89 89 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c9 00 00 00 06 f7 03 89 89 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 c9 00 00 00 06 f7 03 89 89 00 02
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 c9 00 00 00 07 f7 03 04 00 00 68 7a
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 ca 00 00 00 06 f7 03 90 8f 00 01
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 ca 00 00 00 06 f7 03 90 8f 00 01
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 ca 00 00 00 06 f7 03 90 8f 00 01
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 ca 00 00 00 06 f7 03 90 8f 00 01
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 ca 00 00 00 06 f7 03 90 8f 00 01
[modbus] TRACE 2024/07/22 17:14:14 modbus: recv 06 ca 00 00 00 05 f7 03 02 00 64
[site ] DEBUG 2024/07/22 17:14:14 battery soc: 100%
[site ] DEBUG 2024/07/22 17:14:14 battery power: 59W
[modbus] TRACE 2024/07/22 17:14:14 modbus: send 06 cb 00 00 00 06 f7 03 8c b9 00 02
[... cut ...]
[delta ] TRACE 2024/07/22 17:16:09 modbus: recv 02 31 00 00 00 07 01 04 04 00 00 00 66
[lp-1 ] DEBUG 2024/07/22 17:16:09 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.1
[delta ] TRACE 2024/07/22 17:16:09 modbus: send 02 32 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2024/07/22 17:16:10 modbus: recv 02 32 00 00 00 03 01 84 02
[lp-1 ] ERROR 2024/07/22 17:16:10 charger status: modbus: exception '2' (illegal data address), function '4'
[mqtt ] TRACE 2024/07/22 17:16:13 recv evc/power: 'NaN' 

Till the relays clicked off, i saw the values in evcc.

LingLing-de, 2024-07-22:
@Klaus3d3 
what is your configuration on the Delta AC MAX Smart, can you send a screenshot so we are all on the same configuration?
I didn't even get the errors that you have in your log file, since my Car won't start charging.

Klaus3d3, 2024-07-22:
> @Klaus3d3 what is your configuration on the Delta AC MAX Smart, can you send a screenshot so we are all on the same configuration? I didn't even get the errors that you have in your log file, since my Car won't start charging.

@LingLing-de  Follow this Reset Procedure :https://github.com/evcc-io/evcc/discussions/13031#discussioncomment-10116066

The Box seems to get blocked pretty easy.

Klaus3d3, 2024-07-22:
> I got the [fix/delta branch](https://github.com/evcc-io/evcc/tree/fix/delta) installed - cup of coffee helped... Situation improved ( `charge rater` error is gone), but there's still something odd with getting the wallboxe's status (trace log below).

@madmat17 Could you please explain the trick?

premultiply, 2024-07-22:
We can not support the charger if it does not support any status.

From the HA config above (https://github.com/evcc-io/evcc/issues/14890#issue-2409464457) the remaining status register was marked as compatible but the vendor spec says not supported.

If the last is the case we have to drop the support completly - at least for some specific models / firmware version i am not really aware of yet.

premultiply, 2024-07-22:
@madmat17 @Klaus3d3 Your trace logs don't appear to be created by https://github.com/evcc-io/evcc/tree/fix/delta as register 1001 is still queried for status.

closed 2024-07-23 by premultiply

Klaus3d3, 2024-07-23:
> > `[delta ] TRACE 2024/07/22 17:16:09 modbus: send 02 32 00 00 00 06 01 04 03 e9 00 01`
> 
> @madmat17 @Klaus3d3 Your trace logs don't appear to be created by https://github.com/evcc-io/evcc/tree/fix/delta as register 1001 is still queried for status. This branch is using register 1000 for status.
> 
> Please run `evcc charger --log trace --diagnose` with the executable of this branch to verify.

As soon as i can figure out how to switch to fix/delta branch, i can do this. :-)

madmat17, 2024-07-23:
> @madmat17 @Klaus3d3 Your trace logs don't appear to be created by https://github.com/evcc-io/evcc/tree/fix/delta as register 1001 is still queried for status. This branch is using register 1000 for [...]

@premultiply You're totally right. I thought that HA would compile evcc on the fly during installation (I forked the repo and set the branch /fix/delta as my main branch) - which obviously is not the case. Hence the release binaries were executed.
I saw that you already merged the fix into `master` - thanks for that!
I was able to compile the latest version, but cannot test it right now with the Delta AC MAX Smart right now, as I don't have access to my home network atm, but will test it ASAP.

@Klaus3d3: You can compile your own binary following the following documentation https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md for testing it outside of HA.
I didn't try publishing my own docker image for testing it inside HA, but I assume that the Home Assistant Builder might be a good starting point:
https://github.com/home-assistant/builder

premultiply, 2024-07-23:
You can just use the latest nightly build now.
There is no need to run it in HA! Just run evcc on any other computer like Windows Laptop etc.

premultiply, 2024-07-24:
It's already in the nightly build.

Stefan135, 2024-07-27:
Hey folks,
I can also confirm that the current nightly build is working fine with Delta AC Max Smart WB with 
MPU Firmware version :  01.26.34.00 and MCU Firmware version :  01.00.36.00. Many thanks to @andig and @premultiply !

Unfortunately the Modbus Connection is being refused when using a Wifi Connection. However, that's a solvable problem :)

Best regards,
Stefan

Felsblick, 2024-07-28:
Hi all!

seems that this fix for the AC Max Smart is causing now issues with my AC Max Basic. I shortly tested evcc 0.129.0 but evcc was now always showing a connected vehicle (although nothing was plugged in) i am conneting through a waveshare Ethernet modbus Adapter as the AC Max Basic has no Ethernet Port and did not have and problems with the charging status before up to evcc 0.128.4. downgrade now to this Version (everything OK again…)

andig, 2024-07-28:
Please show

    evcc charger —log trace,db:error

Felsblick, 2024-07-28:
here is the trace log from UI ( running in as home assistant add on...)

switch on and off with the key at the wallbox disables the loading point, this is still working... but when key is switch on, lvcc is always showing an connected vehicle...

`[main ] INFO 2024/07/28 16:46:36 evcc 0.129.0
[main ] INFO 2024/07/28 16:46:36 using config file: /config/evcc.yaml
[db ] INFO 2024/07/28 16:46:36 using sqlite database: /data/evcc.db
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `settings` 18 <nil>
[mqtt ] INFO 2024/07/28 16:46:36 connecting evcc-1899380482 at tcp://localhost:1883
[mqtt ] DEBUG 2024/07/28 16:46:36 tcp://localhost:1883 connected
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT fk_configs_details %") -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT fk_devices_details %") -1 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "% device_id %") -1 <nil>
[main ] INFO 2024/07/28 16:46:36 listening at :7070
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[delta ] TRACE 2024/07/28 16:46:36 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/28 16:46:36 modbus: recv 01 03 02 00 00 b8 44
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `config_details` WHERE `config_details`.`config_id` = 3 9 <nil>
[db ] TRACE 2024/07/28 16:46:36 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 1 <nil>
[id ] TRACE 2024/07/28 16:46:36 GET https://emea.bff.cariad.digital/user-login/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&code_challenge=rLMonWl1v-zuyB6SzpYX7RaJRjXxpVTEFprKa7K_R54&code_challenge_method=S256&redirect_uri=weconnect%3A%2F%2Fauthenticated&response_type=code+id_token+token&scope=openid+profile+badge+cars+vin&nonce=zAyAkViATnKGiYRIpzGRilnAuERydmsmnZqSbWrHmab&state=d1698cbd-0020-4da7-91dd-8f7f4dcb92fc
[id ] TRACE 2024/07/28 16:46:36 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&scope=openid+profile+badge+cars+dealers+vin&response_type=code+id_token+token&redirect_uri=weconnect%3A%2F%2Fauthenticated&state=97d17c37b3b9bfbce5ca52c39a9f693887cb6e66faeb76b4d4b4e7cc77b3bf25&nonce=zAyAkViATnKGiYRIpzGRilnAuERydmsmnZqSbWrHmab
[id ] TRACE 2024/07/28 16:46:36 GET https://identity.vwgroup.io/signin-service/v1/signin/a24fba63-34b3-4d43-b181-942111e6bda8@apps_vw-dilab_com?relayState=023d21da63e55c87d3fee19b39cd7af42febeaea
[id ] TRACE 2024/07/28 16:46:36 <!DOCTYPE html> <html lang="en" xmlns="http://www.w3.org/1999/html"> <head> <meta charset="utf-8"/> <meta name="viewport" content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/> <meta name="identitykit" content="loginIdentifier"/> <script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"4602 [... cut ...]
[lp-1 ] DEBUG 2024/07/28 16:51:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/28 16:51:02 pv charge current: 13.2A = 0A + 13.2A (-3031W @ 1p)
[lp-1 ] DEBUG 2024/07/28 16:51:02 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[delta ] TRACE 2024/07/28 16:51:02 modbus: send 01 10 06 40 00 02 04 00 00 0b d7 9b 51
[delta ] TRACE 2024/07/28 16:51:02 modbus: recv 01 10 06 40 00 02 40 94
[lp-1 ] DEBUG 2024/07/28 16:51:02 max charge current: 13.2A`

andig, 2024-07-28:
That does not help :(

Felsblick, 2024-07-28:
What else can i do to provide helpfull hinzu?

andig, 2024-07-28:
Just run it locally. Theres no need for HA.

Felsblick, 2024-07-29:
Hi andig,

i have it now locally on MacOs, same issue, "Gastfahrzeug" is always shown as plugged in.
Here is the result of 

> evcc charger —log trace,db:error

> [main  ] INFO 2024/07/29 11:27:23 evcc 0.129.0
> Error: accepts at most 1 arg(s), received 2
> Usage:
>   evcc charger [name] [flags]
>   evcc charger [command]
> 
> Available Commands:
>   ramp        Ramp current from 6..16A in configurable steps
> 
> Flags:
>   -i, --current int   Set maximum current
>       --diagnose      Diagnose
>   -d, --disable       Disable
>   -e, --enable        Enable
>   -p, --phases int    Set usable phases (1 or 3)
>   -w, --wakeup        Wake up
> 
> Global Flags:
>   -c, --config string   Config file (default "~/evcc.yaml" or "/etc/evcc.yaml")
>   -h, --help            Help
>       --ignore-db       Run command ignoring service database
>   -l, --log string      Log level (fatal, error, warn, info, debug, trace) (default "info")
>       --log-headers     Log headers
> 
> Use "evcc charger [command] --help" for more information about a command.
> 
> accepts at most 1 arg(s), received 2

and log is showing:

> [delta ] TRACE 2024/07/29 11:16:24 modbus: send 01 03 00 c9 00 01 54 34
> [delta ] TRACE 2024/07/29 11:16:24 modbus: recv 01 03 02 00 00 b8 44
> [delta ] TRACE 2024/07/29 11:16:24 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:16:24 modbus: recv 01 03 04 00 00 10 2c f6 2e
> [delta ] TRACE 2024/07/29 11:16:24 modbus: send 01 10 06 40 00 02 04 00 00 10 2c d0 22
> [delta ] TRACE 2024/07/29 11:16:24 modbus: recv 01 10 06 40 00 02 40 94
> [delta ] TRACE 2024/07/29 11:16:24 modbus: send 01 04 03 ed 00 02 e1 ba
> [delta ] TRACE 2024/07/29 11:16:25 modbus: recv 01 04 04 00 00 00 00 fb 84
> [delta ] TRACE 2024/07/29 11:16:25 modbus: send 01 04 03 e8 00 01 b1 ba
> [delta ] TRACE 2024/07/29 11:16:25 modbus: recv 01 04 02 00 03 f9 31
> [delta ] TRACE 2024/07/29 11:16:25 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:16:25 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:16:25 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:16:25 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:16:25 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:16:25 modbus: recv 01 03 04 00 00 10 2c f6 2e
> [delta ] TRACE 2024/07/29 11:16:54 modbus: send 01 04 03 ed 00 02 e1 ba
> [delta ] TRACE 2024/07/29 11:16:55 modbus: recv 01 04 04 00 00 00 00 fb 84
> [delta ] TRACE 2024/07/29 11:16:55 modbus: send 01 04 03 e8 00 01 b1 ba
> [delta ] TRACE 2024/07/29 11:16:55 modbus: recv 01 04 02 00 03 f9 31
> [delta ] TRACE 2024/07/29 11:16:55 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:16:55 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:16:55 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:16:55 modbus: recv 01 03 04 00 00 10 2c f6 2e
> [delta ] TRACE 2024/07/29 11:16:55 modbus: send 01 10 06 40 00 02 04 00 00 10 43 90 0e
> [delta ] TRACE 2024/07/29 11:16:55 modbus: recv 01 10 06 40 00 02 40 94
> [delta ] TRACE 2024/07/29 11:17:24 modbus: send 01 04 03 ed 00 02 e1 ba
> [delta ] TRACE 2024/07/29 11:17:25 modbus: recv 01 04 04 00 00 00 00 fb 84
> [delta ] TRACE 2024/07/29 11:17:25 modbus: send 01 04 03 e8 00 01 b1 ba
> [delta ] TRACE 2024/07/29 11:17:25 modbus: recv 01 04 02 00 03 f9 31
> [delta ] TRACE 2024/07/29 11:17:25 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:17:25 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:17:25 modbus: send 01 03 06 40 00 02 c5 57
[... 204 lines cut ...]
> [delta ] TRACE 2024/07/29 11:27:25 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:27:25 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:27:25 modbus: recv 01 03 04 00 00 10 ba 76 40
> [delta ] TRACE 2024/07/29 11:27:25 modbus: send 01 10 06 40 00 02 04 00 00 10 eb 91 b0
> [delta ] TRACE 2024/07/29 11:27:25 modbus: recv 01 10 06 40 00 02 40 94
> [delta ] TRACE 2024/07/29 11:27:54 modbus: send 01 04 03 ed 00 02 e1 ba
> [delta ] TRACE 2024/07/29 11:27:55 modbus: recv 01 04 04 00 00 00 00 fb 84
> [delta ] TRACE 2024/07/29 11:27:55 modbus: send 01 04 03 e8 00 01 b1 ba
> [delta ] TRACE 2024/07/29 11:27:55 modbus: recv 01 04 02 00 03 f9 31
> [delta ] TRACE 2024/07/29 11:27:55 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:27:55 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:27:55 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:27:55 modbus: recv 01 03 04 00 00 10 eb b7 bc
> [delta ] TRACE 2024/07/29 11:27:55 modbus: send 01 10 06 40 00 02 04 00 00 10 e8 d1 b1
> [delta ] TRACE 2024/07/29 11:27:55 modbus: recv 01 10 06 40 00 02 40 94
[... cut ...]
> [delta ] TRACE 2024/07/29 11:28:25 modbus: send 01 04 04 4c 00 14 30 e2
> [delta ] TRACE 2024/07/29 11:28:25 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
> [delta ] TRACE 2024/07/29 11:28:25 modbus: send 01 03 06 40 00 02 c5 57
> [delta ] TRACE 2024/07/29 11:28:25 modbus: recv 01 03 04 00 00 10 e8 f7 bd
> [delta ] TRACE 2024/07/29 11:28:25 modbus: send 01 10 06 40 00 02 04 00 00 10 f8 d0 7d
> [delta ] TRACE 2024/07/29 11:28:25 modbus: recv 01 10 06 40 00 02 40 94

premultiply, 2024-07-29:
Und jetzt nochmal mit richtiger Syntax: `evcc charger --log trace,db:error`

Felsblick, 2024-07-29:
OK, macht Sinn ;) sorry für meine Laienhafte Art, hier der log:
```
[main  ] INFO 2024/07/29 13:02:05 evcc 0.129.0
[main  ] INFO 2024/07/29 13:02:05 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 04 02 00 03 f9 31
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 03 04 00 00 05 64 f8 88
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       true
Identifier:    <none>
```

Wenn ich das richtig interpretiere, sollte der charger status richtigerweise A sein und nicht B. Zumindest vor dem fix stand er auf A wenn kein Fahrzeug angeschlossen ist.

Gruß
Johannes

andig, 2024-07-29:
@premultiply einfach alle 0x00 aus dem String trimmen?

premultiply, 2024-07-29:
Ist doch schon gefixt

Felsblick, 2024-07-29:
@premultiply aber noch nicht in der Version 0.129.0 oder? Zumindest habe ich da das Problem… lasst mich wissen wenn ich was testen soll.

premultiply, 2024-07-29:
Sorry, da bin ich wohl in den Threads durcheinander gekommen.

premultiply, 2024-07-29:
```
[delta ] TRACE 2024/07/29 13:02:06 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 13:02:06 modbus: recv 01 04 02 00 03 f9 31
```

Hier wird Register 1000 (0x03e8) für den Status abgefragt.
Dort steht 0x0003 also 3 = 'Preparing' drin. Das wird normalerweise auf Status B gemappt.

> Wenn ich das richtig interpretiere, sollte der charger status richtigerweise A sein und nicht B. Zumindest vor dem fix stand er auf A wenn kein Fahrzeug angeschlossen ist.

Was heisst "kein Fahrzeug angeschlossen" hier genau? Kabel steckt in der Wallbox mit Dose und nur das Ende zum Fahrzeug ist nicht angeschlossen?

Mache mal bitte einen Vergleich in deinem Fall für

1) Fahrzeug angeschlossen (aber keine Ladefreigabe)
2) Fahrzeug nicht angeschlossen

Jeweils bitte `evcc charger --log trace` dazu ausführen und und den Registerinhalt dazu ermitteln und mitteilen.

Felsblick, 2024-07-29:
Hallo @premultiply 
Kann ich gerne später noch machen. Aber ich glaube da haben wir den Fehler schon.
Das Register 1000 ist der Status der Wallbox selber, nicht der Ladestatus. Im Register 1001 ist der Ladestatus zu finden.
In Version 0.128.4 wurde auch noch das Register 1001 (0x03e9) abgefragt, und da hat es einwandfrei funktioniert:

```
[delta ] TRACE 2024/07/29 13:05:38 modbus: send 01 04 **03 e9** 00 01 e0 7a
[delta ] TRACE 2024/07/29 13:05:38 modbus: recv 01 04 02 00 00 b9 30
[lp-1 ] DEBUG 2024/07/29 13:05:38 charger status: A
```
und dort steht korrekterweise 0x0000 drin (0: Charging process not started (no vehicle connected))

Infos aus: (Seite 8 bzw 9)
[EVCS ModBus Register Table V1.1 working (1).pdf](https://github.com/user-attachments/files/16416497/EVCS.ModBus.Register.Table.V1.1.working.1.pdf)

premultiply, 2024-07-29:
1001 kann nicht genutzt werden, da es das nicht bei allen Delta-Boxen gibt. Siehe oben!!!
Daher wurde dies durch Register 1000 ersetzt. Darum geht es eigentlich in diesem Issue!

Dahr brauche ich jetzt von dir die angefragten Aussagen https://github.com/evcc-io/evcc/issues/14890#issuecomment-2256386724 zum steckenden Ladekabel(?) und der jeweilige Zustand des **Registers 1000**.

Felsblick, 2024-07-29:
OK, sorry, das hatte ich nicht komplett nachvollzogen. anbei die Logs.

Meiner Meinung nach gäbe es zwei Möglichkeiten:
Status 1, 2 und 3 auf Charger Status "A" Mappen, sollte wieder den alten Zustand herstellen.
Oder eine Unterscheidung zwischen der AX Max Smart und der AC Max basic einführen, für die Basic müsste dann nach wie vor das Register 1001 ausgewertet werden

Kabel abgesteckt (Kein Kabel in Wallbox)
Register 1000 meldet 0x00 03 --> "preparing"
```
[main  ] INFO 2024/07/29 19:29:50 evcc 0.129.0
[main  ] INFO 2024/07/29 19:29:50 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 19:29:51 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 19:29:51 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 19:29:51 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 19:29:51 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/07/29 19:29:51 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 19:29:51 modbus: recv 01 04 02 00 03 f9 31
[delta ] TRACE 2024/07/29 19:29:51 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 19:29:51 modbus: recv 01 03 04 00 00 00 00 fa 33
[delta ] TRACE 2024/07/29 19:29:51 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 19:29:51 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       false
Identifier:    <none>

```

Kabel eingesteckt nur in Wallbox
Register 1000 meldet 0x00 03 --> "preparing"
```
[main  ] INFO 2024/07/29 19:31:00 evcc 0.129.0
[main  ] INFO 2024/07/29 19:31:00 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 19:31:00 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 19:31:00 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 19:31:00 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 19:31:00 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/07/29 19:31:00 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 19:31:00 modbus: recv 01 04 02 00 03 f9 31
[delta ] TRACE 2024/07/29 19:31:00 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 19:31:00 modbus: recv 01 03 04 00 00 00 00 fa 33
[delta ] TRACE 2024/07/29 19:31:00 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 19:31:00 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       false
Identifier:    <none>

```
Kabel in WB und Fahrzeug eingesteckt, Laden AUS über evcc
Register 1000 meldet 0x00 06 --> "Suspended EV"
```
[main  ] INFO 2024/07/29 19:32:02 evcc 0.129.0
[main  ] INFO 2024/07/29 19:32:02 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 19:32:02 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 19:32:02 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 19:32:02 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 19:32:03 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/07/29 19:32:03 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 19:32:03 modbus: recv 01 04 02 00 06 39 32
[delta ] TRACE 2024/07/29 19:32:03 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 19:32:03 modbus: recv 01 03 04 00 00 00 00 fa 33
[delta ] TRACE 2024/07/29 19:32:03 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 19:32:03 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       false
Identifier:    <none>
```

Kabel in WB in Fzg, Laden ein
Register 1000 meldet 0x00 04 --> "Charging"

```
[main  ] INFO 2024/07/29 19:33:21 evcc 0.129.0
[main  ] INFO 2024/07/29 19:33:21 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 19:33:21 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 19:33:21 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 19:33:21 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 19:33:21 modbus: recv 01 04 04 00 00 1b f4 f0 f3
[delta ] TRACE 2024/07/29 19:33:21 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 19:33:21 modbus: recv 01 04 02 00 04 b8 f3
[delta ] TRACE 2024/07/29 19:33:21 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 19:33:21 modbus: recv 01 03 04 00 00 1c c0 f2 a3
[delta ] TRACE 2024/07/29 19:33:21 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 19:33:21 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         7156W
Charge status: C
Enabled:       true
Identifier:    <none>
```

Kabel in WB und Fog, Laden über das Fzg gestoppt:
Register 1000 meldet 0x00 06 --> "Suspended EV"
```
[main  ] INFO 2024/07/29 19:34:14 evcc 0.129.0
[main  ] INFO 2024/07/29 19:34:14 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/07/29 19:34:14 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/07/29 19:34:14 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/07/29 19:34:14 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/07/29 19:34:14 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/07/29 19:34:14 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/07/29 19:34:14 modbus: recv 01 04 02 00 06 39 32
[delta ] TRACE 2024/07/29 19:34:14 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/07/29 19:34:14 modbus: recv 01 03 04 00 00 1c c0 f2 a3
[delta ] TRACE 2024/07/29 19:34:14 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/07/29 19:34:14 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       true
Identifier:    <none>
```

Felsblick, 2024-07-29:
Danke! Super, dass du das so schnell geändert hast. Wie kann ich den fix auf mein Testsystem aufspielen?

Felsblick, 2024-08-19:
@premultiply 
mit dem Release von 0.130.0 konnte ich es auf dem Livesystem testen. Funktionalität ist nicht ganz wieder da. Was mir leider erst jetzt aufgefallen ist: Nach dem Laden gibt des Register 1000 den Wert 9 aus (was eigentlich "Faulted" heißt).
Der Status 9 ist nicht gemappt, daher bleibt der Charger Status auf laden. Ein Wiederstart des Ladens ist dann nicht möglich, nur durch Aus- und Einstecken des Ladekabels.
Es muss aber einen Grund geben, warum die WB auf Fehler geht, den konnte ich aber noch nicht finden. Auf jeden Fall ist es reproduzierbar.

hier der Ausschnitt nach dem evcc das Laden stoppt.

```
[delta ] TRACE 2024/08/19 18:25:17 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/19 18:25:17 modbus: recv 01 04 02 00 04 b8 f3
[delta ] TRACE 2024/08/19 18:25:17 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/08/19 18:25:17 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
[delta ] TRACE 2024/08/19 18:25:17 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/08/19 18:25:17 modbus: recv 01 03 04 00 00 05 64 f8 88
[delta ] TRACE 2024/08/19 18:25:17 modbus: send 01 10 06 40 00 02 04 00 00 00 00 dc 3f
[delta ] TRACE 2024/08/19 18:25:17 modbus: recv 01 10 06 40 00 02 40 94
[delta ] TRACE 2024/08/19 18:25:27 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/08/19 18:25:27 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/08/19 18:25:27 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/19 18:25:27 modbus: recv 01 04 02 00 09 79 36
```

Traces aus dem Testsystem:
Laden:
```
[main  ] INFO 2024/08/19 18:57:02 evcc 0.130.0
[main  ] INFO 2024/08/19 18:57:02 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/08/19 18:57:02 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/08/19 18:57:02 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/08/19 18:57:02 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/08/19 18:57:02 modbus: recv 01 04 04 00 00 02 65 3a cf
[delta ] TRACE 2024/08/19 18:57:02 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/19 18:57:02 modbus: recv 01 04 02 00 04 b8 f3
[delta ] TRACE 2024/08/19 18:57:02 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/08/19 18:57:02 modbus: recv 01 03 04 00 00 05 64 f8 88
[delta ] TRACE 2024/08/19 18:57:02 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/08/19 18:57:02 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         613W
Charge status: C
Enabled:       true
Identifier:    <none>
```

Laden gestoppt (über evcc auf aus)
```
[main  ] INFO 2024/08/19 18:58:21 evcc 0.130.0
[main  ] INFO 2024/08/19 18:58:21 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/08/19 18:58:21 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/08/19 18:58:21 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/08/19 18:58:21 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/08/19 18:58:21 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/08/19 18:58:21 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/19 18:58:21 modbus: recv 01 04 02 00 09 79 36
[delta ] TRACE 2024/08/19 18:58:21 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/08/19 18:58:21 modbus: recv 01 03 04 00 00 00 00 fa 33
[delta ] TRACE 2024/08/19 18:58:21 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/08/19 18:58:21 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: invalid status: 9
Enabled:       false
Identifier:    <none>
```

premultiply, 2024-08-19:
https://github.com/evcc-io/evcc/commit/da91ea630318eae7e3569e6895963f50cd69cf39

Felsblick, 2024-08-19:
2, 3 mal getestet - scheint jetzt zu funktionieren! Es kommen keine Fehler mehr.
Danke!!
```[main  ] INFO 2024/08/19 22:54:28 evcc 0.130.1
[main  ] INFO 2024/08/19 22:54:28 using config file: /etc/evcc.yaml
[delta ] TRACE 2024/08/19 22:54:28 modbus: send 01 03 00 c9 00 01 54 34
[delta ] TRACE 2024/08/19 22:54:28 modbus: recv 01 03 02 00 00 b8 44
[delta ] TRACE 2024/08/19 22:54:28 modbus: send 01 04 03 ed 00 02 e1 ba
[delta ] TRACE 2024/08/19 22:54:28 modbus: recv 01 04 04 00 00 00 00 fb 84
[delta ] TRACE 2024/08/19 22:54:28 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/19 22:54:28 modbus: recv 01 04 02 00 09 79 36
[delta ] TRACE 2024/08/19 22:54:28 modbus: send 01 03 06 40 00 02 c5 57
[delta ] TRACE 2024/08/19 22:54:28 modbus: recv 01 03 04 00 00 00 00 fa 33
[delta ] TRACE 2024/08/19 22:54:28 modbus: send 01 04 04 4c 00 14 30 e2
[delta ] TRACE 2024/08/19 22:54:28 modbus: recv 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 95 56
Power:         0W
Charge status: B
Enabled:       false
Identifier:    <none>
```

PR #14970 (2024-07-23): Delta: fix register usage
Fixes #14890