evcc-io/evcc #14891: panic: runtime error: invalid memory address or nil pointer dereference

oberraeuber, 2024-07-15:
### Describe the bug

Seit ca. gestern (14.7.24) Nachmittag (ca. 15:00 Uhr) hat sich eine laufende Instanz beim Laden "verabschiedet". Neustart im debug Log-Modus produzierte das im Anhang. Seither mit verschiedenen Versionen versucht. Eine ältere Version 0.126.2 von der ich noch einen Container hatte, verhält sich etwas anders, aber schlussendlich mit dem gleichen Ergebnis, dass der Container beendet wird. Was anders ist, ist dass das mehrere Minuten dauert. Bei der 0.128.2 nur wenige Sekunden. 

### Steps to reproduce

1. Container im Docker mit der latest (0.128.2) starten. 
2. Container mit einer alten Config (0.126.2) starten
3. Container mit einer nightly (0.128.2 (a0bbd4a3)) starten 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
#  cache: error
#  site: debug
#  lp-1: debug
  

# unique installation id
plant: 36f22147c343e601605474dc03d040e0dc522a76b628e08434ea9088af74b3e8

interval: 30s # control cycle interval

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.74  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  endianness: little
  minsoc: 5  
  maxsoc: 80  
  watchdog: 60s  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.74  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  minsoc: 5  
  maxsoc: 80  
  watchdog: 60s 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.75  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: ellipro 
  ski: 618A BFE2 24CC 100A 6AC4 F8F1 E421 2DD1 E836 D073  
  name: wallbox5
[... 454 lines cut ...]
          *****
          -----END CERTIFICATE-----
[main  ] FATAL 2024/07/15 20:55:01 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'eebus': i/o timeout
[cache ] TRACE 2024/07/15 20:55:01 fatal: [cannot create charger 'wallbox5' cannot create charger type 'template' cannot create charger type 'eebus' i/o timeout]
[main  ] FATAL 2024/07/15 20:55:01 will attempt restart in: 5m0s
[eebus ] DEBUG 2024/07/15 20:56:02 trying to connect to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 at wallbox-2117A1BR9T.local.
[eebus ] DEBUG 2024/07/15 20:56:02 initiating connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 at wallbox-2117A1BR9T.local.:4712
[eebus ] DEBUG 2024/07/15 20:56:02 connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 failed: dial tcp: lookup wallbox-2117A1BR9T.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/07/15 20:56:02 restarting mdns search
[eebus ] DEBUG 2024/07/15 20:56:02 delaying connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 by 3m23.57s to minimize double connection probability
[eebus ] DEBUG 2024/07/15 20:59:25 trying to connect to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 at wallbox-2117A1BR9T.local.
[eebus ] DEBUG 2024/07/15 20:59:25 initiating connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 at wallbox-2117A1BR9T.local.:4712
[eebus ] DEBUG 2024/07/15 20:59:25 connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 failed: dial tcp: lookup wallbox-2117A1BR9T.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/07/15 20:59:25 restarting mdns search
[eebus ] DEBUG 2024/07/15 20:59:25 delaying connection to 618abfe224cc100a6ac4f8f1e4212dd1e836d073 by 5m3.075s to minimize double connection probability
[eebus ] DEBUG 2024/07/15 21:00:01 mdns: stop announcement
```

### What type of operating system are you running?

Docker container

### Version

0.128.2

andig, 2024-07-15:
/cc @DerAndereAndi

DerAndereAndi, 2024-07-16:
Die panic kann ich lösen, wird aber EEBUS nicht zum Laufen bekommen. Denn es gibt davor gibt es folgendes Problem:

```
[eebus ] DEBUG 2024/07/15 20:49:12 mdns: failure announcing service could not determine host IP addresses
[eebus ] DEBUG 2024/07/15 20:49:12 error during mdns setup: could not determine host IP addresses
```

EEBUS funktioniert hier nicht, es konnte kein mDNS Announcement gemacht werden. Das scheint ein Problem mit diesem Container Setup zu sein, der Code konnte weder eine IPv4 noch eine IPv6 Adresse finden. Habe aber momentan keine Idee was das sein könnte und wie man das beheben könnte.

Bei Version 0.126.2 ist das Problem etwas leicht anderes: der Code konnte keine Verbindung zum lokalen IP Netz haben.

Hypothese: Kann es sein dass der Container NICHT im Host Modus läuft? Das ist aber unbedingt notwendig! Bitte damit mal versuchen.

DerAndereAndi, 2024-07-16:
The panic is fixed via https://github.com/evcc-io/evcc/pull/14900

closed 2024-07-16 by andig

andig, 2024-07-17:
@DerAndereAndi warum gibts egtl. die initiating und connecting Meldungen doppelt? Kann davon evtl. eine entfallen?

DerAndereAndi, 2024-07-17:
Da passiert noch einiges dazwischen, also nein, ist keine Doppelung. Und die zweite kommt nur im Problemfall.