evcc-io/evcc #14892: Seat/Cupra funktioniert nicht mehr

blenni, 2024-07-15:
### Describe the bug

Die Verbindung zu Seat/Cupra funktioniert nicht mehr.
In einer aktiven evcc-Instanz läuft es noch, weil scheinbar das Token noch gültig ist.
Bei einem neuen Aufruf von `evcc vehicle` Aufruf oder in einer anderen, neu gestarteten Instanz läuft es nicht mehr.

Neue Nutzungsbedingungen in der App wurden schon akzeptiert.

### Steps to reproduce

1. Seat/Cupra konfigurieren
2. evcc vehicle ausführen

### Configuration details

```yaml
- type: template
  template: seat
  title: Cupra Leon
  user: xxx
  password: xxx
  vin: xxx
  capacity: 11
  phases: 1
  icon: car
  cache: 15m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  name: ev_cupra
```

### Log details

```text
[main  ] INFO 2024/07/15 22:19:01 evcc 0.128.2
[seat  ] TRACE 2024/07/15 22:19:02 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=30e33736-c537-4c72-ab60-74a7b92cfe83%40apps_vw-dilab_com&code_challenge=zlhrrjFy3BUwXPVjeyhylSxRoqwyWlQTLcXZtZLTR2M&code_challenge_method=S256&redirect_uri=cupraconnect%3A%2F%2Fidentity-kit%2Flogin&response_type=code+id_token&scope=openid+profile+mbb&nonce=BcQXlJpysCNAGbqCNtwFEAWfZetaEyvAOiAJtKgVRsJ&state=85303692-0816-448c-8c64-884f513dc53b
[seat  ] TRACE 2024/07/15 22:19:02 GET https://identity.vwgroup.io/signin-service/v1/signin/30e33736-c537-4c72-ab60-74a7b92cfe83@apps_vw-dilab_com?relayState=7e9acfb0a93c8c6d6b415d9c715ccd5a0a505efb
[seat  ] TRACE 2024/07/15 22:19:02 <!DOCTYPE html>
    <html lang="en" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>

<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;/*! For license information please see nr-loader-spa-1.262.0.min.js.LICENSE.txt */
(()=>{var e,t,r={2983:(e,t,r)=>{"use strict";r.d(t,{D0:()=>v,gD:()=>y,Vp:()=>s,oC:()=>R,fr:()=>_,jD:()=>C,hR:()=>T,xN:()=>b,x1:()=>c,aN:()=>A,V:()=>j});var n=r(384),i=r(7864);const o={beacon:n.NT.beacon,errorBeacon:n.NT.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},a={};function s(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!a[e])throw new Error("Info for ".concat(e," was never set"));return a[e]}function c(e,t){if(!e)throw new Error("All info objects require an agent identifier!");a[e]=(0,i.a)(t,o);const r=(0,n.nY)(e);r&&(r.info=a[e])}var u=r(993);const d=e=>{if(!e||"string"!=typeof e)return!1;try{document.createDocumentFragment().querySelector(e)}catch{return!1}return!0};var l=r(2614),f=r(944);const h="[data-nr-mask]",g=()=>{const e={mask_selector:"*",block_selector:"[data-nr-block]",mask_input_options:{color:!1,date:!1,"datetime-local":!1,email:!1,month:!1,number:!1,range:!1,search:!1,tel:!1,text:!1,time:!1,url:!1,week:!1,textarea:!1,select:!1,password:!0}};return{ajax:{deny_list:void 0,block_internal:!0,enabled:!0,harvestTimeSeconds:10,autoStart:!0},distributed_tracing:{enabled:void 0,exclude_newrelic_header:void 0,cors_use_newrelic_header:void 0,cors_use_tracecontext_headers:void 0,allowed_origins:void 0},feature_flags:[],harvest:{tooManyRequestsDelay:60},jserrors:{enabled:!0,harvestTimeSeconds:10,autoStart:!0},logging:{enabled:!0,harvestTimeSeconds:10,autoStart:!0,level:u.p_.INFO},metrics:{enabled:!0,autoStart:!0},obfuscate:void 0,page_action:{enabled:!0,harvestTimeSeconds:30,autoStart:!0},page_view_event:{enabled:!0,autoStart:!0},page_view_timing:{enabled:!0,harvestTimeSeconds:30,long_task:!1,autoStart:!0},privacy:{cookies_enabled:!0},proxy:{assets:void 0,beacon:void 0},session:{expiresMs:l.wk,inactiveMs:l.BB},session_replay:{autoStart:!0,enabled:!1,harvestTimeSeconds:60,preload:!1,sampling_rate:10,error_sampling_rate:100,collect_fonts:!1,inline_images:!1,inline_stylesheet:!0,mask_all_inputs:!0,get mask_text_selector(){return e.mask_selector},set mask_text_selector(t){d(t)?e.mask_selector="".concat(t,",").concat(h):""===t||null===t?e.mask_selector=h:(0,f.R)("An invalid session_replay.mask_selector was provided. '*' will be used.",t)},get block_class(){return"nr-block"},get ignore_class(){return"nr-ignore"},get mask_text_class(){return"nr-mask"},get block_selector(){return e.block_selector},set block_selector(t){d(t)?e.block_selector+=",".concat(t):""!==t&&(0,f.R)("An invalid session_replay.block_selector was provided and will not be used",t)},get mask_input_options(){return e.mask_input_options},set mask_input_options(t){t&&"object"==typeof t?e.mask_input_options={...t,password:!0}:(0,f.R)("An invalid session_replay.mask_input_option was provided and will not be used",t)}},session_trace:{enabled:!0,harvestTimeSeconds:10,autoStart:!0},soft_navigations:{enabled:!0,harvestTimeSeconds:10,autoStart:!0},spa:{enabled:!0,harvestTimeSeconds:10,autoStart:!0},ssl:void 0}},p={},m="All configuration objects require an agent identifier!";function v(e){if(!e)throw new Error(m);if(!p[e])throw new Error("Configuration for ".concat(e," was never set"));return p[e]}function b(e,t){if(!e)throw new Error(m);p[e]=(0,i.a)(t,g());const r=(0,n.nY)(e);r&&(r.init=p[e])}function y(e,t){if(!e)throw new Error(m);var r=v(e);if(r){for(var n=t.split("."),i=0;i<n.length-1;i++)if("object"!=typeof(r=r[n[i]]))return;r=r[n[n.length-1]]}return r}const w={accountID:void 0,trustKey:void 0,agentID:void 0,licenseKey:void 0,applicationID:void 0,xpid:void 0},x={};function R(e){if(!e)throw new Error("All loader-config objects require an agent identifier!");if(!x[e])throw new Error("LoaderConfig for ".concat(e," was never set"));return x[e]}function A(e,t){if(!e)throw new Error("All loader-config objects require an agent identifier!");x[e]=(0,i.a)(t,w);const r=(0,n.nY)(e);r&&(r.loader_config=x[e])}const T=(0,n.dV)().o;var E=r(6154),N=r(9324);const S={buildEnv:N.F3,distMethod:N.Xs,version:N.xv,originTime:E.WN},O={customTransaction:void 0,disabled:!1,isolatedBacklog:!1,loaderType:void 0,maxBytes:3e4,onerror:void 0,origin:""+E.gm.location,ptid:void 0,releaseIds:{},appMetadata:{},session:void 0,denyList:void 0,harvestCount:0,timeKeeper:void 0},I={};function _(e){if(!e)throw new Error("All runtime objects require an agent identifier!");if(!I[e])throw new Error("Runtime for ".concat(e," was never set"));return I[e]}function j(e,t){if(!e)throw new Error("All runtime objects require an agent identifier!");I[e]={...(0,i.a)(t,O),...S};const r=(0,n.nY)(e);r&&(r.runtime=I[e])}function C(e){return function(e){try{const t=s(e);return!!t.licenseKey&&!!t.errorBeacon&&!!t.applicationID}catch(e){return!1}}(e)}},7864:(e,t,r)=>{"use strict";r.d(t,{a:()=>i});var n=r(944);function i(e,t){try{if(!e||"object"!=typeof e)return(0,n.R)("Setting a Configurable requires an object as input");if(!t||"object"!=typeof t)return(0,n.R)("Setting a Configurable requires a model to set its initial properties");const r=Object.create(Object.getPrototypeOf(t),Object.getOwnPropertyDescriptors(t)),o=0===Object.keys(r).length?e:r;for(let a in o)if(void 0!==e[a])try{if(null===e[a]){r[a]=null;continue}Array.isArray(e[a])&&Array.isArray(t[a])?r[a]=Array.from(new Set([...e[a],...t[a]])):"object"==typeof e[a]&&"object"==typeof t[a]?r[a]=i(e[a],t[a]):r[a]=e[a]}catch(e){(0,n.R)("An error occurred while setting a property of a Configurable",e)}return r}catch(e){(0,n.R)("An error occured while setting a Configurable",e)}}},9324:(e,t,r)=>{"use strict";r.d(t,{F3:()=>i,Xs:()=>o,Yq:()=>a,xv:()=>n});const n="1.262.0",i="PROD",o="CDN",a="2.0.0-alpha.12"},6154:(e,t,r)=>{"use strict";r.d(t,{A4:()=>s,OF:()=>d,RI:()=>i,Vr:()=>g,WN:()=>p,bv:()=>o,gm:()=>a,lR:()=>f,lT:()=>h,m:()=>u,mw:()=>c,sb:()=>l});var n=r(1863);const i="undefined"!=typeof window&&!!window.document,o="undefined"!=typeof WorkerGlobalScope&&("undefined"!=typeof self&&self instanceof WorkerGlobalScope&&self.navigator instanceof WorkerNavigator||"undefined"!=typeof globalThis&&globalThis instanceof WorkerGlobalScope&&globalThis.navigator instanceof WorkerNavigator),a=i?window:"undefined"!=typeof Wor [... cut ...] Xv:()=>a,gX:()=>i,iW:()=>o});var n=[];function i(e)
[main  ] ERROR 2024/07/15 22:19:03 creating vehicle ev_cupra failed: cannot create vehicle type 'template': cannot create vehicle type 'seat': missing id_token

ev_cupra
--------
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'seat': missing id_token
Capacity: 11.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.2

andig, 2024-07-15:
Accept new Tos

blenni, 2024-07-15:
Hab ich schon, steht auch oben:
_Neue Nutzungsbedingungen in der App wurden schon akzeptiert._

blenni, 2024-07-15:
Sorry, habe gestern die Nutzungsbedingungen akzeptiert, dann gings trotzdem nicht.
Gerade nochmal geschaut, jetzt musste ich wieder akzeptieren, jetzt gehts auch wieder...

closed 2024-07-15 by blenni