evcc-io/evcc #14899: Add automatic 1phase/3phase switch in generic OCPP API

abaginski, 2024-07-16:
Hi guys, my mobile charger is connected via generic OCPP API and is not able to automatically switch between 1 phase and 3 phase charging.

This results in either using too much power (when sun is low in the case of fixed 3 phase charge) or too less power (when sun is high in case of fixed 1 phase charging).

What are the limitations for implementing such a functionality into the generic OCPP API?

abaginski, 2024-07-18:
Yes I am using the nightly build and here is the debug Log when switching from 1p to 3p during charging.

You can also see in the screenshot that I have allegedly switched to 1p but it is still using 4,1kwh.
[image]
 

`Jul 18 08:21:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:01 phase scale1p in 16s
Jul 18 08:21:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:01 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jul 18 08:21:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:01 pv charge current: min 6A > 3.78A (1529W @ 3p, battery: false)
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 ----
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 charge power: 4109W
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 pv power: 2824W
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 battery soc: 0%
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 battery power: 0W
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 grid meter: 1545W
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 grid currents: [2.93 3.11 3.46]A
Jul 18 08:21:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:21:31 site power: 1545W
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 !! session: chargeRater.chargedEnergy=0.3 - chargedAtStartup=0.0
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 charge total import: 112.827kWh
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 charger status: C
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 available power 2564W < 4140W min 3p threshold
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 phase scale1p in 0s
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 switched phases: 1p
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 phase timer inactive
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
Jul 18 08:21:31 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:21:31 pv charge current: min 6A > 0A (1545W @ 1p, battery: false)
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 ----
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 charge power: 4109W
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 pv power: 2830W
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 battery soc: 0%
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 battery power: 0W
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 grid meter: 1534W
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 grid currents: [2.92 3.1 3.45]A
Jul 18 08:22:01 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:01 site power: 1534W
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 !! session: chargeRater.chargedEnergy=0.3 - chargedAtStartup=0.0
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 charge total import: 112.827kWh
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 charger status: C
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
Jul 18 08:22:01 evcc evcc[11979]: [lp-1  ] DEBUG 2024/07/18 08:22:01 pv charge current: min 6A > 0A (1534W @ 1p, battery: false)
Jul 18 08:22:31 evcc evcc[11979]: [site  ] DEBUG 2024/07/18 08:22:31 ----`

andig, 2024-07-18:
/cc @premultiply if this still doesn't work it seems that the charger will only switch phases if the session is stopped. Seems this is not something we could do?

abaginski, 2024-07-18:
have a look at, maybe it is indeed solvable: According to the mobile charger developers there is a way to fix that: "It is necessary to set the OCPP ChargingProfile, within which the ChargingSchedule specifies the numberPhases argument to define the number of phases used. "

andig, 2024-07-18:
Please show ocpp trace log. This should already be the case.

abaginski, 2024-07-18:
[tracelogs.txt](https://github.com/user-attachments/files/16279435/tracelogs.txt)

Attached are the full trace logs

andig, 2024-07-18:
evcc is doing that:

```
[ocpp  ] TRACE 2024/07/18 09:33:43 send 9EX-ANV-AWA: [2,"1992951973","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6,"numberPhases":3}]}}}]
[ocpp  ] TRACE 2024/07/18 09:33:43 recv 9EX-ANV-AWA: [3,"1992951973",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/18 09:33:53 send 9EX-ANV-AWA: [2,"2101812968","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6,"numberPhases":1}]}}}]
[ocpp  ] TRACE 2024/07/18 09:33:53 recv 9EX-ANV-AWA: [3,"2101812968",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/18 09:34:01 recv 9EX-ANV-AWA: [2,"1001764","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-18T07:34:00.371Z","sampledValue":[{"value":"117589.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"4084.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
[ocpp  ] TRACE 2024/07/18 09:34:01 send 9EX-ANV-AWA: [3,"1001764",{}]
[ocpp  ] TRACE 2024/07/18 09:34:28 recv 9EX-ANV-AWA: [2,"1001765","Heartbeat",{}]
[ocpp  ] TRACE 2024/07/18 09:34:28 send 9EX-ANV-AWA: [3,"1001765",{"currentTime":"2024-07-18T07:34:28Z"}]
[ocpp  ] TRACE 2024/07/18 09:34:29 send 9EX-ANV-AWA: [2,"2673932340","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":3,"numberPhases":1}]}}}]
[ocpp  ] TRACE 2024/07/18 09:34:29 recv 9EX-ANV-AWA: [3,"2673932340",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/18 09:34:52 send 9EX-ANV-AWA: [2,"858371244","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":3,"numberPhases":3}]}}}]
[ocpp  ] TRACE 2024/07/18 09:34:52 recv 9EX-ANV-AWA: [3,"858371244",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/18 09:35:01 recv 9EX-ANV-AWA: [2,"1001766","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-07-18T07:35:01.266Z","sampledValue":[{"value":"117655.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"4084.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
[ocpp  ] TRACE 2024/07/18 09:35:01 send 9EX-ANV-AWA: [3,"1001766",{}]
```

premultiply, 2024-07-18:
Bitte auch mal mit https://github.com/evcc-io/evcc/pull/14888 bzw. https://github.com/evcc-io/evcc/tree/feature/ocpp-defaultprofile testen. Dort wird `TxDefaultProfile` statt `TxProfile` verwendet.

Ich habe leider kein Testsystem was sich für `numberPhases` interessiert.

abaginski, 2024-07-25:
@premultiply wie kann ich das testen? Oder wann werden diese Changes in den Master gemergt?

abaginski, 2024-07-29:
Nein, auch mit diesen Changes klappt das noch nicht. Also liegt das wohl an der Wallbox oder?

[image]

premultiply, 2024-07-29:
Ohne OCPP-Tracelog nicht zu beantworten.

closed 2024-08-10 by github-actions[bot]