evcc-io/evcc #14906: warp 3: charger logic error: current mismatch

StevieC121176, 2024-07-17:
### Describe the bug

seit neuestem bekomme beim Laden immer `charger logic error: current mismatch`
Konfiguration wurde nicht geändert und die Warp3 zeigt auch keine Fehler an.

### Steps to reproduce

1. Laden starten
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: tinkerforge-warp3
  host: 192.168.178.77
  port: 1883
  user: ****
  password: *****
  topic: warp3/2avk
  timeout: 30s
  name: wallbox1
```

### Log details

```text
sh. Anhang, Trace logs komplett und von warp3
```

### What type of operating system are you running?

Linux

### Version

0.128.2

StevieC121176, 2024-07-17:
hier noch ein screenshot:

[image]

andig, 2024-07-17:
Ich kann den Fehler im Log nicht sehen.

closed 2024-07-17 by andig

VolkerK62, 2024-07-17:
```
[lp-1  ] DEBUG 2024/07/17 08:56:57 pv charge current: 13.1A = 16A + -2.87A (660W @ 1p)
[warp  ] TRACE 2024/07/17 08:56:57 send warp3/2avk/evse/external_current_update: '{ "current": 13130 }'
[lp-1  ] DEBUG 2024/07/17 08:56:57 max charge current: 13.1A

[lp-1  ] DEBUG 2024/07/17 08:57:01 charge currents: [15.8 0 0.01]A

[lp-1  ] WARN 2024/07/17 08:57:03 charger logic error: current mismatch (got 15.8A measured, expected 13.1A)
```

@StevieC121176  hast du ein Intervall von 3s eingestellt?

StevieC121176, 2024-07-17:
@VolkerK62  ja, ich habe 3s eingestellt. Das hab ich aber schon immer und hat bisher ohne Probleme funktioniert.

VolkerK62, 2024-07-18:
Diese Warnung kommt, wenn der tatsächliche Ladestrom um mehr als 1A vom freigegebenen abweicht. (Das ist ziemlich neu).

premultiply, 2024-07-18:
...und tritt u. a. auch vor allem dann auf wenn das `interval` viel zu kurz für die beteiligten Komponenten ist, da deren Reaktionszeiten und Dämpfungsparameter trotz deutlicher Warnung ignoriert werden.

`interval` darf man nur so klein wählen wie sichergestellt ist, dass das Gesamtsystem nach einer Stellgrößenveränderung seitens evcc vollständig eingeschwungen ist!