evcc-io/evcc #14907: Wrong End-of-line in entrypoint.sh (CR/LF->LF)

cbshopper, 2024-07-17:
### Describe the bug

I tried out the build process of the docker image. When I build the image an installed it in portainer, i got the following error:
`exec /app/entrypoint.sh: no such file or directory`
When i change the "End-of-Line" character ( Notepad++, E) from CR/LF to  CR etherything works fine!
Can you please adapt this in the code?

### Steps to reproduce

1. Build docker image (Makefile)
2. Start it (Portainer or manual)
3.  Errormessage occurs: exec /app/entrypoint.sh: no such file or directory
4.
...

### Configuration details

```yaml
I use Windows WSL2 with Ubuntu 24.10 for testing  and Ubuntu24.10 in my production env, same behavior
```

### Log details

```text
exec /app/entrypoint.sh: no such file or directory
```

### What type of operating system are you running?

Linux

### Version

0.128.0

andig, 2024-07-17:
Please open PR if this needs be changed. This is not an issue for the evcc images per se.

closed 2024-07-17 by andig

cbshopper, 2024-07-17:
CR is standard for Linux as far as i know CR/LF ist standard for windows. The file in the repro has CR/LF.

andig, 2024-07-17:
Imho both statements are wrong.

cbshopper, 2024-07-17:
look here: https://stackoverflow.com/questions/419291/historical-reason-behind-different-line-ending-at-different-platforms