evcc-io/evcc #14912: SoC Limit wird nicht gespeichert

owanckel, 2024-07-17:
### Describe the bug

Wenn ich in der GUI das SoC Limit auf einen Wert <100% setze, dann gilt der Wert scheinbar nur für diese eine Ladesession und wird nicht gespeichert. Ziehe ich den Stecker und stecke ihn dann wieder ein, dann ist das SoC Limit wieder bei 100%.

### Steps to reproduce

1. Stecker rein
2. SoC Limit in der GUI auf einen Wert <100% setzen
3. Stecker raus
4. Warten bis Evcc mitbekommen hat, dass kein Auto mehr verbunden ist
5. Stecker wieder rein
6. => SoC Limit ist wieder bei 100%

### Configuration details

```yaml
network:
  schema: http
  host: localhost 
  port: 7070

log: info
levels:
  cache: error

plant: ***
interval: 30s 
sponsortoken: ***e
telemetry: true

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.68  
  port: 80  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.68  
  port: 80  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.68  
  port: 80  
  name: battery3

chargers:
- type: template
  template: easee 
  user: ***
  password: ***
  charger: EH96GH9Q  
  timeout: 20s  
  authorize: true  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
[... 130 lines cut ...]
Jul 17 16:54:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:54:30 !! active phases: 1p = min(1p measured 0p vehicle 0p physical 0p charger)
Jul 17 16:54:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:54:30 pv charge current: min 6A > 1.38A (1063W @ 1p, battery: true)
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 charge power: 1388W
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 charge currents: [5.74 0 0]A
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 detected active phases: 1p
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 charge total import: 7914.801kWh
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 charger status: C
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 vehicle soc: 44%
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 vehicle range: 114km
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 !! active phases: 1p = min(1p measured 0p vehicle 0p physical 0p charger)
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 !! active phases: 1p = min(1p measured 0p vehicle 0p physical 0p charger)
Jul 17 16:55:00 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:00 pv charge current: min 6A > 1.42A (1053W @ 1p, battery: true)
Jul 17 16:55:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:30 charge power: 1388W
Jul 17 16:55:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:30 charge currents: [5.74 0 0]A
Jul 17 16:55:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:30 detected active phases: 1p
Jul 17 16:55:30 raspberrypi evcc[17391]: [lp-1  ] DEBUG 2024/07/17 16:55:30 charge total import: 7914.801kWh
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.2

VolkerK62, 2024-07-17:
Normal. Du kannst aber für jedes Auto ein individuelles Default-Limit speichern.
Auto auswählen, im Loadpoint unten Mitte auf "Plan", dann Tab "Ankunft" auswählen.

owanckel, 2024-07-17:
Okay, prima, danke!

closed 2024-07-17 by owanckel