evcc-io/evcc #14923: Loadstatistic wrong calculated in EVCC with Easee Wallbox. Meter Count in EVCC correct

free118, 2024-07-18:
### Describe the bug

I'm using EVCC on HA together with an easee wallbox. 
Yesterday I pluged in my BEV with ~50% SoC in to be charged via PV only.
So with 39kwh in total I assume that in total ~ 18kwh will be charged. 
I let the vehicle charge and sit over night. So charging continiues today morning. 
After that i took a look into the loading stats. 
And they look wrong. 
I verfied that Easee stats and conters are correct. 
Counteres in EVCC stats are correctly show, but the math is wrong. 

[image]
[image]

### Steps to reproduce

1.Plugin BEV start charging in PV only mode
2. Charging stop due to night 
3. let it connected over night 
4. charging starts with sun in the morning
5. --> startistics are wrong shown in EVCC, EASEE statistics are correct

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168..
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 25s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  #lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: my_grid
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.. # IP-Adresse oder Hostname 

chargers:
  - name: charger
    type: template
    template: easee
    user: 
    password:  #wie Login für Easee App oder Web Portal (https://easee.cloud)
    charger: 
    
    

vehicles:
  - name: kona
    type: template
    template: evnotify
[... 195 lines cut ...]
[site  ] DEBUG 2024/07/18 12:55:09 pv power: 14511W
[site  ] DEBUG 2024/07/18 12:55:09 site power: -14511W
[lp-1  ] DEBUG 2024/07/18 12:55:09 !! session: chargeRater.chargedEnergy=21.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/18 12:55:09 charge total import: 1793.140kWh
[lp-1  ] DEBUG 2024/07/18 12:55:09 charger status: A
[site  ] DEBUG 2024/07/18 12:55:33 ----
[lp-1  ] DEBUG 2024/07/18 12:55:33 charge power: 0W
[lp-1  ] DEBUG 2024/07/18 12:55:33 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/18 12:55:33 grid meter: -14753W
[site  ] DEBUG 2024/07/18 12:55:33 grid powers: [-4977 -4883 -4893]W
[site  ] DEBUG 2024/07/18 12:55:34 grid currents: [-20 -19.9 -20]A
[site  ] DEBUG 2024/07/18 12:55:34 pv power: 14753W
[site  ] DEBUG 2024/07/18 12:55:34 site power: -14753W
[lp-1  ] DEBUG 2024/07/18 12:55:34 !! session: chargeRater.chargedEnergy=21.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/18 12:55:34 charge total import: 1793.140kWh
[lp-1  ] DEBUG 2024/07/18 12:55:34 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.128.2

andig, 2024-07-18:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

We'll need a log file that allows to reproduce the issue. The log above is not related. Closing until matching log available.

free118, 2024-07-18:
First Screenshot show the charging in the Easee App. 
So you can see that 21,5kWh were charged and measured by them. Expected ~18kWh. So this would fit. I guess.

If you took a look into the 4 loadings from EVCC (why ever this one charging is shown as four individual charges here) 
you will see that the "Geladen" amount is wrongly calculated in 3 of 4 times.

Start at the top left of the screenshot: 
Counter Value: 1792,9 kWh - 1784,7 kWh = 8,2kWh --> but EVCC shows 20,4 kWh as charged.

Lower left of the screenshot: 
Counter Value: 1784,7 kWh - 1782,7 kWh = 2,0 kWh --> but EVCC shows 12,2 kWh as charged.

Overall EVCC shows for this charge an total amount of: 42,87kWh. 
Which i think didn't make any sense because the Cars Battery is just 39kWh and it was ~50% charged, when I started charging.

free118, 2024-07-18:
I can't reach the old log. 
But i did a new charging. 
Attached the log and here the screenshot

[image]

[evcc-20240718-165112-debug.log](https://github.com/user-attachments/files/16285013/evcc-20240718-165112-debug.log)

Total calculation is 0,4kWh off, but meters ar showing up correctly

andig, 2024-07-20:
Here's what I can see:

```
[lp-1  ] DEBUG 2024/07/18 16:06:24 charge total import: 1793.140kWh
[lp-1  ] INFO 2024/07/18 16:06:24 start charging ->

[lp-1  ] DEBUG 2024/07/18 16:11:24 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/07/18 16:11:24 charge total import: 1793.142kWh

[lp-1  ] DEBUG 2024/07/18 16:50:59 charger status: B
[lp-1  ] INFO 2024/07/18 16:50:59 stop charging <-
[lp-1  ] DEBUG 2024/07/18 16:50:59 charge total import: 1799.455kWh
[lp-1  ] DEBUG 2024/07/18 16:50:59 !! session: chargedEnergy=6.0 > chargedEnergy=0.0
[lp-1  ] DEBUG 2024/07/18 16:50:59 !! session: start=1793.140 stop=1799.455 chargedEnergy=6.010
```

    chargedEnergy=6.0

comes from the Easee Api (set `easee: trace` to validate yourself). This value is wrong when the session stops. We can either:

- add some sort of update mechanism for Easee (usually not a good idea)
- remove `api.ChargeRater` and rely on meter readings
- add more logic to the loadpoint to use `api.ChargeRater` if and only if the resulting value is _bigger_ than the meter difference. This could also help with https://github.com/evcc-io/evcc/issues/13726#issuecomment-2225494242.

@GrimmiMeloni wdyt?

GrimmiMeloni, 2024-07-24:
Regression of https://github.com/evcc-io/evcc/issues/9884 ?

Not sure if it is the same issue though. @allcoolusernamesaregone do you by any chance remember from the top of your head if the issue described here matches what we discussed back in the day?

allcoolusernamesaregone, 2024-07-24:
Moin, also ja, sieht das so aus, als wäre das der Fehler, den ich in #9884 schonmal beschrieben habe.
Insb. im Kommentar https://github.com/evcc-io/evcc/issues/9884#issuecomment-1722300053
bzw. im Kommentar https://github.com/evcc-io/evcc/issues/9884#issuecomment-1720129364
```
So oder so, wenn man die Werte von meter_start_kwh,meter_end_kwh,charged_kwh sieht, diese sind ja nicht konsistent.
Entweder müsste man start + end Angabe vertrauen, dann kann charged errechnet werden als Differenz.
Oder start nehmen charged addieren um end zu erhalten.
Beim zweiten Ladevorgang ist meter_end_kwh und charged_kwh falsch.
```

Ist hier ja ähnlich bzw. wohl das Problem.  Die initialen Screenshots oben zeigen ja 21,5 kWh laut Easee App.
Das ist konsistent zum Startzählerstand 1771,6 vom Screenshot mit grüner Einrahmung, bis zum Endzählerstand im Screenshot oben links mit roter Einrahmung 1792,9, Rundungsdifferenzen berücksichtigt.

EVCC liegt hier insofern falsch, als dass, wie von mir im anderen Issue beschrieben, nicht die geladene Menge errechnet wird als Differenz Start-End  (zumindest mein Stand von damals, k.A. ob das noch so ist, sieht aber so aus!), sondern der API Wert von SESSION_ENERGY , der in einem Ladevorgang von Easee fortgeschrieben wird, welcher aber in EVCC in verschiedene Teil-Ladevorgänge gesplittet wird.

Ich selbst konnte das in der Vergangenheit nicht mehr zufällig beobachten nach #9884, habe auch nicht mehr aktiv danach gesucht, aber leider kann ich es jetzt bestätigen, wenn ich mir meine aktuellen Ladevorgänge ansehe. 
Da ich EVCC per cronjob neustarte morgens
`03     06    * * *           /usr/sbin/service evcc restart`
werden aus EINER Session lt. Easee ZWEI Sessions in EVCC. 
Das u.a. löst klar bei mir das Problem aus. 

Warum bei @free118 das in verschiedene Ladevorgänge geteilt wird ist mir nicht klar.

Von gestern auf heute habe ich zumindest die Logs von heute. Der Easee Ladevorgang ging von gestern Abend bis heute morgen bis zum Abstecken. Wird von EVCC in mehrere Sessions geteilt, u.a. wg. Neustart EVCC.

Zeitraum	Di., 23. Juli, 17:20 -> Di., 23. Juli, 18:52   (Grund für Session Split ist mir nicht klar - könnte Neustart gewesen sein wg. Update - oder unbekannt)
Geladen	3,8 kWh    richtig!
Zählerstand	4.066,2 kWh -> 4.070,0 kWh

Zeitraum	Di., 23. Juli, 18:53 -> Mi., 24. Juli, 06:03   (Cronjob Neustart EVCC)
Geladen	4,6 kWh     falsch --> 0,8
Zählerstand	4.070,0 kWh -> 4.070,8 kWh   

Zeitraum	Mi., 24. Juli, 08:02 (Fortsetzung Ladevorgang) -> Mi., 24. Juli, 08:50 
Geladen	6,3 kWh   falsch, 1,2 
Zählerstand	4.070,8 kWh -> 4.072,0 kWh

Das Log zeigt, dass die 4,6 kWh an Menge zu viel in der zweiten Session aus dem Wert von der Easse Api kommen, die die Menge in einer Session fortschreibt. Die Gesamtmenge der Session lt. Easee Api und App ist 6,25.
```
Jul 24 06:03:09 evcc[2615154]: [easee ] TRACE 2024/07/24 06:03:09 ProductUpdate : (2024-07-23 17:30:00 +0000 UTC) SESSION_ENERGY 4.603124141693115
Jul 24 08:03:13 evcc[2615154]: [easee ] TRACE 2024/07/24 08:03:13 ProductUpdate : (2024-07-23 17:30:00 +0000 UTC) SESSION_ENERGY 4.603124141693115
Jul 24 08:43:00 evcc[2615154]: [easee ] TRACE 2024/07/24 08:43:00 ProductUpdate : (2024-07-24 06:42:59 +0000 UTC) SESSION_ENERGY 5.857141494750977
Jul 24 08:50:00 evcc[2615154]: [easee ] TRACE 2024/07/24 08:50:00 ProductUpdate : (2024-07-24 06:49:58 +0000 UTC) SESSION_ENERGY 6.254399299621582
```

Ich denke EVCC müsste beim Session Start den aktuellen Wert von SESSION_ENERGY, falls > 0, von späteren SESSION_ENERGY Werten abziehen. Im vorigen Log also die 4.603124141693115 von 08:03:13 abziehen von 6.254399299621582 um 08:50.
Ergibt dann ca. 1,65 kWh. 

Allerdings stimmen auch sonstige Protokolle nicht.
Auch von kurzen Ladevorgängen nicht, bei denen keine falschen SESSION_ENERGY Werte vorlagen beim Start.
Z.B. habe ich am 23.7. einen Ladevorgang mit 0,895 kWh lt. EVCC, Easee App sagt 0,9 kWh. Soweit ok.
Aber Start->Endstand der Easee lt. EVCC waren 4065,3 bzw. 4066 kWh. 

Vorige Sitzung war von 4063,4 bis 4064,8 kWh lt. EVCC, geladen 1,9 lt. EVCC  UND(!) lt. Easee App, ich errechne da 1,4.
Sitzung lt. EVCC von 16:57 bis 17:50, lt. Easee App 16:57 bis 17:49.

Da ich minütlich den Zählerstand der Easee, aus der Cloud gepollt, protokoliere in einer SQL Datenbank sehe ich folgende Änderungen:
```
Zeitstempel	     Stand
21.07.2024 16:56:01	4063,395
21.07.2024 16:57:01	4063,395
..
21.07.2024 17:22:01	4063,965
21.07.2024 17:23:01	4064,231
..
21.07.2024 17:32:01	4064,231
21.07.2024 17:33:01	4064,512
..
21.07.2024 17:42:01	4064,512
21.07.2024 17:43:01	4064,841
..
21.07.2024 18:01:02	4064,841
21.07.2024 18:02:02	4065,278
```

Erst über 10 Minuten nach Ladeende wird der aktuelle Zählerstand übermittelt bzw. ist in der Cloud abrufbar.
SESSION_ENERGY ist aber korrekt.

Irgendwas stimmt da nicht bei Easee momentan, ungeachtet des Problems, dass EVCC Ladevorgänge unterteilt bei @free118 aus unbekannten, bei mir aus bekanntem Grund.

Mir erscheint die Cloud und App in den letzten Tagen aber auch ziemlich zäh, bzw. EVCC zeigt anderen Status als die App als das Auto. App aktualisiert nicht richtig beim "aus dem Hintergrund holen".
Vielleicht oder vermutlich haben diese Inkonsistenzen auch damit zu tun.

andig, 2024-07-27:
> Das Log zeigt, dass die 4,6 kWh an Menge zu viel in der zweiten Session aus dem Wert von der Easse Api kommen, die die Menge in einer Session fortschreibt. Die Gesamtmenge der Session lt. Easee Api und App ist 6,25.

Eine Session i.S. der WB geht von An- bis Abstecken, egal ob evcc läuft.

> Ich denke EVCC müsste beim Session Start den aktuellen Wert von SESSION_ENERGY, falls > 0, von späteren SESSION_ENERGY Werten abziehen.

Das tut evcc.

Siehe https://github.com/evcc-io/evcc/issues/14923#issuecomment-2241086080- da liefert die Easee schon den falschen Wert.

allcoolusernamesaregone, 2024-07-27:
Bei mir tut evcc das aber ganz offensichtlich gerade nicht.
Vielleicht sollte evcc das tun, aber im Ergebnis macht es das nicht.

Zum Beginn des Ladevorgangs (in evcc, nicht der Wallbox!) ist der Wert von SESSION_ENERGY 4.60... . 
Was auch völlig richtig ist, Auto wurde nicht abgesteckt. Zum Ende des Ladevorgang steht der Wert bei 6.2543. 
In evcc steht auch genau dieser falsche Wert als geladene Menge in der Datenbank. 
Richtig wäre das Merken von 4,6 beim Ladestartvorgang in evcc, und abziehen dieser 4,6 am Ende von den 6,2 => 1,6.

andig, 2024-07-27:
Gibts ein Log das das zeigen würde?

allcoolusernamesaregone, 2024-07-27:
nein, leider sind die Logs bei meinem SBC mangels größerer Ramdisk eher kurz.
Aber es gibt das hier:
[image]
Das zeigt den Datenbankeintrag zum Ladevorgang, mit den 6,3 kWh - das ist genau der SESSION_ENERGY Wert zum Ladeende.
Hier hätte der Wert vom Start vom Ladevorgang abgezogen werden müssen, um es richtig zu haben.
Der Golf lädt auch nur mit 7,4 kWh, ist also 100% garantiert nicht richtig mit den 8,1 kW..

Die Frage ist ja, warum teilt evcc den Wallbox-Ladevorgang in mehrere "Sitzungen". Bei mir ist mir das klar mit dem Restart von evcc, zumindest um 06:03 Uhr. Beim Threaderöffner dagegen k.A. warum.

Mir ist das bislang nicht aufgefallen, da ich diese Protokolle nicht weiter beachte. Der geladene Wert selbst, was damals Anlass meines Issues (oder Discussion) war, war ja dass auch tatsächlich Wallbox-Ladevorgang übergreifend nicht mit 0 gestartet wurde beim Ladevorgang, da die Easee Api noch einmal kurz den letzten SESSION_ENERGY in der neuen Sitzung sendet. 
Das wurde ja gelöst mit einem PR.

andig, 2024-07-27:
> Aber es gibt das hier:

Das nutzt nichts. Ohne Log gibt es hier nichts zu tun. Falls es nicht möglich ist, eines zu produzieren würde ich hier schließen.

Falls das Problem aus https://github.com/evcc-io/evcc/issues/14923#issuecomment-2241086080 nicht mehr besteht gibts auch da nichts zu tun.

allcoolusernamesaregone, 2024-07-27:
> Ohne Log gibt es hier nichts zu tun.

Sehe ich ein wenig anders, das Problem ist offensichtlich, der Bug da, kann auch ohne Logs angegangen werden - wenn man saubere Statistiken möchte. 
Da ich die nicht nutze: mir egal ;)

andig, 2024-07-27:
Gute, dann zurück zum Thema.

andig, 2024-07-27:
> Regression of https://github.com/evcc-io/evcc/issues/9884 ?

@GrimmiMeloni warum? Das Log zeigt m.E. klar, dass der falsche Wert von der Easee kommt.

allcoolusernamesaregone, 2024-07-27:
Das IST das Thema.. Die Werte Zählerstand Start & Stop bzw. SESSION_ENERGY sind nicht immer konsistent.
Ich frage mich schon ein wenig, ob Du mit Deiner rabiaten Art und Weise glaubst dem Projekt Gutes zu tun.  Beleidigte Leberwurst?  Verletzter Ehrgefühl? Hier ist doch kein einziger Kommentar offtopic.  Alles ein großes Thema.
Wozu brauchst Du bitte schön ein Log um anzuerkennen, dass was schief gelaufen ist? Das sollte man problemlos im Code nachvollziehen können, es liegen alle Fakten auf dem Tisch.

andig, 2024-07-27:
> Wozu brauchst Du bitte schön ein Log um anzuerkennen, dass was schief gelaufen ist? Das sollte man problemlos im Code nachvollziehen können, es liegen alle Fakten auf dem Tisch.

Mach das bitte wenn du es kannst. Ich kann es nicht. Weiterhin ist Dein Problem- Energiemenge >0 beim evcc Start ein anderes als das Theme eingangs- abweichende Menge vs. Zählerstände _nach_ evcc Start. Sei bitte so nett und bleib beim Thema. Danke.

GrimmiMeloni, 2024-07-27:
> > Regression of #9884 ?
> 
> @GrimmiMeloni warum? Das Log zeigt m.E. klar, dass der falsche Wert von der Easee kommt.

War nur geraten, daher auch als Frage formuliert.

Soweit ich es überblicke ist es anders als #9884. Die Daten von der API kommen bekanntermaßen verzögert. Das lässt sich nur bedingt gut mit der Logik/Ablauf in evcc vereinbaren.

@andig , welche Idee hattest du denn mit dem meter oben? Kannst du das noch mal genauer erklären, wie das funktionieren würde?

allcoolusernamesaregone, 2024-07-27:
Ja dann, alles klar. Hier https://github.com/evcc-io/evcc/issues/14923#issuecomment-2247452634 wird jedenfalls genug Optimierungspotential aufgezeigt. Ist zwar zur Hälfte der Easee Api geschuldet, ändert aber nichts am Ergebnis.

> Weiterhin ist Dein Problem- Energiemenge >0 _beim_ evcc Start ein anderes als das Theme eingangs- abweichende Menge vs. Zählerstände _nach_ evcc Start.

Und NEIN, es ist für mich eben gerade kein anderes Thema.  
Ich habe zwar einen forcierten Auslöser für das Problem mit dem Neustart per Cron. 

Aber auch der Threaderöffner hat mehrere (4) Ladevorgänge in EVCC, obwohl es nur EIN Ladevorgang in der Wallbox ist.
Und damit ist es doch genau das gleiche Thema.

andig, 2024-07-27:
Ändert doch nix. Ohne Logfile nicht nachvollziehbar. Ich würde jetzt wirklich darum bitten, dieses zu liefern oder den Thread in Ruhe zu lassen. Danke.

closed 2024-08-08 by github-actions[bot]