evcc-io/evcc #14935: Eebus connection failed after some time (Elli charger)

sarsonj, 2024-07-19:
### Describe the bug

Hi,
I successfully integrated evcc with my Elli based charger (Skoda IV Connect). Everything seems to work first - the charger is correctly charging right amonut of power based on actual solar conditions.

However, after few minutes, the charger disconnect from evcc. The charging continues, but with low (probably default) power. The evcc doesn't reconnect.

In logs it looks like that websocket connection failed and the evcc didn't try to reconnect again (there are no more logs from eebus after disconnection).

Would not be possible to reconnect to Elli charger in case of this websocket failure?

### Steps to reproduce

1. Connect car (in my case e-Golf) to Elli charger
2. Charging starts and everything works as expected
3. After few minutes, the evcc disconnect from charger (there is No car is connected message). In fact the charging continue, but with default low power
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: f828bdbc64de5fce5bd54a942d3a97e57b974d8948a779b72516ceefa052a1da

interval: 30s # control cycle interval

meters:
  - type: template
    template: solax
    id: 1
    host: 192.168.0.54
    port: 502
    usage: grid
    modbus: tcpip
    name: grid1
  - type: template
    template: solax
    id: 1
    host: 192.168.0.54
    port: 502
    usage: pv
    modbus: tcpip
    name: pv2
  - type: template
    template: solax
    id: 1
    host: 192.168.0.54
    port: 502
    usage: battery
    modbus: tcpip
    name: battery3

chargers:
  - type: template
    template: elliconnect
    ski: 2AFC E767 842C CFAE 3000 400B 5ACF 2910 DC26 594C
    name: wallbox4

vehicles:
  - type: template
[... 513 lines cut ...]
[eebus ] DEBUG 2024/07/19 14:51:04 Recv: d:_i:47859_Elli-Wallbox-2201A1WPGP:Measurement to Measurement notify 432 MeasurementListData
[eebus ] DEBUG 2024/07/19 14:51:04 Recv: d:_i:47859_Elli-Wallbox-2201A1WPGP:Measurement to Measurement notify 435 MeasurementListData
[eebus ] DEBUG 2024/07/19 14:51:04 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 334 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:07 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 335 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:10 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 336 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:13 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 337 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:16 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 338 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:19 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 339 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:22 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 340 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:25 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 341 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:28 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 342 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:31 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 343 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:34 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 344 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/19 14:51:37 2afce767842ccfae3000400b5acf2910dc26594c websocket read error:  read tcp 192.168.0.77:37186->192.168.0.61:4712: i/o timeout
[eebus ] DEBUG 2024/07/19 14:51:37 mdns: announce
[eebus ] DEBUG 2024/07/19 14:51:37 mdns: using zeroconf
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.3

premultiply, 2024-07-19:
/cc @DerAndereAndi
Irgendeine Idee was das nun schon wieder sein könnte? Oder Elli-Standardfehler?

augard, 2024-07-19:
@sarsonj I am not sure if there is any difference, but you can try set template to `template: ellipro`.

DerAndereAndi, 2024-07-19:
evcc can only reconnect when the wallbox announced itself via mDNS, which it doesn't seem to do. Also the wallbox could also connect which it doesn't do. One reason might be that the EEBUS software crashes on the wallbox.

- 14:27:13 EEBUS Software in the wallbox crashes
- 14:28:03  evcc initiates a connection successfully
- 14:51:37 connection is lost, no mDNS announcement seen in the log, nothing evcc can do about

Please note: every time evcc connects to the wallbox while an EV is connected, charging will not work as expected. Disconnected can be expected to happen when using a WLAN connection and is more reliable using a LAN connection.

deThommy, 2024-07-19:
Ich habe genau dasselbe Problem. Elli Charger Pro, IONIQ 5. Meine Elli Wallbox ist mit LAN Kabel verbunden. Bis 0.127.3 lief alles ohne Probleme. Danach ging es dann los mit den Problemen (ab ca. 0.128.x). Erst die Thematik, dass das Auto immer anlädt und direkt wieder abschaltet. Mit 0.128.3 kann ich evcc gar nicht mehr starten. Mein Log ist identisch mit dem OP. 

Kann ich noch was zur Identifikation beisteuern? Bin ab heute Nacht im Urlaub und hoffe, dass es danach wieder alles geht :P

DerAndereAndi, 2024-07-20:
@deThommy das gleiche hätte auch mit der neuen Version funktioniert.

Olf1337, 2024-07-20:
Im Moment habe ich beim Laden mit Überschuss mit 0.128.3 keine Probleme. 
Ich schaue mal ob ich heute Nacht wieder dieses Phantom on/off Phänomen habe.

andig, 2024-07-21:
> Mein Log ist identisch mit dem OP.

Das Problem hier ist fehlende Verbindung? .Bitte probieren ob erneute Kopplung hilft.

Olf1337, 2024-07-22:
ich muss mich korrigieren.
leider ist das verhalten recht instabil. Mal verliert die WB die Kopplung, mal lässt sich das Laden über EVCC (AUS/PC/Schnell) nicht einstellen und mal wird kein EV erkannt.
Eine Wallbox ließ sich seid heute gar nicht mehr koppeln. Alles probiert: Neustart EVCC, Neustart WB ohne EV, Factory Reset WB, alles ohne Erfolg.
Ich bin dann aus Verzweiflung zurück auf die 0.127.3 und zack sofort beide WB gekoppelt und das ansteuern ist "stabiler".

andig, 2024-07-22:
Ohne ein Logfile mit dem sich die Unterschiede analysieren lassen wird das nicht helfen Workarounds für die Box zu finden.

DerAndereAndi, 2024-07-27:
Im neuen Nightly Build sind noch ein paar Änderungen ins Blaue rein gemacht. Keine Ahnung ob die was helfen können. Aber auch hier, ohne ein Logfile mit EEBUS auf Trace Level und auch den restlichen Logdaten von evcc, kann man nicht mal gut raten.

andig, 2024-07-28:
@deThommy probierst Du mal das nightly? Bitte trace log!

deThommy, 2024-07-28:
Ich habe mir gerade die aktuelle 0.128.4 nightly geladen (ist die noch aktuell, jetzt wo gerade die 0.129 raus ist?). Damit habe ich bislang keine Probleme.

Auch das durchschalten der Modi (On, PV, MIN+PV, now) ging gerade ohne Probleme. Gerade ist aber zu wenig Sonne da. Ich lasse das über Nacht einmal laufen, den Wagen angestöpselt und melde mich morgen dazu - aktuell wirkt noch alles stabil.

deThommy, 2024-07-29:
Gab bislang heute keine weiteren Probleme und es läuft stabil. Auch nach mehrmaligem (automatischem) umschalten aufgrund zu geringer PV-Leistung.

Ist das alles so in der 129er drin, sodass ich von der Nightly weg kann?

Danke euch fürs fixen!!!

DerAndereAndi, 2024-07-29:
@deThommy Ja, das ist in der 129er drin.

sarsonj, 2024-07-30:
Hi,

the version 129 make it more stable. To better isolate the charger problem, I have site with charger only, no solar panels / battery. So that now I can select only Off or Fast charging.

However, still, after some time, I got "Charger timeout" error in the logs:

[eebus ] DEBUG 2024/07/30 11:15:20 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1693 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:22 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1694 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:24 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1695 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:26 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1696 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:28 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1697 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:30 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1698 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:32 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1699 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:34 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1700 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:36 Send: d:_i:47859_Elli-Wallbox-2201A1WPGP notify 1701 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/30 11:15:38 550ea237e1c2739b159bd8367fdd6621b8b50bc6 websocket read error:  read tcp 192.168.0.77:46754->192.168.0.61:4712: i/o timeout
[eebus ] DEBUG 2024/07/30 11:15:38 ski 550ea237e1c2739b159bd8367fdd6621b8b50bc6 disconnected
[eebus ] DEBUG 2024/07/30 11:15:38 mdns: announce
[eebus ] DEBUG 2024/07/30 11:15:38 mdns: using zeroconf
[site  ] DEBUG 2024/07/30 11:15:39 ----
[lp-1  ] DEBUG 2024/07/30 11:15:39 charge power: 7360W
[site  ] DEBUG 2024/07/30 11:15:39 pv power: 0W
[site  ] DEBUG 2024/07/30 11:15:39 site power: 7360W
[lp-1  ] DEBUG 2024/07/30 11:15:39 !! active phases: 2p = min(0p measured 2p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/30 11:15:39 !! session: chargeRater.chargedEnergy=5.7 - chargedAtStartup=0.0
[lp-1  ] ERROR 2024/07/30 11:15:39 charger status: timeout
[site  ] DEBUG 2024/07/30 11:16:09 ----

The logs continues like this - timeout forever. It never restore back. What helps is restart of evcc - then everything works again perfectly. Could be that "connection reset" made by evcc. When restart of evcc helps, wouldn't help some internal "connectin restart" in app itself?

In attachment are logs where the timeout error is shown and second log after restart.

[logs-when-timeout-before-restart.log](https://github.com/user-attachments/files/16425891/logs-when-timeout-before-restart.log)
[logs-when-timeout-after-restart.log](https://github.com/user-attachments/files/16425890/logs-when-timeout-after-restart.log)

DerAndereAndi, 2024-07-31:
Thanks for the log.

- So the connection is lost because the Wallbox doesn't respond any more. Are you connected via WLAN? If so, a LAN connection will definitely help if that is possible to do.
- Then the mDNS entry for the wallbox can't be found, so no connection is initiated, but also the wallbox doesn't initiate a connection (which it should)
- On restart the mDNS entry is found again, and evcc can initiate a connection

So there could be an issue with the mDNS entry not correctly being reported even though it should at some time be there. I'll check if I can simulate this.

andig, 2024-07-31:
Super Idee @DerAndereAndi neues Nightly in 30min zum Test.

DerAndereAndi, 2024-07-31:
One additional note: with this fix, the wallbox should be connected again. BUT it will likely not be able to change the charge settings unless you unplug the EV from the charger. This is due to software bugs in the wallbox that can not be worked around.

Only solution for this scenario: use a LAN connection.

sarsonj, 2024-07-31:
> One additional note: with this fix, the wallbox should be connected again. BUT it will likely not be able to change the charge settings unless you unplug the EV from the charger. This is due to software bugs in the wallbox that can not be worked around.

I don't thing so. When "connection lost", the charging changes to "low" power changing. But when evcc restarts and everything reconnects, I can set again "full" charging or turn charging off. I don't have to reconnect car (it is e-golf 2018 model). I have the latest firmware on the Elli charger.
> 
> Only solution for this scenario: use a LAN connection.
I have very good WiFi connection so that I am not sure, that this is problem with my WiFi. Or do you think, that this is problem with WiFi stack on the charger and that it will works better over the lan?

DerAndereAndi, 2024-07-31:
I can assure you, that this problem will happen, as it has been faces over the past two years by many many users.

Also, no matter how good your WIFI connection is, the connection loss does not happen on LAN and so does the problem not occur there.

sarsonj, 2024-07-31:
@DerAndereAndi OK, I will try the LAN connection.

jomach, 2024-07-31:
see https://github.com/evcc-io/hassio-addon/issues/99#issuecomment-2258490968

jomach, 2024-07-31:
and https://github.com/evcc-io/docs/pull/590/files#diff-7da3fb7b91f600ad31b84ed50c36c310f23142a4e1df7971809383874f018220R1065

closed 2024-08-09 by andig

andig, 2024-08-18:
https://github.com/evcc-io/evcc/releases/tag/0.130.0 contains a refactor of the EEbus charger that might or might not help here.

jomach, 2024-08-18:
@DerAndereAndi  I just tested this and I have the same issue still with 1.30.0 

```
@andig  @DerAndereAndi  I just tested 1.30.0 and I think we still have a problem with : 

```
[eebus ] DEBUG 2024/08/18 21:41:24 ski: 57a55237e44bd3838818586f239d55443a3a1de1 name: Livo-EVB-500-015-503 brand: EVBox model: Livo typ: ChargingStation identifier: EVBox-Livo-EVB-500-015-503 register: false host: EVB-500-015-503.local. port: 4712 addresses: [192.168.1.219]
[eebus ] DEBUG 2024/08/18 21:41:24 trying to connect to 57a55237e44bd3838818586f239d55443a3a1de1 at EVB-500-015-503.local.
[eebus ] DEBUG 2024/08/18 21:41:24 initiating connection to 57a55237e44bd3838818586f239d55443a3a1de1 at EVB-500-015-503.local.:4712/ship/
[eebus ] DEBUG 2024/08/18 21:41:34 connection to 57a55237e44bd3838818586f239d55443a3a1de1 failed: dial tcp 192.168.1.219:4712: i/o timeout
[eebus ] DEBUG 2024/08/18 21:41:34 trying to connect to 57a55237e44bd3838818586f239d55443a3a1de1 at 192.168.1.219
[eebus ] DEBUG 2024/08/18 21:41:34 initiating connection to 57a55237e44bd3838818586f239d55443a3a1de1 at 192.168.1.219:4712/ship/
a[eebus ] DEBUG 2024/08/18 21:41:44 connection to 57a55237e44bd3838818586f239d55443a3a1de1 failed:  dial tcp 192.168.1.219:4712: i/o timeout
[eebus ] DEBUG 2024/08/18 21:41:44 mdns: announce
[eebus ] DEBUG 2024/08/18 21:41:44 mdns: using zeroconf
[eebus ] DEBUG 2024/08/18 21:41:44 trying to connect to 57a55237e44bd3838818586f239d55443a3a1de1 at EVB-500-015-503.local.
[eebus ] DEBUG 2024/08/18 21:41:44 initiating connection to 57a55237e44bd3838818586f239d55443a3a1de1 at EVB-500-015-503.local.:4712/ship/
[eebus ] DEBUG 2024/08/18 21:41:54 connection to 57a55237e44bd3838818586f239d55443a3a1de1 failed: dial tcp 192.168.1.219:4712: i/o timeout
[eebus ] DEBUG 2024/08/18 21:41:54 trying to connect to 57a55237e44bd3838818586f239d55443a3a1de1 at 192.168.1.219
[eebus ] DEBUG 2024/08/18 21:41:54 initiating connection to 57a55237e44bd3838818586f239d55443a3a1de1 at 192.168.1.219:4712/ship/
```