evcc-io/evcc #14939: Ford connect (again): unexpected status: 500 (Internal Server Error)

tobox, 2024-07-20:
### Describe the bug

About a month ago I successfully connected my ford to evcc using the new API and now it broke again...

### Steps to reproduce

Configured ford-connect as documented, worked properly for a while, does not work anymore. 

### Configuration details

```yaml
...
vehicles:
  - name: mustang
    type: template
    template: ford-connect
    title: Mustang
    clientid: 30990062-9618-40e1-a27b-7c6bcb23658a
    clientsecret: XXXXX~XXXXXXXXXXXXXXXXXXX-XXXX-
    accessToken: eyJhbGciOiJSUzI1NiIsImtpZCI6IlMxUEZhdzdkR2s3bHNF.....
    refreshToken: eyJraWQiOiI2cjIzQ2FTeTF4cFdUUFBxYVRtX01Vc2RKZGo.....
    capacity: 88
```

### Log details

```text
root@evcc:~# evcc vehicle --log trace
 
[main  ] INFO 2024/07/20 08:09:33 evcc 0.128.3 (9e75d8c7)
[main  ] INFO 2024/07/20 08:09:33 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/20 08:09:33 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/07/20 08:09:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/07/20 08:09:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/07/20 08:09:33 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/20 08:09:33 SELECT * FROM `settings` 31 <nil>
[mqtt  ] INFO 2024/07/20 08:09:33 connecting evcc-240278577 at tcp://mosquitto.fritz.box:1883
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[mqtt  ] DEBUG 2024/07/20 08:09:34 tcp://mosquitto.fritz.box:1883 connected
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2024/07/20 08:09:34 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[ford  ] TRACE 2024/07/20 08:09:34 POST https://dah2vb2cprod.b2clogin.com/914d88b1-3523-4bf6-9be4-1b96b4f6f919/oauth2/v2.0/token?p=B2C_1A_signup_signin_common
[ford  ] TRACE 2024/07/20 08:09:35 grant_type=refresh_token&refresh_token=eyJraWQiOiI2cjIzQ2FTeTF4cFdUUFBxYVRtX01Vc2RKZGo1RWlDTnRtME4yVTAxNTdFIiwidmVyIjoiMS4wIiwiemlwIjoiRGVmbGF0ZSIsInNlciI6IjEuMCJ9.SHt84y647OBcyAuh0tF9T9P53gp4Z-0fNVO7nEkPBnGiF1No0f0FcE2MnSDkmcF8m8ag0qOPgTZI6SHVEoBYAEI3U9C8LPWu0oa-8jVb3ue3Rp7eFaXhmQHp9v7SVgg-LC7gvA1gRAu8LuXGbfd6pkX3rD9A541nrzxEuFwShmkbKZVc4sCCJaQ8VD1FPYymEnVudjVDJNorRcBdECF9E8eaiE3aTEM193VNdqFauHaiVvHbrUELokpPqp4BGp4aZL3kSh038GFk2xztNnusEvEeBBziXQl6Zaz-OPGKqbUXO--tU2qpbtAWg0J4v64OW6G9GIjyxmxaM45Zbd6fiQ.nXhNIIjY8tZL9BjR.6oowp9lBL0A2OtCk-lDfIf9tuLRVlbkXJUfvffrY9LLhPLA95zsDYu05IDUSdEd2dsVLqOtVMc1jbTQCOApka682rtz_3EylBGyzhPn5IdNnfNqdYiAMfLjSPS3ASHda8OvcJh0vPzE6h-3xiLN4iisohm3XTvnYIwqfwkAVI18sLWQopWZ0-vW3GBApneva6g4sgOxiHzQGsLpGjsFThfhdHtFc49A70zYoWyC83cduPzXoOEKzmE5qtGn077Q2RabA7847jExk9XjLyGJe_fb65af-1apAFdfSezHewdqKYqF4d69JU1T4F6thGgbsJZ6olXkBG05TF_h7VSXBwOUcyROR4H7HsfNh_bbveUI8hgVVDdnk9Z5kiFtWPWKEhnxFUe04lErZR6HEk-ueuEAw34LpGY7VdTDxy5WMA8rnmoCQkpwAb9zo6FG2HL3EQjSME39-Xyi68dYlYa86OLzf5bZaBhNDmo7t3eSqDk612OGRTPZLi_mMsRzdxTPRymN55XQi97EaLa7mftifVtpzfpWyaZAEc7WVypH-CA9fxLRiseIZTIWh2-xoav6mYdxEPRlq4cs2XCX_svBKiC80VOnnl2Oh98oZJVKsF_9QWUBdNnuKg8BKPyvePHXulxQuIxvQZHPxywAZF0_1WnYKz30qL0GJl0m2idh2qFxLvzDvmCS0Gb0jUwULmW8pglEjpzthyR-yuBBUKjxIXAwf1oSlULwc1OOuo78E4WahasGsLEbm1QE4wsQFLCshLP2rLMgbZLMwF7rUzynASYVysBZ2QEgRJo-uW3FKYxmgvJwV7uPxIZdasuITqDHwYrsoeK9PWkyq9_FIF7P6G4Nk-bKZ_5ai3h6n4bO7nBo605tvFxRSI4gz5XfB5Q-niou0E-zMGv9Axj14A3x4IIdshB2dpByEqDwwfIOoc8KwBkimu5ugiusvbs4NOr1kBg.WncEI95oOVkvhoh1YiZVAA
--
{"access_token":"eyJhbGciOiJSUzI1NiIsImtpZCI6IlMxUEZhdzdkR2s3bHNFQmEzUjVWMnRLSzVYYnNIWEJsemFXZGhjNUVNdW8iLCJ0eXAiOiJKV1QifQ.eyJhdWQiOiJjMWU0YzFhMC0yODc4LTRlNmYtYTMwOC04MzZiMzQ0NzRlYTkiLCJpc3MiOiJodHRwczovL2RhaDJ2YjJjcHJvZC5iMmNsb2dpbi5jb20vOTE0ZDg4YjEtMzUyMy00YmY2LTliZTQtMWI5NmI0ZjZmOTE5L3YyLjAvIiwiZXhwIjoxNzIxNDU2OTc1LCJuYmYiOjE3MjE0NTU3NzUsImNsaWVudF9pZCI6IjMwOTkwMDYyLTk2MTgtNDBlMS1hMjdiLTdjNmJjYjIzNjU4YSIsImxvY2FsZSI6ImVuLVVTIiwiaWRwIjoiYjJjX0RwSzFPQW44ZFEiLCJtdG1JZCI6IjkwNWY2MjRkLTAzODMtNGE4ZC04NmEzLWMwZmEwZWFjOTUyYiIsInVzZXJHdWlkIjoiWkVKN2RTL0tsZTZsNE1DTUNCUllCK1FoL1hQZVJGQjFlL1RJVlBLNWJrQWJsZjUxZlZrMS9yNE41ZlZQbUZDNyIsImp0aSI6ImUyNTY2MWJlLTBmMzktNDgzMy1hMGI4LWNkNzgxMDQ3MWVjMSIsInNjb3BlIjoib3BlbmlkIG9mZmxpbmVfYWNjZXNzIGFjY2VzcyIsInN1YiI6ImIyOGY0NTFlLTRkOWEtNDdjYi04MmU5LWE4ZDExM2E1NmJjMyIsIm5vbmNlIjoiMTIzNDU2Iiwic2NwIjoiYWNjZXNzIiwiYXpwIjoiMzA5OTAwNjItOTYxOC00MGUxLWEyN2ItN2M2YmNiMjM2NThhIiwidmVyIjoiMS4wIiwiaWF0IjoxNzIxNDU1Nzc1fQ.fSUwCAKurH4OLZaCk06xnSUZGY6XKc7z8Jwv9O7gYCNOGts0DxOrS8A0ql7JU_AjjM2w9K6c5EDrGiSpbNpirHgyFdjsh-NjRFahfay9upDhPZLIzxXasP67U6jAWEyyJGMM1A9eG1weqbVxBHDwn4_FvIeOd7kxlxe3TqvM8J9_MP2UjYM45wcKFrNBhmLWObfjZyLOSl_-t5eGOl98FbPnLmvkJtXWphbrQQLwJcG1h3iymgX_iPp8g9i8TvSSqAtX0naHCfIcij80z-UpsQFzikiyTN4i-mIiRD9a3ObDRTpUtbEMp1jpyJ6SgNgJtcb8sMBxE-r1DFZPxzIYNg","id_token":"eyJhbGciOiJSUzI1NiIsImtpZCI6IlMxUEZhdzdkR2s3bHNFQmEzUjVWMnRLSzVYYnNIWEJsemFXZGhjNUVNdW8iLCJ0eXAiOiJKV1QifQ.eyJleHAiOjE3MjE0NTY5NzUsIm5iZiI6MTcyMTQ1NTc3NSwidmVyIjoiMS4wIiwiaXNzIjoiaHR0cHM6Ly9kYWgydmIyY3Byb2QuYjJjbG9naW4uY29tLzkxNGQ4OGIxLTM1MjMtNGJmNi05YmU0LTFiOTZiNGY2ZjkxOS92Mi4wLyIsInN1YiI6ImIyOGY0NTFlLTRkOWEtNDdjYi04MmU5LWE4ZDExM2E1NmJjMyIsImF1ZCI6IjMwOTkwMDYyLTk2MTgtNDBlMS1hMjdiLTdjNmJjYjIzNjU4YSIsIm5vbmNlIjoiMTIzNDU2IiwiaWF0IjoxNzIxNDU1Nzc1LCJhdXRoX3RpbWUiOjE3MTc0ODQ1NjEsImNsaWVudF9pZCI6IjMwOTkwMDYyLTk2MTgtNDBlMS1hMjdiLTdjNmJjYjIzNjU4YSIsImxvY2FsZSI6ImVuLVVTIiwiaWRwIjoiYjJjX0RwSzFPQW44ZFEiLCJtdG1JZCI6IjkwNWY2MjRkLTAzODMtNGE4ZC04NmEzLWMwZmEwZWFjOTUyYiIsInVzZXJHdWlkIjoiWkVKN2RTL0tsZTZsNE1DTUNCUllCK1FoL1hQZVJGQjFlL1RJVlBLNWJrQWJsZjUxZlZrMS9yNE41ZlZQbUZDNyIsImp0aSI6ImUyNTY2MWJlLTBmMzktNDgzMy1hMGI4LWNkNzgxMDQ3MWVjMSIsInNjb3BlIjoib3BlbmlkIG9mZmxpbmVfYWNjZXNzIGFjY2VzcyIsImF0X2hhc2giOiJpSGg4QjVsWmtLY2pBYV9Pem5CdDVnIn0.Myc-SnFg4t2bB_PQZVwt7T7AVcKrLqina1CyiNE7OX7kWwjgWpJoCO051wmCxji_h4v5pDkGP2zKJ-JHC6dimIXMMXX5J6SgiqkSY_d8CDjVUlmSgIjsOlwz3R0dsC-EmHxnoyzgwvqsDu8dMXF4FaN8bkkjrA60tSy-OC4KcfKegddbTeYAPdMMKBD_5BoHPezTcCJ_S7knWv85vD-b66Fr4v3MTwgFpyIADpCtinfJH4bTDbuyV9lgBerS0y_zJGLCE6dMTBIASNTZfyauR9WuvsIut6RZh-FwFRLf96mVnK1q5_HuzWQz6XNwYOzchf2b3BwuxA10V8sozJiIjQ","token_type":"Bearer","not_before":1721455775,"expires_in":1200,"expires_on":1721456975,"resource":"c1e4c1a0-2878-4e6f-a308-836b34474ea9","id_token_expires_in":1200,"profile_info":"eyJ2ZXIiOiIxLjAiLCJ0aWQiOiI5MTRkODhiMS0zNTIzLTRiZjYtOWJlNC0xYjk2YjRmNmY5MTkiLCJzdWIiOm51bGwsIm5hbWUiOm51bGwsInByZWZlcnJlZF91c2VybmFtZSI6bnVsbCwiaWRwIjoiYjJjX0RwSzFPQW44ZFEifQ","scope": [... cut ...]
[main  ] ERROR 2024/07/20 08:09:36 creating vehicle mustang failed: cannot create vehicle type 'template': cannot create vehicle type 'ford-connect': cannot get vehicles: unexpected status: 500 (Internal Server Error)
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford-connect': cannot get vehicles: unexpected status: 500 (Internal Server Error)
Capacity: 88.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.3 (9e75d8c7)

andig, 2024-07-20:
Token looks ok (remove from post!). Just wait a few days or needs information what might have changed in the Ford API...

tobox, 2024-07-21:
I had the same error since about a week ago, but I wanted to update from 0.128.2 to 0.128.3 to make sure it occurs with the current version.

tobox, 2024-07-22:
Funktionieren tut es immer noch nicht. Aber dafür jetzt mit andere Fehlermeldung:
```

[ford  ] TRACE 2024/07/22 08:24:05 GET https://api.mps.ford.com/api/fordconnect/v3/vehicles
[ford  ] TRACE 2024/07/22 08:24:06 {"Error": "Too Many Request"}
[main  ] ERROR 2024/07/22 08:24:06 creating vehicle mustang failed: cannot create vehicle type 'template': cannot create vehicle type 'ford-connect': cannot get vehicles: unexpected status: 429 (Too Many Requests)
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford-connect': cannot get vehicles: unexpected status: 429 (Too Many Requests)

```

GrimmiMeloni, 2024-07-26:
500 rarely (never?) is an indication for a client side problem. The server side is acting up - Ford needs to fix there stuff I'd say.

tobox, 2024-07-26:
The home assistant integration for my ford works for weeks without any issues. With evcc it fails since weeks. 

So there must be a difference somewhere, maybe a different api endpoint or similar. But I have no idea how to fix/debug this.

tobox, 2024-07-26:
I created new tokens and it works again. 

I think the official documentation needs to be updated. I found the token generation process here:

https://github.com/evcc-io/evcc/discussions/14117

closed 2024-07-26 by tobox

GrimmiMeloni, 2024-07-27:
Still interesting. So you are saying that sending Ford‘s servers an invalid/expired token, makes there servers reply http ststus 500 ?

miniluma, 2024-08-09:
> Still interesting. So you are saying that sending Ford‘s servers an invalid/expired token, makes there servers reply http ststus 500 ?

Right. I can confirm.
Error 500 -> New token generation -> works.