evcc-io/evcc #14943: Add Solar Analytics

derdew111, 2024-07-20:
Could you please add support to use data from Solar Analytics? The API is here:

The credentials below are generic and can be provided to other third parties:

https://api-docs.solaranalytics.com/
username: public-api@solaranalytics.com
password: weL0veDa2a!

https://info.solaranalytics.com/s/article/Solar-Analytics-Public-API

https://api-docs.solaranalytics.com/OpenAPI-Specs/sapublic-openapi.yaml

thanks

andig, 2024-07-20:
I have no idea how to use this. This doesn't work:

```
curl --location --request GET 'https://api-docs.solaranalytics.com/api/v2/site_data/?tstart&tend&gran&raw&trunc&all' \
--header 'User-Agent: Apidog/1.0.0 (https://apidog.com)' \
--header 'Authorization: Basic cHVibGljLWFwaUBzb2xhcmFuYWx5dGljcy5jb206d2VMMHZlRGEyYSE=' \
```

Need some example of the actual api call for the relevant data.

andig, 2024-07-28:
No more feedback on the API, closing.

closed 2024-07-28 by andig