evcc-io/evcc #14951: TeslaBleHttpProxy + 0.128.3

Neophyte85, 2024-07-21:
### Describe the bug

Hey Leute,
bin super Dankbar für euren Job und ihr macht das Toll, auch wenn ich bei der 0.128.3 graue Haare bekommen habe.
[image]
Ich hatte nach eurer Warnung mich doch dazu entschieden die 0.123.2 drauf zu lassen aber leider klappte es bei mir nicht.

Ich hatte dann doch die 0.123.3 drauf. Egal darum soll es mir nicht gehen.

Ich habe daraufhin die Anweisungen befolgt und die Vehicle config entfernt und in EVCC online Konfiguration eingepflegt.

Leider scheint er hier die den HTTP Proxy zu verwenden.
Ich hatte schon in der 0.123.2 das Gefühl das EVCC zunächst die 50 Kommandos über die TeslaAPI aufbraucht. Dann hab ich evcc neugestartet und dann lief es über TeslaBleHttpProxy. Jetzt bekomme ich aber keine Regelung mehr.

Und habe die Charger Too Many request Meldung dauerhaft. 
[image]

![config](https://github.com/user-attachments/assets/6ba
[image]
c6367-158c-462b-b1df-79e78f55c6d7)

Habt ihr einen Tipp für mich. Was ich noch prüfen kann. Oder wie ich drum herum arbeiten kann?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: dfda0d2e12df53dc1ae4bb7a419fa478ed6499952fe200bb0313fce9c6936074

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.34 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.078 # [currency]/kWh

# trial token, valid until 2024-03-20
sponsortoken: entfernt

interval: 4s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.244  
  port: 502  
  usage: grid  
[... 164 lines cut ...]
[lp-1 ] ERROR 2024/07/21 12:31:01 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:31:11 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:31:21 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:31:31 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:31:41 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:31:51 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:01 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:11 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:21 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:31 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:41 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:32:51 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:01 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:11 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:21 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:31 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:41 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:33:51 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:34:01 max charge current 16A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/21 12:34:11 max charge current 16A: 429 Too Many Requests
```

### What type of operating system are you running?

Linux

andig, 2024-07-21:
Wenn ich jetzt noch wüsste, von welcher Version wir hier überhaupt reden…

andig, 2024-07-21:
Die config ist in jedem Fall falsch, siehe https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/tesla.yaml. Der Parameter heisst `commandProxy`

Neophyte85, 2024-07-21:
Also ich habe doch wirklich oft reingeschrieben das es um die 0.128.3 geht. 

Und die Parameter kann ich doch in der Version gar nicht mehr selbst in die Config schreiben oder etwa doch?

Also es geht um die 0.128.3.

andig, 2024-07-21:
@naltatis hier stimmt etwas nicht im experimental UI, siehe https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/tesla.yaml. Im ui taucht als Parameter `control` auf, aber nicht `commandProxy`?

andig, 2024-07-21:
> Also ich habe doch wirklich oft reingeschrieben

Das Issuetemplate hat ein Feld dafür. Wenns da drin steht muss niemand suchen…

andig, 2024-07-23:
@naltatis das kommt vom vehicle api:

```json
{
    "result": {
        "Template": "tesla",
        "Covers": [
            "tesla-command",
            "tesla-proxy"
        ],
        "Products": [
            {
                "Brand": "Tesla",
                "Description": ""
            }
        ],
        "Params": [
            {
                "Name": "title",
                "Description": "Title",
                "Help": "Will be displayed in the user interface",
                "Type": "String"
            },
            {
                "Name": "icon",
                "Description": "Icon",
                "Help": "Icon as shown in user interface",
                "Advanced": true,
                "Default": "car",
                "Type": "Choice",
                "ValidValues": [
                    "car",
                    "bike",
                    "bus",
                    "moped",
                    "motorcycle",
                    "scooter",
                    "tractor",
                    "shuttle",
                    "van",
                    "heater",
                    "heatpump",
                    "waterheater",
                    "laundry",
                    "dishwasher",
                    "airpurifier",
                    "climate",
                    "cooler",
                    "generic",
                    "device",
                    "meter",
                    "compute",
                    "tool"
                ]
            },
            {
                "Name": "accessToken",
                "Description": "",
                "Help": "",
                "Required": true,
                "Mask": true,
                "Type": "String"
            },
            {
                "Name": "refreshToken",
                "Description": "",
                "Help": "",
                "Required": true,
                "Mask": true,
                "Type": "String"
            },
            {
                "Name": "vin",
                "Description": "Vehicle Identification Number",
                "Help": "Required if you own multiple vehicles of the same brand",
                "Example": "W...",
                "Type": "String"
            },
            {
                "Name": "capacity",
                "Description": "Battery capacity in kWh",
                "Help": "Battery capacity in kWh",
                "Example": "50",
                "Usages": [
                    "vehicle",
                    "battery"
                ],
                "Type": "Float"
            },
            {
                "Name": "phases",
                "Description": "Maximum number of phases",
                "Help": "The maximum number of phases which can be used",
                "Advanced": true,
                "Example": "3",
                "Type": "Number"
            },
            {
                "Name": "control",
                "Description": "",
                "Help": "",
                "Deprecated": true,
                "Type": "String"
            },
            {
                "Name": "commandProxy",
                "Description": "",
                "Help": "When using a TWC3 (or other 'dumb' charger not capable of control), evcc can manage the charge directly by communicating with the vehicle through a Command Proxy. By default, the proxy provided by evcc is used. With this parameter, you set the base URL of a custom Command Proxy to use instead of the default evcc one. See for example https://github.com/wimaha/TeslaBleHttpProxy for a proxy sending commands via bluetooth.",
                "Advanced": true,
                "Default": "https://tesla.evcc.io/",
                "Type": "String"
            },
            {
                "Name": "mode",
                "Description": "Default charging mode when a vehicle is connected",
                "Help": "Possible values are Off, Now, MinPV and PV, or empty if none should be set",
                "Advanced": true,
                "Type": "ChargeModes"
            },
            {
                "Name": "minCurrent",
                "Description": "Minimum amperage (A)",
                "Help": "The minimum amperage per connected phase with which the car should be charged",
                "Advanced": true,
                "Example": "6",
                "Type": "Number"
            },
            {
                "Name": "maxCurrent",
                "Description": "Maximum amperage (A)",
                "Help": "The maximum amperage per connected phase with which the car shuold be charged",
                "Advanced": true,
                "Example": "16",
                "Type": "Number"
            },
            {
                "Name": "identifiers",
                "Description": "Identification",
                "Help": "Mostly this can be added later, see: https://docs.evcc.io/en/docs/features/vehicle",
                "Advanced": true,
                "Type": "StringList"
            },
            {
                "Name": "priority",
                "Description": "Priority",
                "Help": "Priority of the loadpoint or vehicle in relation to other loadpoints or vehicles for allocating pv energy",
                "Advanced": true,
                "Type": "Number"
            }
        ]
    }
}
```

Fehler ist hier
- `"Deprecated": true`, wird angezeigt
- `"Advanced": true`, wird nicht angezeigt

Damit ist es nicht möglich, Tesla über das UI zu konfigurieren.

@Neophyte85 du kannst das aber über die yaml machen.

Neophyte85, 2024-07-23:
Kannst du mir nen Tipp geben wie?
Bekomme jetzt immer Fehlermeldungen sobald ich was unter "vehicles:" eingebe.

vehicles:
commandProxy: http://raspberrypi:8080
ging nicht.

Dann habe ich
vehicles:
- name: my_car
type: template
template: tesla 
title: Model 3
commandProxy: http://raspberrypi:8080
probiert.

Aber kommt:

Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
failed parsing config file: decoding failed due to the following error(s): '' has invalid keys: commandproxy, template, type

andig, 2024-07-23:
Unformatiert nicht lesbar :(. Bei mir geht das wunderbar:

```yaml
vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Model 3
    commandProxy: http://raspberrypi:8080
```

```
evcc vehicle -c tesla-neo.yaml

[main  ] ERROR 2024/07/23 09:17:27 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': missing access and/or refresh token, use `evcc token` to create
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla': missing access and/or refresh token, use `evcc token` to create
Capacity: 0.0kWh
Features: [Offline Retryable]
```

Neophyte85, 2024-07-23:
Vielen Dank, nach dem ich deine Formatierung hier raus kopiert habe kam ich weiter.

Wird nach dem hin und her probieren irgendwas dummes gewesen sein.

[image]

So bekomme ich evcc wieder gestartet. Bekomme aber immer noch den 
[lp-1 ] ERROR 2024/07/23 09:34:37 max charge current 13A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/23 09:34:47 max charge current 13A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/23 09:34:57 max charge current 14A: 429 Too Many Requests
[lp-1 ] ERROR 2024/07/23 09:35:07 max charge current 13A: 429 Too Many Requests

andig, 2024-07-23:
Bitte mal `tesla: trace` und vollständiges Log. Der Ladestrom sollte über BT gesetzt werden, NICHT über HTTP. Der Fehler kann egtl. gar nicht auftreten mit der Config.

Neophyte85, 2024-07-23:
[evcc-20240723-133532-trace.log](https://github.com/user-attachments/files/16348003/evcc-20240723-133532-trace.log)
[evcc-20240723-133640-trace.log](https://github.com/user-attachments/files/16348006/evcc-20240723-133640-trace.log)

Also bei mir ist wirklich nicht auszuschließen dass ich hier irgendetwas falschen machen. Aber bin mir recht sicher, nachdem die Commands der Tesla API beschränkt wurden ging es bei mir ne ganze teile nicht. Dann habe ich mir den TeslaBleHTTPProxy eingerichtet und dann war es zwar holrig, da ich das gefühl hatte immer wenn die 50 Commands aufgebraucht waren musste ich evcc einmal neustartet aber danach wurde meine auto über BLE geregelt und das laden klappte super.

Erst hatte ich auch vihicle Commands installiert aber da war mir die einrichtung in EVCC zu kompliziert. Mit dem TeslaBLEHTTPProxy lief es jetzt aber in der 0.128.2 ganz gut.

andig, 2024-07-23:
Dein evcc steuert ab Start über HTTP, nicht über BT:

> [tesla ] TRACE 2024/07/23 13:34:25 POST https://tesla.evcc.io/api/1/vehicles/LRW3E7EL3PC703169/command/set_charging_amps
[tesla ] TRACE 2024/07/23 13:34:25 {"charging_amps": 11}

Angesichts

> using config file: /etc/evcc.yaml

mach doch mal bitte

    cat /etc/evcc.yaml

Neophyte85, 2024-07-23:
[evcc.yaml.txt](https://github.com/user-attachments/files/16348103/evcc.yaml.txt)

Neophyte85, 2024-07-23:
okay aber ich darf ja die tesla token ja auch nicht entfernt. 
Dachte sobald ich den CommandProxy angebe weis er das er darüber steuern sollte.

andig, 2024-07-23:
Zeig bitte 1:1 die Ausgabe von 

    cat /etc/evcc.yaml

andig, 2024-07-23:
Und bitte:

> interval: 4s # control cycle interval

Bei jedem Programmstart bekommst Du eine Warnung, dass das keine gute Idee ist. Bitte korrigieren.

Neophyte85, 2024-07-23:
[image]
ja okay korrigiere ich aber 30sec ist echt lang :-|

Neophyte85, 2024-07-23:
Ah ich glaube ich habe gefunden was ich noch falsch hatte. 
[image]
vehicle unter loadpoints war noch aus kommentiert. Das musste ich machen damit ich die 0.128.3 zum laufen kriege weil ich ja komischerweise bei mir das Fahrzeug entfernen musste.

andig, 2024-07-23:
Dann machen wir hier mal zu...

andig, 2024-07-23:
Screenshot wg. der Tokens gelöscht. Mir fehlt weiter die Ausgabe (kein Screenshot!) vom `cat`. Das Fahrzeug im Ladepunkt einzutragen ist nicht notwendig.

closed 2024-07-23 by andig

Neophyte85, 2024-07-23:
christian@raspberrypi:~$ cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: dfda0d2e12df53dc1ae4bb7a419fa478ed6499952fe200bb0313fce9c6936074

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.34 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.078 # [currency]/kWh

# trial token, valid until 2024-03-20
sponsortoken: entfernt

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.244
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.244
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
#- type: template
#  template: sungrow-hybrid
#  id: 1
#  host: 192.168.178.244
#  port: 502
#  usage: battery
#  modbus: tcpip
#  name: battery3
- type: template
  template: sungrow-inverter
  id: 1
  host: 192.168.178.245
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4

#- type: template
#  name: hoymiles
#  template: hoymiles-opendtu
#  usage: pv
#  host: 192.168.178.106

chargers:
  - name: my_charger
    type: template
    template: twc3
    host: 192.168.178.232 # IP-Adresse oder Hostname

#chargers:
#- type: template
#  template: twc3
#  host: 192.168.178.232
#  name: wallbox5

#vehicles:
#- name: my_car
#type: template
#template: tesla
#-command
#title: Model 3
#accessToken: 
#refreshToken: 
#vin: LRW3E7EL3PC703169
#capacity: 75
#commandProxy: http://raspberrypi:8080

vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Model 3
    accessToken: 
    refreshToken: 
    vin: LRW3E7EL3PC703169
    commandProxy: http://raspberrypi:8080

loadpoints:
- title: Carport
  charger: my_charger
  mode: pv
  #phases: 1
  #mincurrent: 6
  #maxcurrent: 16
  vehicle: my_car
  #resetOnDisconnect: true

site:
  title: Zu Hause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
 #   battery:
 #   - battery3

Neophyte85, 2024-07-23:
Meinst du so? bekomme die Ausgabe von Cat einfach in meiner Console, habs da jetzt mal rauskopiert und die token entfernt.

Aber jetzt nachdem ich das gesetzt habe macht er es.

andig, 2024-07-23:
Das sieht richtig aus und das Log bestätigt das:

```
[tesla ] TRACE 2024/07/23 14:17:58 POST http://raspberrypi:8080/api/1/vehicles/LRW3E7EL3PC703169/command/set_charging_amps
[tesla ] TRACE 2024/07/23 14:17:58 {"charging_amps": 4}
```

Was auch immer Du vorher getan hast- es war sicher nicht diese Config...

Neophyte85, 2024-07-23:
ich schwöre ich hab da jetzt schon 2 tage ständig dieses ding neugestartet aber schon ne weile nix mehr verändert. 
soll ich vehicle nochmal rausnehmen? lag ohne misst daran. Kanns dir bestimmt beweisen.

andig, 2024-07-27:
> aber vielleicht hilft es ja weiter

Das Log ist unnütz da wieder `tesla: trace` fehlt.