evcc-io/evcc #14952: Tesla BLE Module keeps Car awake after reaching cars SOC Limit in PV Mode

Frahmi12, 2024-07-21:
### Describe the bug

Hello,
I have two Tesla Model Ys and two Tesla TWC3s. The control runs via the new function of the BLE Proxy.
If one Vehicles has reached its Car Charging Limit in PV Mode, then EVCC continues to send Charging Currents to the BLE proxy and keeps the Car awake:
`[tesla ] TRACE 2024/07/21 12:32:43 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/set_charging_amps`

and wakes up the Vehicle via Web API Proxy:
`[tesla] TRACE 2024/07/21 12:15:42 POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN1/wake_up`

On the EVCC homepage, the Charging Process is shown as complete and also that the vehicle SOC limit has been reached.
I have the same behavior when the vehicle has no Charging Limit set and it has reached 100%.

When I charge in fast Mode, EVCC sends wake up and Charge Start Commands
```
[tesla ] TRACE 2024/07/21 12:59:43 POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN1/wake_up 

[tesla ] TRACE 2024/07/21 12:55:39 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/charge_start
```

NOTE: I have made some places unrecognizable in the Log File and the Config.

### Steps to reproduce

1. Charging the car in PV Mode and reaching the car charging limit
2. EVCC continues to send control commands and keeps the vehicle awake

Workaround: Set charging control from PV to Stop. Only then does EVCC stop.

[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: trace

# unique installation id
plant: *****

interval: 30s # control cycle interval

vehicles:
  - name: Marwin
    type: template
    template: tesla
    title: Marwin
    accessToken: *****
    refreshToken: *****
    vin: *****
    capacity: 60
    commandProxy: http://192.168.178.121:8080

  - name: Pia
    type: template
    template: tesla
    title: Pia
    accessToken: *****
    refreshToken: *****
    vin: *****
    capacity: 60
    commandProxy: http://192.168.178.121:8080

[... 123 lines cut ...]
[tesla ] TRACE 2024/07/21 12:27:43 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/set_charging_amps
[tesla ] TRACE 2024/07/21 12:27:43 {"charging_amps": 12}
--
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"VIN1","command":"set_charging_amps"}}
[tesla ] TRACE 2024/07/21 12:28:44 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/set_charging_amps
[tesla ] TRACE 2024/07/21 12:28:44 {"charging_amps": 6}
--
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"VIN1","command":"set_charging_amps"}}
[tesla ] TRACE 2024/07/21 12:31:43 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/set_charging_amps
[tesla ] TRACE 2024/07/21 12:31:43 {"charging_amps": 8}
--
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"VIN1","command":"set_charging_amps"}}
[tesla ] TRACE 2024/07/21 12:32:43 POST http://192.168.178.121:8080/api/1/vehicles/VIN1/command/set_charging_amps
[tesla ] TRACE 2024/07/21 12:32:43 {"charging_amps": 7}
--
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"VIN1","command":"set_charging_amps"}}
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.3

andig, 2024-07-21:
Did you try setting the limit in evcc instead of the vehicle?

Maddimax, 2024-07-21:
The https://github.com/wimaha/TeslaBleHttpProxy proxy automatically tries to wake up the car any time a command fails. It does not use the mechanism you build to return errors about the wake state. The two options are to update the proxy, which I think is not possible as it executes the requested commands async after it already ended the http request, or to change EVCC to not send power changes unless the vehicle is actually charging.

andig, 2024-07-21:
We could change the TWC. But whats the logic? When not charging we still need to enable the charger at some point

andig, 2024-07-21:
> The https://github.com/wimaha/TeslaBleHttpProxy proxy automatically tries to wake up the car any time a command fails.

evcc over HTTP doesn‘t do this and handles wakeup in the loadpoint. Seems this might be needed here, too?

Frahmi12, 2024-07-21:
> Did you try setting the limit in evcc instead of the vehicle? As long as the eallbox is enabled, evcc will update the power envelope.
> 
> /cc @Maddimax seems this needs be handled by the proxy? Not sure how we did this with the http api.

I just tested it.
Vehicle SOC limit to 100% and EVCC charging limit to 85%. Start Charging from 82% to 85% in PV Mode.
EVCC charges up to 85% and stops charging. EVCC then does not send any further control commands.

What strikes me:
When EVCC sets the charging limit, the LP1 log says:
`[lp-1 ] DEBUG 2024/07/21 13:30:42 limitSoc reached: 85.0% > 85%`

This information is missing if the vehicle specifies the charging limit and therefore ends charging automatically. Although the WebGUI shows that the vehicle has reached the vehicle's charging limit and the charging process is complete, the above-mentioned note is missing in the log file.

The question for me is, why does EVCC continue to send control commands when it detects that the vehicle has reached the vehicle's charging limit?
And why is this information missing from the LP1 log? Is there something wrong in the LP1 module?

LP1 Log - Charging Limit sets in EVCC
```
[lp-1  ] INFO 2024/07/21 13:28:42 stop charging <-
[lp-1  ] DEBUG 2024/07/21 13:28:43 !! session: chargedEnergy=2.2 > chargedEnergy=0.0
[lp-1  ] DEBUG 2024/07/21 13:28:43 !! session: start=0.000 stop=0.000 chargedEnergy=2.176
[lp-1  ] DEBUG 2024/07/21 13:28:43 soc estimated: 85.02% (vehicle: 85.00%)
[lp-1  ] DEBUG 2024/07/21 13:28:43 vehicle soc: 85%
[lp-1  ] DEBUG 2024/07/21 13:28:43 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/07/21 13:28:43 vehicle range: 346km
[lp-1  ] DEBUG 2024/07/21 13:28:43 limitSoc reached: 85.0% > 85%
[lp-1  ] DEBUG 2024/07/21 13:29:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 13:29:10 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 13:29:40 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 13:29:40 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 13:29:42 charge voltages: [0 4 0]V
[lp-1  ] DEBUG 2024/07/21 13:29:42 !! session: chargeRater.chargedEnergy=7.2 - chargedAtStartup=5.1
[lp-1  ] DEBUG 2024/07/21 13:29:42 charger status: B
[lp-1  ] DEBUG 2024/07/21 13:29:42 limitSoc reached: 85.0% > 85%
[lp-1  ] DEBUG 2024/07/21 13:30:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 13:30:10 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 13:30:40 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 13:30:40 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 13:30:42 charge voltages: [0 4 0]V
[lp-1  ] DEBUG 2024/07/21 13:30:42 !! session: chargeRater.chargedEnergy=7.2 - chargedAtStartup=5.1
[lp-1  ] DEBUG 2024/07/21 13:30:42 charger status: B
[lp-1  ] DEBUG 2024/07/21 13:30:42 limitSoc reached: 85.0% > 85%
[lp-1  ] DEBUG 2024/07/21 13:31:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 13:31:10 charge currents: [0.2 0 0.4]A
```

LP1 Log - Charging Limit sets in Car
```
[lp-1  ] INFO 2024/07/21 14:14:43 stop charging <-
[lp-1  ] DEBUG 2024/07/21 14:14:43 wake-up timer: start
[lp-1  ] DEBUG 2024/07/21 14:14:43 !! session: chargedEnergy=5.0 > chargedEnergy=2.2
[lp-1  ] DEBUG 2024/07/21 14:14:43 !! session: start=0.000 stop=0.000 chargedEnergy=5.018
[lp-1  ] DEBUG 2024/07/21 14:14:43 vehicle status: B
[lp-1  ] DEBUG 2024/07/21 14:14:43 vehicle soc: 90%
[lp-1  ] DEBUG 2024/07/21 14:14:43 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/07/21 14:14:43 vehicle range: 365km
[lp-1  ] DEBUG 2024/07/21 14:14:43 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/21 14:14:43 pv charge current: 8.52A = 0A + 8.52A (-5882W @ 3p)
[lp-1  ] DEBUG 2024/07/21 14:15:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:15:10 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 14:15:40 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:15:40 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 14:15:45 charge voltages: [0 4 0]V
[lp-1  ] DEBUG 2024/07/21 14:15:45 !! session: chargeRater.chargedEnergy=10.1 - chargedAtStartup=5.1
[lp-1  ] DEBUG 2024/07/21 14:15:45 charger status: B
[lp-1  ] DEBUG 2024/07/21 14:15:45 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/21 14:15:45 pv charge current: 6.66A = 0A + 6.66A (-4595W @ 3p)
[lp-1  ] DEBUG 2024/07/21 14:15:45 max charge current: 6A
[lp-1  ] DEBUG 2024/07/21 14:16:10 charge power: 0W
[... 36 lines cut ...]
[lp-1  ] DEBUG 2024/07/21 14:19:43 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/21 14:19:43 pv charge current: 9.76A = 0A + 9.76A (-6735W @ 3p)
[lp-1  ] DEBUG 2024/07/21 14:20:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:20:10 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 14:20:40 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:20:40 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 14:20:42 charge voltages: [0 4 0]V
[lp-1  ] DEBUG 2024/07/21 14:20:42 !! session: chargeRater.chargedEnergy=10.1 - chargedAtStartup=5.1
[lp-1  ] DEBUG 2024/07/21 14:20:43 charger status: B
[lp-1  ] DEBUG 2024/07/21 14:20:43 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/21 14:20:43 pv charge current: 9.79A = 0A + 9.79A (-6758W @ 3p)
[lp-1  ] DEBUG 2024/07/21 14:20:43 wake-up vehicle, attempts left: 0
[lp-1  ] DEBUG 2024/07/21 14:21:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:21:10 charge currents: [0.2 0 0.4]A
[lp-1  ] DEBUG 2024/07/21 14:21:40 charge power: 0W
[... cut ...]
[lp-1  ] DEBUG 2024/07/21 14:21:42 charge voltages: [0 4 0]V
[lp-1  ] DEBUG 2024/07/21 14:21:42 !! session: chargeRater.chargedEnergy=10.1 - chargedAtStartup=5.1
[lp-1  ] DEBUG 2024/07/21 14:21:42 charger status: B
[lp-1  ] DEBUG 2024/07/21 14:21:42 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/21 14:21:42 pv charge current: 9.45A = 0A + 9.45A (-6518W @ 3p)
[lp-1  ] DEBUG 2024/07/21 14:22:10 charge power: 0W
[lp-1  ] DEBUG 2024/07/21 14:22:10 charge currents: [0.2 0 0.4]A

```

Maddimax, 2024-07-21:
I set the vehicle max soc to 90% to stop charging, interestingly I now see evcc trying to restart the charge. From the log of my personal ble proxy:

```
Jul 21 15:03:00 raspberrypible bash[426]: Starting charge ...
Jul 21 15:03:04 raspberrypible bash[426]: WithCar Error: car could not execute command: complete
```

Maddimax, 2024-07-21:
Even after adding "limitsoc:" to my custom vehicle definition, evcc is still trying to start charging.

andig, 2024-07-21:
Was spräche dagegen, bei schlafendem Auto ErrAsleep zurück zu geben und das Wakeup evcc zu überlassen? So ist es bei Http auch. Der Wakeup auf Proxy-seite ist hier unerwartet.

Frahmi12, 2024-07-21:
Wenn ich das Log richtig verstehe, dann hat EVCC selber das Fahrzeug über den Fleet API Dienst von Tesla geweckt. Dann würde das an der BLE HTTP Proxy Mechanik nichts ändern.
`[tesla] TRACE 2024/07/21 12:15:42 POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN1/wake_up`

Liegt das Problem nicht vielmehr darin, dass EVCC den Ladevorgang, obwohl er laut Web GUI beendet ist, wieder starten möchte und zusätzlich den Ladestrom kontinuierlich anpasst? Das dürfte ja nicht passieren, wenn EVCC den Ziel SOC des Fahrzeugs bereits kennt und auch den aktuellen SOC.

closed 2024-08-02 by github-actions[bot]

andig, 2024-09-15:
Wenn der limit soc bekannt und erreicht ist sollte das nicht mehr passieren. Hilfreich wäre ein Log mit Tesla und custom charger auf trace.

premultiply, 2024-11-27:
There is no signalisation for "finished". SOC 100% does not tell you anything about the energy consumption of the whole vehicle.
It is the car that decided to stop drawing power from the loadpoint. But it can continue at any time if it needs more energy e.g. for pre-conditioning, continue charging etc.

geoffreys, 2024-11-27:
I don’t think that’s what’s happening here. The car isn’t drawing a charge. Instead EVCC is repeatedly telling TesleBLE the new charge rate every time the available solar changes which in turn prevents the vehicle ever sleeping (unless unplugged)

This doesn’t seem right?

andig, 2024-11-29:
evcc is doing what it does with every other charger- adjust the rate as long as the vehicle is connected. It does specifically not intentionally wake the vehicle up. 

Given that BLE connection requires awake vehicle we'd probably need to change this in evcc- not sure we should do this.

geoffreys, 2024-12-01:
Thanks for considering it at least. I'm not familiar with Go but from reading the codebase it would seem like something as simple as changing like below would prevent it changing the amps/keeping vehicle awake when not charging.

core/loadpoint.go line 1861 
`original: err = lp.setLimit(targetCurrent)`
 
to:
```
if lp.enabled && lp.status == api.StatusC {
	err = lp.setLimit(targetCurrent)
}
```

I was going to run this to test impacts locally but it seems like due to not having a TESLA_CLIENT_ID or secret the template for Tesla won't initialise so can't really validate it, and command_proxy doesn't seem supported for Tessie or TeslaMate integrations.

Anyway, hope you could consider implementing this or some parameterised version of it if there's impacts on other vehicle/charges.

andig, 2024-12-01:
Can't do that (at least in this place) since it's also handling on/off there.

andig, 2024-12-01:
> Seeing this same issue on single vehicle (Tesla M3) with TWC and Tesla BLE module. Even after reaching 100% and showing as such in the evcc UI, evcc continues to send set_charging_amp messages to Tesla BLE.

Seems this is only happening when there is enough power. So this seems acceptable, if not perfect?