evcc-io/evcc #14954: UI: Ladeplan „morgen“ wird am Folgetag nicht aktualisiert

An-Wi, 2024-07-21:
### Describe the bug

Nur ein Mini-Bug in der Anzeige - keine funktionale Einschränkung.

In Hauptansicht steht am Folgetag bei „PLAN“ noch „morgen“ statt „heute“:
[image]

Klick man auf die Ladeplanung, dann steht dort im Popup aktualisiert „heute“ hinter dem korrektem Tag:
[image]

### Steps to reproduce

1. Ladeplan am Vortag für Start und Ende am nächsten Tag aktivieren
2. Auf Folgetag warten
3. Hauptansicht mit Ladeplanung vergleichen

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Windows

### Version

0.128.2

naltatis, 2024-07-22:
Heißt, du hast das UI über die Tagesgrenze hinweg offen, richtig? Nach einem Reload sollte das dann korrekt sein. Aber ich versteh das Problem. Die Anzeige müssen wir aktualisieren, auch wenn sich der dahinter liegende Wert nicht ändert.

An-Wi, 2024-07-22:
Jein, die App läuft meist auf meinem iPhone als native Browser-App - dort ist es mir aufgefallen. Nachts war das Hand im Flugzeugmodus - keine Ahnung, ob die App das "überlebt" oder sich reconnected.
Aber deine Aussagen bringt mich auf einige Ideen, das Problem nochmal genauer von unterschiedlichen Seiten/Geräten zu betrachten. Da das Auto eh noch angesteckt ist, habe ich den Ladeplan jetzt nochmal für Morgen früh eingerichtet und gebe anschließend Rückmeldung.

An-Wi, 2024-07-23:
Nun ein klares "Ja". Trotz Flugzeugmodus war das UI auf dem iPhone noch "im Speicher" geblieben, hat sich reconnected und "morgen" weiterhin angezeigt. Nach dem Neustart der App war das Problem in der Hauptansicht verschwunden.

Interessanterweise konnte ich das Problem auf meinem iPad (identisches iOS) nicht reproduzieren. Da wurde die gestern Abend geöffnete App scheinbar komplett "refreshed".

An-Wi, 2024-07-23:
Kleiner Ergänzungswunsch, weil er hier das Thema zufällig tangiert:
Es wäre schön, wenn man einen alten (nicht vollständig erreichten) Ladeplan über die Checkbox "Aktiv" deaktivieren/löschen könnte. Die ist wird leider ausgegraut, wenn der Zielzeitpunkt in der Zukunft liegt:
[image]
Momentan muss man zuerst den Tag anpassen, damit die Checkbox wieder anklickbar ist.

mule1972, 2024-07-23:
> Kleiner Ergänzungswunsch, weil er hier das Thema zufällig tangiert: Es wäre schön, wenn man einen alten (nicht vollständig erreichten) Ladeplan über die Checkbox "Aktiv" deaktivieren/löschen könnte. Die ist wird leider ausgegraut, wenn der Zielzeitpunkt in der Zukunft liegt: ![ScreenShot_9546_](https://private-user-images.githubusercontent.com/139048232/351205536-bf808c07-bc84-4178-8259-c3894237730c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE3MjQwMjUsIm5iZiI6MTcyMTcyMzcyNSwicGF0aCI6Ii8xMzkwNDgyMzIvMzUxMjA1NTM2LWJmODA4YzA3LWJjODQtNDE3OC04MjU5LWMzODk0MjM3NzMwYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyM1QwODM1MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNGY1OWJiMjVkYzY5MzAxM2U2NDdmZjM1YzBkMjUwODVhZDliYmViM2Q4MWY1ZmMxMTQ4ODEwZWM0ZDIxM2JjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1Bjx7veY8UlHKm2aHfmsQdIK-1Jed5EI02mzk09-0AU) Momentan muss man zuerst den Tag anpassen, damit die Checkbox wieder anklickbar ist.

Ergänzend hierzu: Vorallem versucht EVCC dann auch wenn man bspw. um 18Uhr nach Hause kommt, den eigentlich schon seit morgens abgelaufenen Ladeplan weiter zu erfüllen und lädt so auch mit Netzstrom trotz Einstellung PV.

kauTzinho, 2024-07-23:
Habe das Verhalten heute auch beobachtet.

Ich hatte gestern für heute um 12 Uhr einen Ladeplan für 100% SoC erstellt. Gegen 10 Uhr war ich kurz unterwegs und habe das Fahrzeug anschließend wieder angesteckt. Der Ladeplan wurde fortgesetzt. Das war ok. Um 12 Uhr war der SoC bei 98% und ich bin zum geplanten Ziel gefahren. 

Als ich um 17 Uhr zurück kam wurde der Ladeplan weiterhin ausgeführt, scheinbar weil die 100% nicht erreicht wurden. EVCC hätte jetzt das Auto mit Netzstrom nochmal voll gepumpt.

Evtl. wäre es besser wenn ein Ladeplan der in der Vergangenheit liegt nicht mehr ausgeführt wird.

mule1972, 2024-07-23:
> Habe das Verhalten heute auch beobachtet.
> Evtl. wäre es besser wenn ein Ladeplan der in der Vergangenheit liegt nicht mehr ausgeführt wird.

Ja, dies wäre auch mein Vorschlag. @naltatis: Besser ein neues Issue öffnen?

naltatis, 2024-07-24:
Ja, das Thema "Umgang mit Ladeplänen in der Vergangenheit" ist definitiv ein anderes Issue.

closed 2024-07-27 by andig

An-Wi, 2024-07-29:
Funktioniert nun prima mit Version 0.129.0 👍
Danke.