evcc-io/evcc #14957: EVCC started nach Update auf 128.3 nicht mehr - status=2/INVALIDARGUMENT

BachmannErik, 2024-07-21:
### Describe the bug

Hallo, 
nach dem Update auf 128.3 startet EVCC nicht mehr. im Logfile sehe ich nur, dass der Service immer wieder gestartet wird. Woran kann das liegen ? 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 03214988b5917ecb8575cc641b984d625565a33566dc0b3b8de226794a71f93a

interval: 10s # control cycle interval

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.34  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.34  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.60  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv4

chargers:
- type: template
  template: openwb 
  host: 192.168.178.22  
  name: wallbox5

loadpoints:
- title: OpenWB
  charger: wallbox5
  mode: minpv

site:
  title: Mein Zuhause
  meters:
    grid:
    pv:
    - pv1
    - pv4
    battery:
    - battery2

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3864 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.10 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: meine "07407" # PLZ mit führender Null mit in "" setzen

influx:
  url: http://localhost:8086
  database: EVCC
  token: 5-GHvqdZH2xx-80cq9dwSCKn6AIgGq5qH8rzaQCM0PPyC5d1hwE6JJRRr-9poDkQZmO1f_5aq48ksUbIjzSynw==
  org: EVCC

telemetry: true # default: false

sponsortoken:
```

### Log details

```text
evcc.service: Scheduled restart job, restart counter is at 136.
Jul 21 17:49:11 raspberrypi systemd[1]: Stopped evcc.
Jul 21 17:49:11 raspberrypi systemd[1]: Started evcc.
Jul 21 17:49:12 raspberrypi evcc[4588]: [main  ] INFO 2024/07/21 17:49:12 evcc 0.128.3
Jul 21 17:49:12 raspberrypi evcc[4588]: [main  ] INFO 2024/07/21 17:49:12 using config file: /etc/evcc.yaml
Jul 21 17:49:12 raspberrypi evcc[4588]: [db    ] INFO 2024/07/21 17:49:12 using sqlite database: /var/lib/evcc/evcc.db
Jul 21 17:49:12 raspberrypi evcc[4588]: [main  ] INFO 2024/07/21 17:49:12 listening at :7070
Jul 21 17:49:14 raspberrypi evcc[4588]: [openwb] INFO 2024/07/21 17:49:14 connecting evcc-645887714 at tcp://192.168.178.22:1883
Jul 21 17:49:14 raspberrypi evcc[4588]: [openwb] DEBUG 2024/07/21 17:49:14 tcp://192.168.178.22:1883 connected
Jul 21 17:49:14 raspberrypi evcc[4588]: [openwb] DEBUG 2024/07/21 17:49:14 tcp://192.168.178.22:1883 subscribe openWB/system/Timestamp
Jul 21 17:49:16 raspberrypi evcc[4588]: panic: invalid kind: ptr
Jul 21 17:49:16 raspberrypi evcc[4588]: goroutine 1 [running]:
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/server/db/settings.DecodeOtherSliceOrMap({0x2382560, 0x400192a1d0}, {0x23431a0, 0x400058bf10})
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/server/db/settings/setting.go:203 +0x200
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/server/db/settings.Yaml({0x2a13cfb, 0x7}, {0x2382560, 0x400192a1d0}, {0x23431a0, 0x400058bf10})
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/server/db/settings/setting.go:226 +0x280
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/cmd.configureTariffs({{0x4000e6dbb8, 0x3}, {{0x4000e6dbf0, 0x5}, 0x4000440d80}, {{0x4000e6dc54, 0x5}, 0x4000461140}, {{0x4000e6dcb0, 0xf}, ...}, ...})
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/cmd/setup.go:808 +0xe8
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x59563e0)
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/cmd/setup.go:909 +0x190
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/cmd.runRoot(0x5954cc0, {0x2a13d25?, 0x7?, 0x2a06cf3?})
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/cmd/root.go:182 +0x91c
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/spf13/cobra.(*Command).execute(0x5954cc0, {0x400004c0a0, 0x0, 0x0})
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/spf13/cobra@v1.8.1/command.go:989 +0x828
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/spf13/cobra.(*Command).ExecuteC(0x5954cc0)
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/spf13/cobra.(*Command).Execute(...)
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/spf13/cobra@v1.8.1/command.go:1041
Jul 21 17:49:16 raspberrypi evcc[4588]: github.com/evcc-io/evcc/cmd.Execute()
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
Jul 21 17:49:16 raspberrypi evcc[4588]: main.main()
Jul 21 17:49:16 raspberrypi evcc[4588]: 	github.com/evcc-io/evcc/main.go:44 +0x38
Jul 21 17:49:16 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.3

andig, 2024-07-21:
Könntest du deine DB und Yaml bitte an info@evcc.io schicken?

andig, 2024-07-21:
Startet die 0.128.2 noch?

StefanSchoof, 2024-07-21:
Steht in deiner CO2 Config wirklich "meine" oder hast du das für das Issue geändert?

```
  co2:
    type: grünstromindex
    zip: meine "07407"
```

andig, 2024-07-22:
@BachmannErik could you please run

    evcc settings get tariffs

Make sure to use the same database as the crash (if this is the correct one):

    EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db evcc settings get tariffs

andig, 2024-07-22:
Der Fehler hier ist vmtl, dass die YAML Konfiguration einfach falsch ist und nicht dekodiert werden kann. Das darf natürlich kein Panic ergeben.

BachmannErik, 2024-07-22:
> evcc settings get tariffs

das ist die Rückmeldung auf den Befehl.

tariffs: # currency: EUR

# grid: # price using energy from the grid
#   type: fixed
#   price: 0.294 # EUR/kWh

# feedin: # price for feeding solar energy to the grid
#   type: fixed
#   price: 0.08 # EUR/kWh

# co2: # carbon intensity forecast data
#   type: grünstromindex
#   zip: 07407

andig, 2024-07-22:
Kannst Du das bitte formatieren? Die Struktur ist nicht erkennbar.

BachmannErik, 2024-07-22:
Hallo, ich hatte in meiner Konfiguration vor dem Update auf 128.3 zwei Autos konfiguriert, die erkannt wurden und zu Ladevorgänge aufgezeichnet wurden. Nach dem Update auf 128.3 funktioniert es nicht mehr und die Autos sind auch nicht mehr in der YAML enthalten. Jetzt nach dem Update auf 128.4 kommt der Fehler das die Konfiguration falsch ist und der Server startet nicht. Wie bekomme ich jetzt den Initialen Zustand wieder hergestellt ? Oder was ist der Weg um die Konfiguration wieder neu aufzubauen ?

andig, 2024-07-22:
Finally closed in https://github.com/evcc-io/evcc/commit/cbb1319ebc012c871da14f669d1cb4c70c327eab

andig, 2024-07-22:
Bitte morgen auf Nightly umstellen. Alternativ Migration zurück drehen:

    evcc  migrate --reset

closed 2024-07-22 by andig

BachmannErik, 2024-07-22:
ich habe die Einträge im Dialog Home/Konfiguration nachgezogen. Landen diese Einträge in der EVCC.db und von dort in der YAML? oder ist die YAML Konfiguration überflüssig? Was ist die Empfehlung zur Pflege der Konfiguration?

andig, 2024-07-22:
UI Config landet in der DB und ist ein experimentelles Feature an dem wir arbeiten. Für Tests sind wir dankbar!

michelmalle, 2024-07-22:
> UI Config landet in der DB und ist ein experimentelles Feature an dem wir arbeiten. Für Tests sind wir dankbar!

Hallo @andig. Ich habe auch noch einen Bug gefunden, welcher in der Experimental UI auftritt und die App nach anschließendem Reboot nicht mehr starten lässt.

Fehler tritt ein, sobald man in der UI die Tarife anpasst. 
Lösung war, das Tariffs Key-Value Paar aus der SQLite DB zu löschen.

andig, 2024-07-22:
Mit der Beschreibung kann ich nichts anfangen. Gibts ein Logfile?

michelmalle, 2024-07-22:
> Mit der Beschreibung kann ich nichts anfangen. Gibts ein Logfile?

Ok, hier nochmal der Ablauf beschrieben, wie gesagt, das ist reproduzierbar:

1. Browse web based configuration
2. Enable Experimental Features
3. Go to configuration > Tariffs
4. Change a value, for e.g. 0.294 to 0.293
5. Save & restart
6. EVCC doesn’t start up

Hat eine gewisse Ähnlichkeit zum hier gefixten Issue Report oben. Hier das Log am Macbook (evcc via Brew).

`main  ] INFO 2024/07/22 21:10:32 evcc 0.128.3
[main  ] INFO 2024/07/22 21:10:32 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/22 21:10:32 using sqlite database: /Users/michel/.evcc/evcc.db
[mqtt  ] INFO 2024/07/22 21:10:33 connecting evcc at tcp://192.168.2.10:1883
[mqtt  ] DEBUG 2024/07/22 21:10:33 tcp://192.168.2.10:1883 connected
[eebus ] INFO 2024/07/22 21:10:33 Local SKI:  376799c48b6b82c0837d9e9cbbec253927cf5ad5
[eebus ] DEBUG 2024/07/22 21:10:33 starting websocket server on :4712
[eebus ] DEBUG 2024/07/22 21:10:33 mdns: announce
[eebus ] DEBUG 2024/07/22 21:10:33 mdns: using zeroconf
[eebus ] DEBUG 2024/07/22 21:10:33 mdns: start search
[main  ] INFO 2024/07/22 21:10:33 listening at :7070
[eebus ] DEBUG 2024/07/22 21:10:37 ski: 775b95f4539ec53dc2b4057321beb319bc2a4c18 name: Elli-Wallbox-2045A11VNC brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2045A11VNC register: false host: wallbox-2045A11VNC.local. port: 4712 addresses: [192.168.2.195]
[eebus ] DEBUG 2024/07/22 21:10:37 trying to connect to 775b95f4539ec53dc2b4057321beb319bc2a4c18 at wallbox-2045A11VNC.local.
[eebus ] DEBUG 2024/07/22 21:10:37 initiating connection to 775b95f4539ec53dc2b4057321beb319bc2a4c18 at wallbox-2045A11VNC.local.:4712/ship/
[eebus ] DEBUG 2024/07/22 21:10:37 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/07/22 21:10:37 Recv: d:_i:47859_Elli-Wallbox-2045A11VNC:NodeManagement to NodeManagement read 24 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC reply 2 24 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/07/22 21:10:37 Recv: d:_i:47859_Elli-Wallbox-2045A11VNC:NodeManagement to NodeManagement reply 26 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 6 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 5 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 7 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 10 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 9 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 8 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 11 NodeManagementBindingRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 12 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 13 MeasurementDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 14 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 16 ElectricalConnectionDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 17 MeasurementConstraintsListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 18 DeviceConfigurationKeyValueDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 19 ElectricalConnectionParameterDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 20 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 15 LoadControlLimitDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 21 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 operation is not supported on function loadControlLimitConstraintsListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 22 MeasurementDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 23 MeasurementConstraintsListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 24 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 25 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 26 ElectricalConnectionParameterDescriptionListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC read 27 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2024/07/22 21:10:37 Send: d:_i:47859_Elli-Wallbox-2045A11VNC call 28 NodeManagementSubscriptionRequestCall
[... cut ...]
github.com/spf13/cobra.(*Command).execute(0x106399540, {0x140001181f0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0x828
github.com/spf13/cobra.(*Command).ExecuteC(0x106399540)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38`

Die Lösung, diesen Eintrag in der SQLite DB löschen:
[image]

andig, 2024-07-22:
> Hat eine gewisse Ähnlichkeit zum hier gefixten Issue Report oben

Na dann- mal den Fix probiert?!

michelmalle, 2024-07-26:
> > Hat eine gewisse Ähnlichkeit zum hier gefixten Issue Report oben
> 
> Na dann- mal den Fix probiert?!

Problem tritt auch nach Upgrade auf 0.128.4 auf, mit dem Unterschied, dass EVCC nun hochfährt, im GUI einen Fehler meldet:

Error during startup. Check your configuration and restart.
yaml: line 3: mapping values are not allowed in this context

...analoge Meldung findet sich auch im Logfile. 

Lösung: Tariffs Eintrag aus SQlite DB löschen oder  `evcc migrate --reset` ausführen,