evcc-io/evcc #14958: ELLI PRO lädt im Schnell Modus nur mit max 6 kW

daklemo, 2024-07-21:
### Describe the bug

Hallo,

ich habe gerade mit Version 0.128.3 im Schnell Modus das Problem, dass die Elli Pro WB nur mit max 6 kW lädt.
Der Anschluss ist 3-phasig, max A pro Phase ist 16 A eingestellt, in der WB ist auch kein Lastmanagement aktiv.

Keine Ahnung ob das schon länger so ist, Schnell Modus wird selten verwendet.

Elli Charger Pro, 1 Gen, aktuelle SW.

Woran kanns liegen?

VG Klemo

### Steps to reproduce

1. starte im Modus schnell
2.
3.
...

### Configuration details

```yaml
--
```

### Log details

```text
[site ] DEBUG 2024/07/21 17:16:56 ----
[lp-1 ] DEBUG 2024/07/21 17:16:56 charge power: 5973W
[lp-1 ] DEBUG 2024/07/21 17:16:56 charge currents: [8.78 8.54 8.65]A
[site ] DEBUG 2024/07/21 17:16:56 pv power: 2589W
[site ] DEBUG 2024/07/21 17:16:56 battery soc: 98%
[site ] DEBUG 2024/07/21 17:16:56 battery power: 493W
[site ] DEBUG 2024/07/21 17:16:57 grid meter: 3211W
[site ] DEBUG 2024/07/21 17:16:57 site power: 3904W
[lp-1 ] DEBUG 2024/07/21 17:16:57 detected active phases: 3p
[lp-1 ] DEBUG 2024/07/21 17:16:57 !! session: chargeRater.chargedEnergy=5.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/21 17:16:57 charger status: C
[eebus ] DEBUG 2024/07/21 17:16:58 Send: d:_i:47859_Elli-Wallbox-2123A1EMTI notify 953 DeviceDiagnosisHeartbeatData
[lp-1 ] DEBUG 2024/07/21 17:16:58 vehicle status: C
[lp-1 ] DEBUG 2024/07/21 17:16:58 vehicle soc: 16%
[lp-1 ] DEBUG 2024/07/21 17:16:58 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/07/21 17:16:58 vehicle range: 88km
```

### What type of operating system are you running?

Linux

### Version

0.128.3

daklemo, 2024-07-21:
Gelbe Lampe an der WB leuchtet: WB reduziert die Lade-Leistung wg Übertemperatur
kein Evcc Issue

closed 2024-07-21 by daklemo

Sumpfdotter, 2024-07-29:
Eine Frage, weil ich seit wenigen Tagen auch solches ein Problem reproduzierbar habe: Welches Leuchtmuster ist es genau? Mit der gelben Lampe rechts sagt die WB: "evcc in seiner Rolle als HEMS hat die WB angewiesen, die Leistung zu reduzieren."

Ich habe seit einigen Tagen das Problem, dass in der Kombi trotz "Schnell" Modus nur mit Minmalleistung geladen wird. Workaround ist ein einfacher manueller eingriff: Einmal den Modus per evcc-Frontend verändern (z.B. auf "Aus" oder PV") und zurück auf "Schnell", dann geht er auf die volle Power.

Ich muss jetzt die Tage mal rausfinden, ob es mit einer neuen evcc-Version (benutze ein Nighlty) gekommen ist, oder weil ich einen externen Zähler ergänzt habe.

daklemo, 2024-07-30:
In dem Fall war es die LED links oben: statt grün war sie gelb. Lt. Doku der Elli WB "Einschränkung Übertemperatur"

Passt, die Sonne hat bei 35 Grad frontal auf die WB gebrannt, als der Schatten kam ging die Ladeleistung wieder nach oben

andig, 2024-08-18:
https://github.com/evcc-io/evcc/releases/tag/0.130.0 contains a refactor of the EEbus charger that might or might not help here.

daklemo, 2024-08-18:
> https://github.com/evcc-io/evcc/releases/tag/0.130.0 contains a refactor of the EEbus charger that might or might not help here.

I'll check after I returned from my vacation.
I will seitch from WLAN to Ethernet and add an external meter. Thought about Shelly Pro 3EM (allready use one for my heatpump) - unless you have another suggestion.

Due to my company fleet driver management I will stay on Elli charger