evcc-io/evcc #14962: OCPP panic: runtime error: invalid memory address or nil pointer dereference

jeffborg, 2024-07-21:
### Describe the bug

Updated from 0.128.2 to 0.128.3, now always crashes upon occp communication, working perfectly before on 0.128.2

### Steps to reproduce

1. run evcc 0.128.3 docker on arm64
2. occp connects
3. crash

### Configuration details

```yaml
interval: 30s
site:
  - title: Home
    circuit: main
    meters:
      grid: grid
      pv:
        - pv
circuits:
- name: main
  maxPower: 15000
  meter: grid
log: info
levels:
  site: debug
  lp-1: debug
  ocpp: trace
  cache: error
  db: error
meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      topic: shellies/house-energy-monitor/emeter/0/power
      timeout: 30s
      scale: 1
    energy:
      source: mqtt
      topic: shellies/house-energy-monitor/emeter/0/total
      timeout: 30s
      scale: 0.001
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: energy/solar
      timeout: 30s
      scale: 1
      jq: .OutputPower
    energy:
      source: mqtt
      topic: energy/solar
      timeout: 30s
      jq: .TotalGenerateEnergy
chargers:
  - name: solax
    type: template
    template: ocpp
    timeout: 2m
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
    meterinterval: 10s
loadpoints:
  - title: Garage
    charger: solax
    vehicle: zapdos
    circuit: main
    mode: pv
    enable:
      delay: 1m
      threshold: -700
    disable:
      delay: 3m
      threshold: 1000
mqtt:
  broker: mqtt-mosquitto
vehicles:
  - name: zapdos
    type: template
    template: teslamate
    id: 1
    capacity: 56
tariffs:
  currency: AUD
  grid:
    type: template
    template: amber
    token: XXXXXXXXXXXXXXXXXXXXXX
    siteid: XXXXXXXXXXXXXXXXXXXXX
    channel: general
  feedin:
    type: custom
    price:
      source: mqtt
      topic: amber/feedin
      timeout: 30s
  co2:
    type: custom
    price:
      source: mqtt
      topic: amber/co2
      timeout: 30s
```

### Log details

```text
[main  ] INFO 2024/07/22 09:22:07 evcc 0.128.3
[main  ] INFO 2024/07/22 09:22:07 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/07/22 09:22:07 connecting evcc-884314637 at tcp://mqtt-mosquitto:1883
[main  ] INFO 2024/07/22 09:22:07 listening at :7070
[ocpp  ] INFO 2024/07/22 09:22:22 charge point connected, registering: SN10052307XXXXXX
[ocpp  ] TRACE 2024/07/22 09:22:22 send SN10052307XXXXXX: [2,"233495459","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/07/22 09:22:24 recv SN10052307XXXXXX: [2,"7365","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307XXXXXX\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/07/22 09:22:24 recv SN10052307XXXXXX: [3,"233495459",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/22 09:22:24 recv SN10052307XXXXXX: [2,"7366","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-07-21T23:22:22Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"241.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"26495.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":7}]
[ocpp  ] TRACE 2024/07/22 09:22:24 send SN10052307XXXXXX: [3,"7365",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/22 09:22:24 send SN10052307XXXXXX: [2,"345696950","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2024/07/22 09:22:24 TriggerMessage MeterValues for SN10052307XXXXXX: Accepted
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x127abf0]

goroutine 85 [running]:
github.com/evcc-io/evcc/charger/ocpp.(*Connector).MeterValues(0x4000d361b0, 0x400101d4a0)
	github.com/evcc-io/evcc/charger/ocpp/connector_core.go:56 +0xe0
github.com/evcc-io/evcc/charger/ocpp.(*CP).MeterValues(0x0?, 0x400101d4a0)
	github.com/evcc-io/evcc/charger/ocpp/cp_core.go:91 +0x38
github.com/evcc-io/evcc/charger/ocpp.(*CS).OnMeterValues(0x0?, {0x40006409c5?, 0x0?}, 0x400101d4a0)
	github.com/evcc-io/evcc/charger/ocpp/cs_core.go:91 +0x40
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest.func1()
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:523 +0x27c
created by github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest in goroutine 70
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:512 +0x48c
```

### What type of operating system are you running?

Docker container

### Version

0.128.3

jeffborg, 2024-07-21:
My guess going by the code is conn.status is nil at https://github.com/evcc-io/evcc/blame/0.128.3/charger/ocpp/connector_core.go#L56 because I'm not using getconfiguration and instead using this

```
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
```

andig, 2024-07-22:
It crashes when receiving the MeterValues. The line is somewhat impossible unless the request itself is nil (but that should crash a line above):

```go
	if request.TransactionId != nil && conn.txnId == 0 {
		conn.log.DEBUG.Printf("hijacking transaction: %d", *request.TransactionId)
		conn.txnId = *request.TransactionId
	}
```

Any chance to start the debugger and see what's going on? Please also try current nightly so we get the line numbers right.

closed 2024-07-22 by andig

premultiply, 2024-07-22:
```yaml
chargers:
  - name: solax
    type: template
    template: ocpp
    timeout: 2m
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
    meterinterval: 10s
```

@jeffborg 

What is the reason for this manual configuration here, especially `getconfiguration: false` and `bootnotification: false`?

What type of charger is this exactly?
Could you provide an ocpp trace log without these parameters or both set to true?
My goal is to avoid the need of this parameters if possible and to provide templates even for ocpp chargers.

Did you try the nightly build and setting `autostart: true`?

andig, 2024-07-22:
Issue is already fixed. Cannot access status when its nil.