evcc-io/evcc #14969: keba-modbus': decoding failed due to the following error(s): '' has invalid keys: id, rtu, uri

TobiasHuber1980, 2024-07-22:
### Describe the bug

Nach Update auf 128.4 kommt der Fehler
```[main ] FATAL 2024/07/22 10:39:30 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'keba-modbus': decoding failed due to the following error(s): '' has invalid keys: id, rtu, uri```

### Steps to reproduce

1. update auf 0.128.4
2. nach Downgrade auf 0.128.3 kein Fehler

[evcc-20240722-104407-trace.log](https://github.com/user-attachments/files/16330375/evcc-20240722-104407-trace.log)

...

### Configuration details

```yaml
##################################

site:
  - title: expert Traunreut
    circuit: main
    meters:
      grid: grid
      pv:
        - pv
        
##################################     
   
circuits:
- name: main
  title: Main Circuit
  maxPower: 35000
  #maxCurrent: 100
  meter: grid
  parent:
- name: loadpoints
  #maxCurrent: 120
  parent: main        

##################################

meters:
  - name: grid
    type: template
    template: plexlog
    usage: grid
    host: 192.168.1.26
    port: 503
  - name: pv
    type: template
    template: plexlog
    usage: pv
    host: 192.168.1.26
    port: 503

##################################

chargers:
#- type: template
#  template: keba-modbus 
#  host: 192.168.1.xx
#  name: wallbox1

- type: template
  template: keba-modbus
  host: 192.168.1.29
  name: wallbox2
  modbus: tcpip
  id: 255
  port: 502

- type: template
  template: keba-modbus
  host: 192.168.1.31
  name: wallbox3
  modbus: tcpip
  id: 255
  port: 502

- type: template
  template: keba-modbus
  host: 192.168.1.33
  name: wallbox4
  modbus: tcpip
  id: 255
  port: 502

##################################

loadpoints:
#- title: expert links
#  charger: wallbox1
#  circuit: loadpoints
#  mode: minpv
#  disable:
#    delay: 10m

- title: expert rechts
  charger: wallbox2
  circuit: loadpoints
  mode: minpv
  disable:
    delay: 10m

- title: Fahrschule links
  charger: wallbox3
  circuit: loadpoints
  mode: minpv
  disable:
    delay: 10m

- title: Fahrschule rechts
  charger: wallbox4
  circuit: loadpoints
  mode: minpv
  disable:
    delay: 10m

##################################
```

### Log details

```text
[main ] FATAL 2024/07/22 10:39:30 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'keba-modbus': decoding failed due to the following error(s): '' has invalid keys: id, rtu, uri
[main ] FATAL 2024/07/22 10:39:30 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

0.128.4

Nyarly360, 2024-07-22:
+1, ich hab dasselbe Problem.

andig, 2024-07-22:
Closed in https://github.com/evcc-io/evcc/commit/881c4a36417c154b1e185ce259cc595e90cbfe29

closed 2024-07-22 by andig

githuballrutz, 2024-07-22:
> +1, ich hab dasselbe Problem.

bei mir ebenfalls, also erstmal wieder downgraden?

TobiasHuber1980, 2024-07-22:
> Warum wurde der Eintrag geschlossen? Der Fehler ist immer noch da.

weil der fehler im nächsten nightly / master behoben ist.

stuppeliene, 2024-07-22:
Dankeschön für die Erklärung.

ahackner, 2024-07-23:
@andig wann kann man das Update dann in Homeassistant laden? Bei mir klappt es auch nicht mehr leider :/

TobiasHuber1980, 2024-07-23:
Du kannst ja erst mal auf 0.128.3 zurückgehen....

Matze5593, 2024-07-25:
> Du kannst ja erst mal auf 0.128.3 zurückgehen....

Und wie macht man das in home Assistent?

TobiasHuber1980, 2024-07-25:
> > Du kannst ja erst mal auf 0.128.3 zurückgehen....
> 
> Und wie macht man das in home Assistent?

https://github.com/evcc-io/evcc/discussions/5209#discussioncomment-4183505