evcc-io/evcc #14973: trial token not valid

arnematschke, 2024-07-22:
### Describe the bug

I'm not shure how to write a bug report correctly, so here is a short description,

using evcc as an home assistant addon
addon is not starting up
log shows: sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired

### Steps to reproduce

1. get token from; [https://docs.evcc.io/docs/sponsorship](url)
2. set token in evcc.yaml
3. error at startup: sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired

### Configuration details

```yaml
# trial token, valid until 2024-07-28
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjIxNTM2MDAsImlhdCI6MTcxOTU2MTYwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.81Bd8lvqVYGTOQi0vVQAK9Q6k8aruoD1ic3_88in9Kw
```

### Log details

```text
will provide this when I'm back home
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.128.4

andig, 2024-07-22:
Will be updated

closed 2024-07-22 by andig

naltatis, 2024-07-24:
Token is updated.

arnematschke, 2024-07-24:
Had to reinstall the addon to get it working

andig, 2024-07-24:
That should and is never the case.

Flirina, 2024-08-28:
the trial token is again not working, although stated as valid until 31.08.2024

# trial token, valid until 2024-08-31
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjUwOTQ4MDAsImlhdCI6MTcyNDIzMDgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.61RAc8c7vXkgX_INZgM5Cjbq9nta6XR-7Aodmle6gQ8

[main  ] INFO 2024/08/28 13:48:10 evcc 0.130.6
[main  ] INFO 2024/08/28 13:48:10 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/28 13:48:11 listening at :7070
[main  ] FATAL 2024/08/28 13:48:11 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/08/28 13:48:11 will attempt restart in: 15m0s

andig, 2024-08-28:
The token will always expire. We we renew it in due time. Note that the trial token is not meant for production use.