evcc-io/evcc #14984: Eastron SDM72D-M does not work

dersch81, 2024-07-22:
### Describe the bug

I'm trying to talk via a Waveshare RS485 to Modbus TCP Gateway to my Eastron SDM72D-M meter

But i just get timeouts.

[image]

### Steps to reproduce

Follow the config as in the wiki

### Configuration details

```yaml
- type: template
  template: eastron-sdm72
  id: 1
  host: 10.0.204.3
  port: 502
  usage: charge
  modbus: rs485tcpip
  name: charge6
```
```

### Log details

```text
charge6
-------
Power:  read failed: read tcp 10.0.204.2:57790->10.0.204.3:502: i/o timeout
Energy: read failed: read tcp 10.0.204.2:51210->10.0.204.3:502: i/o timeout

```
```

### What type of operating system are you running?

Linux

### Version

evcc 0.128.4

premultiply, 2024-07-22:
According to your converter config:
`modbus: tcpip`

closed 2024-07-22 by premultiply

dersch81, 2024-07-22:
WOW thanks. Wasn't able to get this by myself. It is working now.

So could you also change the wiki?

https://docs.evcc.io/en/docs/devices/meters#sdm72d-m

premultiply, 2024-07-22:
Just configure your converter the right way and there were no such trouble.
But you set it to translate Modbus RTU to Modbus TCP but you told evcc to talk Modbus RTU to you converter...

dersch81, 2024-07-22:
yes you are right. So to understand. If i wouldn't set the Waveshare to RTO to TCP it would work with my previous setting?