evcc-io/evcc #14995: Switch charging session when vehicle identifier changes

andig, 2024-07-23:
See https://github.com/evcc-io/evcc/discussions/14992. 

Currently this is only being done when we see an actual status change to disconnected. Instead, we could check if the vehicle identifier changes. Need to investigate which id changes should trigger new session.

premultiply, 2024-07-23:
Possible common indicators for intermediate disconnect could be:

- `Status` changed (`A` is the ultimate indicator) 
- `ChargedEnergy` jumps back
- `ChargingTime` jumps back
- `Identifier` changed (be carefull, may change even if session is still running by just swiping any other card. Depends on charger firmware!)
- ...?

tbc

andig, 2024-07-23:
> could evcc not check once in a while which vehicle actually shows connected

It does.

sizzos, 2025-03-05:
> This issue ist _not_ about vehicle identification by vehicle API, but about vehicle identification by charger. Doing via API is not possible as API calls are scarce and can _only_ be triggered if we identify vehicle disconnect by charger.

I experience this as a "bug" currently. Not sure if I should open a new issue. 
Attached two screenshots - one export from wallbox, one from evcc. The "unser Passat" car is identified by rfid token "blau" while the standard car at this loadpoint is "Hyundai Ioniq 5".
The values read from the wallbox export are correct (Passat only has a 10,x kWh battery) while evcc attributes more of the charged power to that session - I assume because of a "fast switch" - both cars were parked and unplugging/plugging takes me 5 seconds.

[image]

I my explanation doesn't make sense - please let me know.

andig, 2025-03-05:
Does this add any new information?

sizzos, 2025-03-05:
> Does this add any new information?

I'd say yes because it shows that my charger is able to identify both cars correctly.

andig, 2025-03-06:
PR zum Test. Da wir die Fahrzeug-APIs dafür nicht nutzen können passiert das nur wenn die Wallbox einen Identifier zurück gibt.