evcc-io/evcc #14996: Easee: Phase Mode set incorrectly for switching phases with multi charger circuit

GrimmiMeloni, 2024-07-23:
### Describe the bug

_**Note: This entire scenario only applies to multiple Easee Chargers on a single circuit. I am specifically disucssing the related code path - this does not cover the simple single charger scenario in which we do Circuit Level control.**_

While looking into #14387 I noticed that the logic for switching phases when multiple chargers are connected to a single circuit looks incorrect. It was introduced many moons ago in #1701.

Specifically this part looks incorrect to me:

https://github.com/evcc-io/evcc/blob/e94cc98058be6ccb01378c50fe63c74caaa42a35/charger/easee.go#L783-L785

According to the [API docs from Easee](https://developer.easee.com/docs/enumerations#phasemode-38), 2 indeed does not mean 3 phases, but automatic phase switching. So what this would mean, is that whenever we are trying to perform a switch to 3 phases, we are actually not switching phases, but simply telling the charger to go to "auto" mode. The outcome is not clear to me (maybe the chargers will use EaseeLink to agree on switching phases?) - but either way we are not getting the evcc intended behavior (i.e. switching phases immediately on that charger).

@naltatis & @andig - since you built all this back in the day, can you chime in here? The error seems so obvious as the docs  from Easee are super clear. I am wondering if there is more to this than it seems at first sight...?

### Steps to reproduce

1. Have at least 2 Easee chargers on a single circuit
2. Have a charger currently using 1 phase
3. Have that charger in automatic  phase switching mode
4. Try to perform phase switch to 3 phases, it becomes a No Op, and charger keeps using 1p.

### Configuration details

```yaml
N/A
```

### Log details

```text
see example shared in https://github.com/evcc-io/evcc/issues/14387#issuecomment-2210638931
```

### What type of operating system are you running?

Docker container

### Version

master

andig, 2024-07-23:
> logic for switching phases when multiple chargers 

Kann ich nicht beurteilen wir häufig das vorkommt...

GrimmiMeloni, 2024-07-23:
> > logic for switching phases when multiple chargers
> 
> Kann ich nicht beurteilen wir häufig das vorkommt...

Das weiß ich auch nicht. Aber sind wir zumindest einig, daß die aktuelle Implementierung nicht das tut, was sie eigentlich soll (gezielt die Phasen umschalten).

andig, 2024-07-23:
Keine Ahnung, aber das wäre ja sehr schnell ausprobiert. Ich kann (leider) auch nicht mehr nachvollziehen, warum wir den Pfad über den Circuit überhaupt jemals genutzt haben?

GrimmiMeloni, 2024-07-24:
> Ich kann (leider) auch nicht mehr nachvollziehen, warum wir den alternativen Pfad über den Circuit überhaupt jemals genutzt haben?

Ich _vermute_ flash wear, aber ganz sicher bin ich nicht. Der Weg über den Circuit (wie wir Ihn für einzelne Charger verwenden) wird in ein "Dynamic" Feld geschrieben, welches für regelmäßige Updates gedacht ist. 

In der Doku steht etwas zum Phase Limiting im Bereich [Loadbalancing](https://developer.easee.com/docs/load-balancing). Dort findet sich in einem Nebensatz folgendes:

```Setting phase 3 to ~0 A will also prevent the any chargers on this circuit from doing 3 phase charging.```

Das heißt, über den Circuit kann man nur alle Charger am Circuit gleichzeitig die Phasen schalten lassen. Wenn man allerdings nur 1 Charger hat, scheint dies der "goldene" Weg zu sein.

Der zweite (hier betrachtete) Weg über die Charger Settings ist gezielter. Hat aber nach meinem Verständnis den Nachteil, daß Flashwear ein Thema werden kann, da die Settings nicht als Dynamic deklariert sind. Auf der anderen Seite findet sich im OCPP Bereich der Doku auch ein [Abschnitt zum PhaseMode locking](https://developer.easee.com/docs/ocpp-smart-charging#lockconnectorphasemode). Ob man dies jetzt als Aufforderung verstehen kann häufiger diesen Wert zu setzen?

Ich fasse mal zusammen.

Es gibt 2 Wege die Phasen zu schalten:
* Auf dem Circuit, über `DynamicCircuitCurrent`, das wirkt auf alle Charger (am Circuit). Dieser Wert kann nur vom Master Charger gelesen/geschrieben werden.
* Auf dem Charger, über `ChargerSettings -> phaseMode`

Soweit passt unsere Implementierung. Das initial beschriebene Problem aber bleibt - weil wir z.B. den Fall #14387 haben wo ein Charger auf PhaseMode 2 (auto) steht, und einen Charger mit 1p am laufen hat. In so einer Situation hilft es nicht wenn wir 3p wollen erneut auf "auto" zu schalten. Ich denke es macht Sinn hier explizit `PhaseMode` auf 3 zu setzen.

GrimmiMeloni, 2024-07-26:
> Was spricht denn dagegen das einfach lokal mal zu tun?

Gar nichts. Ich habe gerade mal direkt mit der API gesprochen. PhaseMode auf der Settings API wird identisch gemappt wie für die Events dokumentiert (1->1p, 3->3p, 2-> auto). Die App hat auch noch etwas beschreibenden Text.

[image]

Ich werde nachher mal mit dem Fahrzeug den Step-Up von 1p auf (1) auf Auto (2) durch spielen.

MaTr74, 2024-08-04:
Hallo! Gibt es irgendwelche Erkenntnisse inzwischen?

GrimmiMeloni, 2024-08-04:
Mangels Zeit leider noch nicht. 😞

andig, 2024-08-04:
Die Änderung wäre ja überschaubar- könntest du evtl ein Testrelease ausprobieren (Docker)?

MaTr74, 2024-08-04:
Kann ich gern machen, habe nur leider keinen Einfluss darauf, wann wieder geladen wird. In der letzten Woche hatten wir nur einen einzigen Ladevorgang.

GrimmiMeloni, 2024-08-08:
> Ich werde nachher mal mit dem Fahrzeug den Step-Up von 1p auf (1) auf Auto (2) durch spielen.

Ich habe das vorhin hier lokal durchgespielt. Um über den PhaseMode von 1p auf 3p zu schalten, reicht der "auto" PhaseMode (2) aus. Hier ist also erstmal nichts falsch.

closed 2024-08-08 by GrimmiMeloni

andig, 2024-08-08:
Vielen Dank!