evcc-io/evcc #14997: Wrong charging current shown using OCPP

hddnPrdtr, 2024-07-23:
### Describe the bug

Setup: ecvv 128.2, Wallbox pulsar+ using OCPP. 
Once the charging process is almost completed and the vehicle reduces the current below what would be permitted via OCPP the actual charging current in the evcc frontend is not (!) the actual current but the max current that would be permitted due to the available solar power.

### Steps to reproduce

1. start Charging process via OCPP
2. load vehicle up to 100%
3. 
4.
...

### Configuration details

```yaml
meters:
- type: template
  template: e3dc-rscp 
  usage: grid  
  host: xxx  
  port: 5033  
  user: xxxx  
  password: xxx  
  key: xxx  
  name: grid1
- type: template
  template: e3dc-rscp 
  usage: pv  
  host: xxx  
  port: 5033  
  user: chris@staudt.in  
  password: xxx  
  key: xxx  
  name: pv2
- type: template
  template: e3dc-rscp 
  usage: battery  
  host: xxx
  port: 5033  
  user: xxx  
  password: xxx  
  key: OOOOOO  
  name: battery3

chargers:
- type: template
  template: pulsarplus 
  timeout: 2m  
  name: wallbox4

loadpoints:
- title: Rechts
  charger: wallbox4
  mode: pv

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  grid:
    type: template
    template: tibber
    token: xxx
[... 485 lines cut ...]
Jul 22 16:58:57 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:58:57 battery power: 12W
Jul 22 16:58:57 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:58:57 grid meter: -12565W
Jul 22 16:58:57 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:58:57 site power: -12553W
Jul 22 16:58:57 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:58:57 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jul 22 16:58:57 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:58:57 !! session: chargeRater.chargedEnergy=29.2 - chargedAtStartup=0.0
Jul 22 16:58:57 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:58:57 charger status: A
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 ----
Jul 22 16:59:27 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:59:27 charge power: 0W
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 pv power: 14376W
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 battery soc: 98%
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 battery power: 12W
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 grid meter: -12501W
Jul 22 16:59:27 NUC evcc[16477]: [site  ] DEBUG 2024/07/22 16:59:27 site power: -12489W
Jul 22 16:59:27 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:59:27 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jul 22 16:59:27 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:59:27 !! session: chargeRater.chargedEnergy=29.2 - chargedAtStartup=0.0
Jul 22 16:59:27 NUC evcc[16477]: [lp-1  ] DEBUG 2024/07/22 16:59:27 charger status: A
```

### What type of operating system are you running?

Linux

### Version

128.2

andig, 2024-07-23:
> Once the charging process is almost completed and the vehicle reduces the current below what would be permitted via OCPP the actual charging current in the evcc frontend is not (!) the actual current but the max current that would be permitted due to the available solar power.

If your charger does not have a meter (or does not send meter values) only the given current can be displayed.

premultiply, 2024-07-23:
Your charger does not push power metering data on its own without request:
```
Jul 22 14:15:27 NUC evcc[16477]: [lp-1  ] INFO 2024/07/22 14:15:27 loadpoint 1:
Jul 22 14:15:27 NUC evcc[16477]: [lp-1  ] INFO 2024/07/22 14:15:27   mode:        pv
Jul 22 14:15:27 NUC evcc[16477]: [lp-1  ] INFO 2024/07/22 14:15:27   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
Jul 22 14:15:27 NUC evcc[16477]: [lp-1  ] INFO 2024/07/22 14:15:27   meters:      charge ✗
```

You can try to add
```yaml
  metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import,Voltage
  meterinterval: 10
```
to your charger configuration.

Autodetection of metering values will be added by https://github.com/evcc-io/evcc/pull/14888

If you could include an occp trace log from your charger boot and getconfiguration calls it will be visible if the charger supports all of the requested metering data and we could include it in the template until autodetection is included.

closed 2024-07-23 by andig