evcc-io/evcc #15016: Override Climater settings (force climater on or off)

pauxus, 2024-07-23:
**Is your feature request related to a problem? Please describe.**
Renault API is bad, often it is way behind the reality. This effects things like vehicle recognition, but this can be worked around. However, it also effects the climater. Current situation: Renault API thinks that the climater is running for the last two days straight. Since evcc takes that at face value, when returning home this evening and plugging in, it happily started charging the car and draining my battery. Only current workaround: completely turn off charging, but in that case I have to remember to switch it on again tomorrow.

The same happened yesterday. 

(this effects the Renault app as well, which shows the climater to be active as well) 

**Describe the solution you'd like**
Simplest solution: provide a flag to turn of climating for certain vehicles. I can live with manually switching to min+PV when starting the climater.

**Describe alternatives you've considered**
The alternatives would be switching to guest vehicle or creating an offline vehicle. Both cases would not allow viewing the current SOC or setting max

andig, 2024-07-23:
We can just remove it. Do you want to provide a PR?

closed 2024-07-27 by andig

maatinh, 2024-09-03:
Same for me with latest NB - v.0.130.7. evcc recognizes a climater being active which is not true and thus using battery to charge the car. Would really prefer to be able to override/disable the climater recognition.

andig, 2024-09-03:
Nothing has changed. See https://github.com/evcc-io/evcc/issues/15016#issuecomment-2246178868. And- as usual- log files are king 🤴

maatinh, 2024-09-03:
https://github.com/evcc-io/evcc/pull/15757#issuecomment-2314472112

Ich hatte das mit 0.130 schon angemerkt, dass sich das Verhalten geändert hatte. So ist evcc für mich leider fast wertlos, weil weder das Laden beim erreichen der Ladegrenze gestoppt wird noch das Limit der Hausbatterie berücksichtigt wird.