evcc-io/evcc #15017: evcc not starting after update to 0.128.4 with keba wallbox

alexduerl, 2024-07-23:
### Describe the bug

After update to 0.128.4 my evcc is hangin in start loop. 
Log shows the following error:

[main  ] FATAL 2024/07/23 21:50:45 cannot create charger 'keba': cannot create charger type 'template': cannot create charger type 'keba-modbus': decoding failed due to the following error(s):
'' has invalid keys: id, rtu, uri
[main  ] FATAL 2024/07/23 21:50:45 will attempt restart in: 15m0s

### Steps to reproduce

1. Configure keba charger in evcc.yaml
2. Update to 0.128.4
3. Try to start evcc

### Configuration details

```yaml
chargers:
  - name: keba
    type: template
    template: keba-modbus
    modbus: tcpip
    id: 255
    host: 192.168.178.78 # Hostname
    port: 502 # Port 

loadpoints:
  - title: Carport # display name for UI
    charger: keba # Wallbox Referenz
    vehicle: 500e # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    soc:
      poll:
        mode: always
        interval: 5m
      estimate: false
    maxcurrent: 10
    
site:
  title: Home # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - my_pv # list of pv inverters/ meters
    battery: # (batteries = veraltet)
      - my_battery # battery meter reference
  residualPower: 200
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/07/23 21:50:35 evcc 0.128.4
[main  ] INFO 2024/07/23 21:50:35 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/23 21:50:35 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/07/23 21:50:35 connecting evcc-1270808091 at tcp://localhost:1883
[main  ] INFO 2024/07/23 21:50:35 listening at :7070
[main  ] FATAL 2024/07/23 21:50:45 cannot create charger 'keba': cannot create charger type 'template': cannot create charger type 'keba-modbus': decoding failed due to the following error(s):

'' has invalid keys: id, rtu, uri
[main  ] FATAL 2024/07/23 21:50:45 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

0.128.4

TobiasHuber1980, 2024-07-23:
Siehe https://github.com/evcc-io/evcc/issues/14969
Du musst dir entweder die aktuelle Nightly installieren oder zurück zu 0.128.3

alexduerl, 2024-07-23:
sorry, das hatte ich nicht gesehen.
Bin zurück auf 0.128.3. Jetzt läuft es wieder.
Danke

closed 2024-07-23 by alexduerl