evcc-io/evcc #15022: Add eProwallbox /Move to chargers

chitufl, 2024-07-24:
If possible, could the eProwallbox / Move be added as charger to the supported devices? Seems to be working ok with default OCPP profile but would be better to have them as standalone. Happy to support with logs. 
Product manuals here: https://www.esolutions.free2move.com/eu/en_it/information-hub/

premultiply, 2024-07-24:
Yes, shure.

Please provide an ocpp tracelog from the start of evcc including the boot and getconfiguration responses of the charger for reference.

chitufl, 2024-07-24:
let me know if this enough:

main  ] INFO 2024/07/24 20:35:28 evcc 0.128.4
[main  ] INFO 2024/07/24 20:35:28 using config file: c:\evcc\evcc.yaml
[db    ] INFO 2024/07/24 20:35:28 using sqlite database: C:\Users\Florin\.evcc\evcc.db
[main  ] INFO 2024/07/24 20:35:28 listening at :7070
[ocpp-1] DEBUG 2024/07/24 20:35:28 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/07/24 20:35:47 charge point connected, registering: 0744AB070003433
[ocpp  ] TRACE 2024/07/24 20:35:47 recv 0744AB070003433: [2,"8ad6bec3-4c1e-4a53-a835-fe633b89bd4f","BootNotification",{"chargePointModel":"F2ME.EPROSE01TXX","chargePointSerialNumber":"0744AB070003433","chargePointVendor":"AB070","firmwareVersion":"2.9.3"}]
[ocpp  ] TRACE 2024/07/24 20:35:47 send 0744AB070003433: [3,"8ad6bec3-4c1e-4a53-a835-fe633b89bd4f",{"currentTime":"2024-07-24T18:35:47Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/07/24 20:35:47 recv 0744AB070003433: [2,"38e1b145-a2d2-482a-a5a3-09c0f762ceb9","MeterValues",{"connectorId":0,"meterValue":[{"sampledValue":[{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Energy.Active.Import.Register","unit":"Wh","value":"13813"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Energy.Reactive.Import.Register","unit":"varh","value":"272"},{"context":"Sample.Clock","format":"Raw","location":"Body","measurand":"Temperature","unit":"Celsius","value":"20"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L1","unit":"V","value":"242"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L2","unit":"V","value":"0"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L3","unit":"V","value":"1"}],"timestamp":"2024-07-24T06:00:00Z"}]}]
[ocpp  ] TRACE 2024/07/24 20:35:48 send 0744AB070003433: [2,"3245209120","GetConfiguration",{}]
[ocpp  ] TRACE 2024/07/24 20:35:48 recv 0744AB070003433: [3,"3245209120",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown, 1.RST"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"35"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register, Energy.Reactive.Import.Register, Temperature, Voltage.L1, Voltage.L2, Voltage.L3"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register, Energy.Active.Import.Interval, Power.Active.Import, Temperature"},{"key":"MeterValueSampleInterval","readonly":false,"value":"150"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"3"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core, FirmwareManagement, LocalAuthListManagement, SmartCharging"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"5"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"2"},{"key":"LightIntensity","readonly":false,"value":"1"},{"key":"MaxEnergyOnInvalidId","readonly":false,"value":"0"},{"key":"WebSocketPingInterval","readonly":false,"value":"20"},{"key":"ACTIVATION_DATE","readonly":true,"value":"2024-06-29T15:08:18Z"},{"key":"APN_NAME","readonly":true,"value":" "},{"key":"APN_SET","readonly":false,"value":""},{"key":"APN_USER","readonly":true,"value":""},{"key":"C2G_ENABLED","readonly":false,"value":"false"},{"key":"CS_URL","readonly":false,"value":"ws://192.168.0.19:8887/"},{"key":"ClockAlignedAdditionalDataInterval","readonly":false,"value":"900"},{"key":"DPM_ENABLED","readonly":false,"value":"true"},{"key":"DPM_PMAX","readonly":false,"value":"10000"},{"key":"DPM_TYPE","readonly":false,"value":"5"},{"key":"ENABLE_RANDOMIZED_DELAY","readonly":false,"value":"false"},{"key":"GUI_STANDBY_TIMEOUT","readonly":false,"value":"0"},{"key":"HWCONFIG_e","readonly":true,"value":"5"},{"key":"LOAD_UNBALANCE_CURRENT","readonly":false,"value":"32000"},{"key":"LOAD_UNBALANCE_ENABLED","readonly":false,"value":"false"},{"key":"MAX_RANDOMIZED_DELAY","readonly":false,"value":"600"},{"key":"MID_ENABLED","readonly":false,"value":"false"},{"key":"MID_TYPE","readonly":false,"value":"3"},{"key":"MODBUS_BAUD_RATE","readonly":false,"value":"4"},{"key":"MODBUS_RTU_ADDRESS","readonly":false,"value":"1"},{"key":"MS_MODE","readonly":false,"value":"0"},{"key":"MeterValueSampleAdditionalInterval","readonly":false,"value":"150"},{"key":"OPERATING_MODE","readonly":false,"value":"3"},{"key":"RFID_LIST","readonly":false,"value":"[]"},{"key":"SIM_ICCID","readonly":true,"value":""},{"key":"SIM_PIN","readonly":false,"value":""},{"key":"SIM_STATUS","readonl [... cut ...] mple.Clock","format":"Raw","location":"Body","measurand":"Temperature","unit":"Celsius","value":"20"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L1","unit":"V","value":"242"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L2","unit":"V","value":"0"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L3","unit":"V","value":"1"}],"timestamp":"2024-07-24T06:00:00Z"}]}]

premultiply, 2024-08-06:
```json
{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core, FirmwareManagement, LocalAuthListManagement, SmartCharging"}
```

Your charger lacks support for RemoteTrigger feature profile.

Tomorrows nightly build includes a very improved OCPP support.

In addition to many other fixes and changes it does not try to use trigger commands if they are not supported by the charger indicated by their `SupportedFeatureProfiles` configuration.
This error will disappear.

It would be best if you could try the nightly build tomorrow and provide the output of `evcc charger --diagnose` here.

chitufl, 2024-08-18:
@premultiply 
I ran version 130, problem is that the cmd window is self closing after running the start up routine.
result of diag prompt below
c:\evcc>evcc charger --diagnose
[main  ] INFO 2024/08/18 16:16:24 evcc 0.130.0
[main  ] INFO 2024/08/18 16:16:24 using config file: c:\evcc\evcc.yaml
[db    ] INFO 2024/08/18 16:16:24 using sqlite database: C:\Users\Florin\.evcc\evcc.db
[ocpp-1] DEBUG 2024/08/18 16:16:24 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/18 16:16:36 charge point connected, registering: 0744AB070003433
[ocpp  ] TRACE 2024/08/18 16:16:36 recv 0744AB070003433: [2,"3481e2e0-4985-45b1-b0cf-18476e73b7ef","StatusNotification",{"connectorId":0,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2024/08/18 16:16:36 send 0744AB070003433: [3,"3481e2e0-4985-45b1-b0cf-18476e73b7ef",{}]
[ocpp  ] TRACE 2024/08/18 16:16:36 recv 0744AB070003433: [2,"28a89cf7-6c57-4a97-ae3f-bcf48114da73","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2024/08/18 16:16:36 send 0744AB070003433: [3,"28a89cf7-6c57-4a97-ae3f-bcf48114da73",{}]
[ocpp  ] TRACE 2024/08/18 16:16:36 recv 0744AB070003433: [2,"6fc0cc6d-b179-4d63-adf6-6b825659e7e9","DataTransfer",{"data":"[{\"MID\":{\"sampledValue\":[{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Current.Export\",\"unit\":\"A\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Current.Import\",\"unit\":\"A\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Energy.Active.Export.Register\",\"unit\":\"Wh\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Energy.Active.Import.Register\",\"unit\":\"Wh\",\"value\":\"327500\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Energy.Reactive.Export.Register\",\"unit\":\"varh\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Energy.Reactive.Import.Register\",\"unit\":\"varh\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Frequency\",\"value\":\"50\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Power.Active.Export\",\"unit\":\"W\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Power.Active.Import\",\"unit\":\"W\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Voltage\",\"phase\":\"L1\",\"unit\":\"V\",\"value\":\"244\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Voltage\",\"phase\":\"L2\",\"unit\":\"V\",\"value\":\"0\"},{\"context\":\"Sample.Clock\",\"format\":\"Raw\",\"location\":\"Inlet\",\"measurand\":\"Voltage\",\"phase\":\"L3\",\"unit\":\"V\",\"value\":\"0\"}],\"timestamp\":\"2024-08-18T14:15:00Z\"}}]","messageId":"MID_DPM_Measures","vendorId":"F2MeS_ePro"}]
[ocpp  ] TRACE 2024/08/18 16:16:36 send 0744AB070003433: [3,"6fc0cc6d-b179-4d63-adf6-6b825659e7e9",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/18 16:16:36 recv 0744AB070003433: [2,"d5ce714e-d5a4-4fbb-a70d-18e4df69621a","MeterValues",{"connectorId":0,"meterValue":[{"sampledValue":[{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Energy.Active.Import.Register","unit":"Wh","value":"117147"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Energy.Reactive.Import.Register","unit":"varh","value":"681"},{"context":"Sample.Clock","format":"Raw","location":"Body","measurand":"Temperature","unit":"Celsius","value":"37"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L1","unit":"V","value":"244"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L2","unit":"V","value":"0"},{"context":"Sample.Clock","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L3","unit":"V","value":"1"}],"timestamp":"2024-08-18T14:15:01Z"}]}]
[ocpp  ] TRACE 2024/08/18 16:16:37 send 0744AB070003433: [2,"20155390","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/18 16:16:37 recv 0744AB070003433: [3,"20155390",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown, 1.RST"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"35"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register, Energy.Reactive.Import.Register, Temperature, Voltage.L1, Voltage.L2, Voltage.L3"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"3"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInval [... cut ...]
[ocpp  ] TRACE 2024/08/18 16:16:44 recv 0744AB070003433: [3,"970531014",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/18 16:16:44 send 0744AB070003433: [2,"4256524427","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2024/08/18 16:16:45 recv 0744AB070003433: [3,"4256524427",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/18 16:16:45 send 0744AB070003433: [2,"3850656645","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"30"}]
[ocpp  ] TRACE 2024/08/18 16:16:46 recv 0744AB070003433: [3,"3850656645",{"status":"Accepted"}]

chitufl, 2024-08-21:
[image]

Driving me nuts ...

premultiply, 2024-08-21:
Try the current nightly.

chitufl, 2024-08-23:
no ideea how to run the current nigthly on a windows system... :(

chitufl, 2024-08-30:
No more errors but the charge starts in full mode regardless of the selected option. 
[image]

andig, 2024-09-05:
Is there anything left to do here? What?

andig, 2024-09-05:
> Seems to be working ok with default OCPP profile but would be better to have them as standalone.

If ocpp works we can close there. Thank you!

chitufl, 2024-09-06:
@andig Still not working
Either I get "expect disable got enabled" or the charging starts in full mode and cannot be stopped from evcc...

andig, 2024-09-06:
As usual: please provide trace log, closing until available.

chitufl, 2024-09-09:
[main  ] INFO 2024/09/09 13:38:57 evcc 0.130.8
[main  ] INFO 2024/09/09 13:38:57 using config file: c:\evcc\evcc.yaml
[main  ] FATAL 2024/09/09 13:39:37 cannot create charger 'wallbox': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/09/09 13:39:37 will attempt restart in: 15m0s

same error with 1.130.7

andig, 2024-09-09:
See post above. Logfile? If thats all on trace level then your charger never connects to evcc in the first place…

chitufl, 2024-09-14:
Hello @andig 
Something happened with EVCC between version 0.130.6 and 0.130.8, see below the snap with different result of the 2 versions, with the same yaml config file...
[image]

andig, 2024-09-14:
Please try latest release.

chitufl, 2024-09-16:
same result
[image]

andig, 2024-09-16:
That's really unexpected now. Please add trace log.

chitufl, 2024-09-16:
that's all I get:

c:\evcc>evcc -log
[main  ] INFO 2024/09/16 18:22:09 evcc 0.130.11
[main  ] INFO 2024/09/16 18:22:09 using config file: c:\evcc\evcc.yaml
[ocpp  ] ERROR 2024/09/16 18:22:40 error replying cp 0744AB070003433 to request 1990b35d-3ddb-47b5-8f2f-b4ed7c610c4b: Key: 'CallResult.Payload.IdTagInfo' Error:Field validation for 'IdTagInfo' failed on the 'required' tag
[ocpp  ] ERROR 2024/09/16 18:22:46 error replying cp 0744AB070003433 to request 1990b35d-3ddb-47b5-8f2f-b4ed7c610c4b: Key: 'CallResult.Payload.IdTagInfo' Error:Field validation for 'IdTagInfo' failed on the 'required' tag
[main  ] FATAL 2024/09/16 18:23:17 cannot create charger 'wallbox': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/09/16 18:23:17 will attempt restart in: 15m0s

andig, 2024-09-16:
`trace` log bitte!

chitufl, 2024-09-16:
lost me after "bitte", my german speaking day are to far in the past  :))

c:\evcc>evcc --log trace
[main  ] INFO 2024/09/16 21:01:08 evcc 0.130.11
[main  ] INFO 2024/09/16 21:01:08 using config file: c:\evcc\evcc.yaml
[db    ] INFO 2024/09/16 21:01:08 using sqlite database: C:\Users\Florin\.evcc\evcc.db
[db    ] TRACE 2024/09/16 21:01:08 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/16 21:01:08 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/16 21:01:08 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/16 21:01:08 SELECT * FROM `settings` 14 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] INFO 2024/09/16 21:01:09 listening at :7070
[db    ] TRACE 2024/09/16 21:01:09 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db    ] TRACE 2024/09/16 21:01:09 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[ocpp-1] DEBUG 2024/09/16 21:01:09 waiting for chargepoint: 1m0s
[ocpp  ] INFO 2024/09/16 21:01:13 charge point connected, registering: 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 enqueued CALL [2250219927, ChangeAvailability] for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 dispatched request 2250219927 for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent JSON message to 0744AB070003433: [2,"2250219927","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/09/16 21:01:13 started timeout timer for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 received JSON message from 0744AB070003433: [2,"28a608ad-8e22-459f-9e17-0c2c02257fec","StatusNotification",{"connectorId":0,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2024/09/16 21:01:13 handling incoming CALL [28a608ad-8e22-459f-9e17-0c2c02257fec, StatusNotification] from 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent CALL RESULT [28a608ad-8e22-459f-9e17-0c2c02257fec] for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent JSON message to 0744AB070003433: [3,"28a608ad-8e22-459f-9e17-0c2c02257fec",{}]
[ocpp  ] TRACE 2024/09/16 21:01:13 received JSON message from 0744AB070003433: [2,"74c3f92e-1b8b-43c6-a562-302494a8552e","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"","status":"SuspendedEV","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2024/09/16 21:01:13 handling incoming CALL [74c3f92e-1b8b-43c6-a562-302494a8552e, StatusNotification] from 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent CALL RESULT [74c3f92e-1b8b-43c6-a562-302494a8552e] for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent JSON message to 0744AB070003433: [3,"74c3f92e-1b8b-43c6-a562-302494a8552e",{}]
[ocpp  ] TRACE 2024/09/16 21:01:13 received JSON message from 0744AB070003433: [2,"7047e335-c28c-4447-8cae-4dd92b2dc754","MeterValues",{"connectorId":1,"meterValue":[{"sampledValue":[{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Current.Offered","unit":"A","value":"0"},{"context":"Sample.Periodic","format":"Raw","location":"Inlet","measurand":"Energy.Active.Import.Register","unit":"Wh","value":"365761"},{"context":"Sample.Periodic","format":"Raw","location":"Inlet","measurand":"Power.Active.Import","unit":"W","value":"0"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Power.Offered","unit":"W","value":"0"}],"timestamp":"2024-09-16T18:45:56Z"}],"transactionId":1726503730}]
[ocpp  ] TRACE 2024/09/16 21:01:13 handling incoming CALL [7047e335-c28c-4447-8cae-4dd92b2dc754, MeterValues] from 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 received JSON message from 0744AB070003433: [3,"2250219927",{"status":"Rejected"}]
[ocpp  ] TRACE 2024/09/16 21:01:13 sent CALL RESULT [7047e335-c28c-4447-8cae-4dd92b2dc754] for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent JSON message to 0744AB070003433: [3,"7047e335-c28c-4447-8cae-4dd92b2dc754",{}]
[ocpp  ] TRACE 2024/09/16 21:01:13 handling incoming CALL RESULT [2250219927] from 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 completed request 2250219927 for 0744AB070003433
[ocpp-1] DEBUG 2024/09/16 21:01:13 failed configuring availability: Rejected
[ocpp  ] TRACE 2024/09/16 21:01:13 0744AB070003433 ready to transmit again
[ocpp  ] TRACE 2024/09/16 21:01:13 dispatched request 3734163909 for 0744AB070003433
[ocpp  ] TRACE 2024/09/16 21:01:13 sent JSON message to 0744AB070003433: [2,"3734163909","GetConfiguration",{}]
[ocpp  ] TRACE 2024/09/16 21:01:13 timeout canceled for 0744AB070003433
[... cut ...] tlet","measurand":"Current.Offered","unit":"A","value":"0"},{"context":"Sample.Periodic","format":"Raw","location":"Inlet","measurand":"Energy.Active.Import.Register","unit":"Wh","value":"365761"},{"context":"Sample.Periodic","format":"Raw","location":"Inlet","measurand":"Power.Active.Import","unit":"W","value":"0"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Power.Offered","unit":"W","value":"0"}],"timestamp":"2024-09-16T18:46:26Z"}],"transactionId":1726503730}]

andig, 2024-09-17:
Afaikt the last log does not show any error. Tbo, I'm lost what to look at. Problem solved? If not we need a trace log of the problem.

mfuchs1984, 2024-12-31:
@andig ich habe mir die logs hier und in https://github.com/evcc-io/evcc/discussions/17920 angeschaut und soweit ich glaube verstanden, was das Problem ist.
1. Die Wallbox kann, wie @premultiply weiter oben schon festgestellt hat, kein RemoteTrigger
2. Evcc startet und empfängt nach der ChangeAvailability Nachricht initial eine StatusNotification
3. CP.Setup läuft durch
4. Danach wird in NewOCPP der connector instanziiert ocpp.NewConnector
5. Am Ende wird conn.Initialized() aufgerufen
6. Da dann keine StatusNotification mehr empfangen wird, triggert diese Funktion nach einem timeout eine StatusNotification, diese wird erwartungsgemäß mit "Not implemented" abgelehnt
7. Timeout

Zusammenfassung:
Der Status aus Anfangs empfangenen StatusNotifications wird nicht gespeichert, somit haben die Konnektoren erstmal keinen Status. Wenn die Wallbox kein RemoteTrigger beherrscht, läuft evcc in ein timeout, da ohne Statusänderung von alleine keine kommt.
Optionen, die ich auf den ersten Blick sehe: 
1. Initialen Status speichern / cachen und an connector übergeben (etwas ähnliches hast du wohl mal in https://github.com/evcc-io/evcc/pull/16908 versucht, um https://github.com/evcc-io/evcc/issues/15677 zu lösen, aber so wie ich sehe nur für charger mit RemoteTrigger https://github.com/evcc-io/evcc/blob/50f135005ff389752c0d9e520fdaeb330b3a7e6b/charger/ocpp/connector.go#L53)
2.  wenn kein RemoteTrigger unterstützt, dann nach Connector Instanziierung andere message schicken, die StatusNotifications triggert (Wie ChangeAvailability in diesem Fall)

Weitere Auffälligkeiten
Während CP.Setup vor dem Triggern einer StatusNotification mit cp.HasRemoteTriggerFeature prüft, ob RemoteTrigger unterstützt wird, ist das in https://github.com/evcc-io/evcc/blob/a558aa2d3ab9664b1ab00e2aa378430c0ddc4dfd/charger/ocpp/connector.go#L98 und https://github.com/evcc-io/evcc/blob/a558aa2d3ab9664b1ab00e2aa378430c0ddc4dfd/charger/ocpp/connector.go#L92 nicht der Fall,.

andig, 2025-01-08:
> Da dann keine StatusNotification mehr empfangen wird

That sounds like the problem I've already failed to solve in https://github.com/evcc-io/evcc/pull/16908.

mfuchs1984, 2025-01-08:
As far as I can see, your last attempt was never tried by @benesolar, the last comment is a question, see https://github.com/evcc-io/evcc/issues/15677#issuecomment-2476793968

There is one important difference, in the original case, the wallbox fails to respond, here, we have a wall box not supporting RemoteTrigger. To support such chargers, I think your pull request is necessary but needs a small change to support them, see my last comment.
I can try to have another look at the issue #15677 and your PR #16908 if you want, I just cannot promise when exactly.

chitufl, 2025-01-09:
@andig @mfuchs1984 
Thanks for looking again at this. On my side I've dropped the idea of using surplus to charge the cars, but I'm ready to support by testing, could be useful for other in the future if a solution is found.

benesolar, 2025-01-09:
I am also ready for testing again.
Would need a nightly because i don`t know how to build myself.

haraldfunk, 2025-01-20:
Facing the same issue with my new eProWallbox that i wanted to connect via evcc. Any chance to fix this? Would be happy to test on my setup.

mfuchs1984, 2025-01-20:
Is one of you able to build and run evcc on your computer?

haraldfunk, 2025-01-20:
I‘m not a Go or Node developer but have a lot of experience in Python and C/C++, so I guess it should be possible for me to get it build and running. I will try the next days and give some feedback.

haraldfunk, 2025-01-21:
@mfuchs1984: Habe Evcc gebaut bekommen und läuft sauber. Was kann ich machen, um den Fehler einzugrenzen und eventuell zu beheben?

mfuchs1984, 2025-01-25:
Schon getestet?

haraldfunk, 2025-01-26:
Ich wollte es jetzt am Wochenende testen. Leider ist jedoch der Testtoken auf der Website gestern abgelaufen, so dass ich aktuell nicht testen kann...

mfuchs1984, 2025-01-26:
Das Token ist aktualisiert.

haraldfunk, 2025-01-26:
Hat zumindest auf den ersten Blick keine Verbesserung gebracht: 

`[main  ] INFO 2025/01/26 16:21:22 evcc 0.133.0 (7c5bd4b4)
[main  ] INFO 2025/01/26 16:21:22 using config file: /workspaces/evcc/evcc.yaml
[main  ] INFO 2025/01/26 16:21:22 listening at :7070
[2592AB030003214-1] DEBUG 2025/01/26 16:21:22 waiting for chargepoint: 2m0s
[ocpp  ] DEBUG 2025/01/26 16:21:31 charge point connected: 2592AB030003214
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [2,"647068650","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [2,"a598044b-df18-4479-bdff-bd1594466696","StatusNotification",{"connectorId":0,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [3,"a598044b-df18-4479-bdff-bd1594466696",{}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [2,"55fa9cfb-ff85-4ecd-83cb-57753b61d583","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [3,"55fa9cfb-ff85-4ecd-83cb-57753b61d583",{}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [3,"647068650",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [2,"ad835f46-b582-44d1-8493-a6a131979194","StatusNotification",{"connectorId":0,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [2,"006b32dd-c05f-4f60-b7cd-d9df764b6fcc","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"","status":"Available","vendorErrorCode":"","vendorId":""}]
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [3,"006b32dd-c05f-4f60-b7cd-d9df764b6fcc",{}]
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [2,"2536226595","GetConfiguration",{}]
[ocpp  ] TRACE 2025/01/26 16:21:31 send 2592AB030003214: [3,"ad835f46-b582-44d1-8493-a6a131979194",{}]
[ocpp  ] TRACE 2025/01/26 16:21:31 recv 2592AB030003214: [3,"2536226595",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown, 1.RST"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"35"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register, Energy.Reactive.Import.Register, Power.Active.Import, Temperature, Voltage.L1, Voltage.L2, Voltage.L3"},{"key":"MeterValuesSampledData","readonly":false,"value":"Current.Import, Current.Offered, Energy.Active.Import.Register, Power.Active.Import, Power.Offered, SoC, Voltage"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"3"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core, FirmwareManagement, LocalAuthListManagement, SmartCharging"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"5"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"2"},{"key":"LightIntensity","readonly":false,"value":"1"},{"key":"MaxEnergyOnInvalidId","readonly":false,"value":"0"},{"key":"WebSocketPingInterval","readonly":false,"value":"30"},{"key":"ACTIVATION_DATE","readonly":true,"value":"2023-12-11T14:14:10Z"},{"key":"APN_NAME","readonly":true,"value":"cpms.esolutions.com"},{"key":"APN_SET","readonly":false,"value":""},{"key":"APN_USER","readonly":true,"value":""},{"key":"C2G_ENABLED","readonly":false,"value":"false"},{"key":"CS_URL","readonly":false,"value":"ws://192.168.178.143:8887/"},{"key":"ClockAlignedAdditionalDataInterval","readonly":false,"value":"900"},{"key":"DPM_ENABLED","readonly":false,"value":"false"},{"key":"DPM_PMAX","readonly":false,"value":"22000"},{"key":"DPM_TYPE","readonly":false,"value":"0"},{"key":"ENABLE_RANDOMIZED_DELAY","readonly":false,"value":"false"},{"key":"GUI_STANDBY_TIMEOUT","readonly":false,"value":"0"},{"key":"HWCONFIG_e","readonly":true,"value":"0"},{"key":"LOAD_UNBALANCE_CURRENT","readonly":false,"value":"32000"},{"key":"LOAD_UNBALANCE_ENABLED","readonly":false,"value":"false"},{"key":"MAX_RANDOMIZED_DELAY","readonly":false,"value":"600"},{"key":"MID_ENABLED","readonly":false,"value":"false"},{"key":"MID_TYPE","readonly":false,"value":"0"},{"key":"MODBUS_BAUD_RATE","readonly":false,"value":"4"},{"key":"MODBUS_RTU_ADDRESS","readonly":false,"value":"1"},{"key":"MS_MODE","readonly":false,"value":"0"},{"key":"MeterValueSampleAdditionalInterval","readonly":false,"value":"150"},{"key":"OPERATING_MODE [... cut ...]
[ocpp  ] TRACE 2025/01/26 16:36:06 recv 2592AB030003214: [3,"2742980092",{"status":"NotImplemented"}]
[ocpp  ] TRACE 2025/01/26 16:36:08 send 2592AB030003214: [2,"1799525380","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/01/26 16:36:08 recv 2592AB030003214: [3,"1799525380",{"status":"NotImplemented"}]
[ocpp  ] TRACE 2025/01/26 16:36:10 send 2592AB030003214: [2,"3748172232","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/01/26 16:36:10 recv 2592AB030003214: [3,"3748172232",{"status":"NotImplemented"}]
`

haraldfunk, 2025-01-26:
Etwas verwundert es mich auch, dass die WebUI erst startet, nachdem der timeout abgelaufen ist. Vorher wird zwar der Port geöffnet, aber der Dienst dahinter reagiert nicht. Scheinbar blockiert hier die OCCP-Suche das Hochlaufen der WebUI? Ist das normal?

haraldfunk, 2025-01-26:
Sorry, hatte das mit dem Fork überlesen und daher zuerst nur den branch fix/ocpp-initial-status-3 aus dem evcc repository getestet. Habe jetzt noch mal deinen Fork geladen, gebaut und folgende Logs erhalten:

[evcc-20250126-165742-trace.log](https://github.com/user-attachments/files/18550908/evcc-20250126-165742-trace.log)

Sieht leider allerdings auch nicht viel besser aus...

mfuchs1984, 2025-01-26:
Ich schau nochmal rein.

mfuchs1984, 2025-01-26:
Hmm, wenn das wirklich mit meiner Version gebaut wurde, ging bei mir irgendwas schief. Da sind Nachrichten drin, die ich eigentlich entfernt habe.

haraldfunk, 2025-01-26:
Ich glaube ich war fälschlicherweise auf deinem main statt im fix branch. Passt der hier besser?:

[evcc-20250126-181754-trace.log](https://github.com/user-attachments/files/18551188/evcc-20250126-181754-trace.log)