evcc-io/evcc #15024: Call to Cupra API results in 404

tgold, 2024-07-24:
### Describe the bug

The API call to the endpoint for getting the car status results in a 404.

### Steps to reproduce

1. configure a car as Cupra 
2. Start charging
3. 
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: {removed}

interval: 30s # control cycle interval

sponsortoken: {removed}

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.91
  port: 1502
  usage: grid
  modbus: tcpip
  timeout: 10s
  name: grid1
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.91
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv2
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.91
  port: 1502
  usage: battery
  modbus: tcpip
  timeout: 10s
  name: battery3

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.0.93
  port: 502
  modbus: tcpip
  name: wallbox5

vehicles:
- type: template
  template: cupra
  title: Cupra Born
  user: {removed}
  password: {removed}
  capacity: 58
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: ev4
  mode: pv
  enable:
    delay: 1m
    threshold: -1600
  disable:
    delay: 10m
    threshold: -500

site:
  title: name
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[cupra ] TRACE 2024/07/24 08:28:27 GET https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/{removed}/vehicles/{removed}/mycar
[cupra ] TRACE 2024/07/24 08:28:27 {"timestamp":"2024-07-24T06:28:27.176470629Z","path":"/v2/users/{removed}/vehicles/{removed}/mycar","status":404,"error":"Not Found","message":"No static resource v2/users/{removed}/vehicles/{removed}/mycar.","requestId":"281f1c4d-8633507"}
[cupra ] TRACE 2024/07/24 09:40:57 POST https://tokenrefreshservice.apps.emea.vwapps.io/refreshTokens
[cupra ] TRACE 2024/07/24 09:40:57 brand=cupra&grant_type=refresh_token&refresh_token={removed}{"access_token":"{removed}","expires_in":3599,"id_token":"{removed}","refresh_token":"{removed}","token_type":"bearer"}
[cupra ] TRACE 2024/07/24 09:40:57 GET https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/{removed}/vehicles/{removed}/mycar
[cupra ] TRACE 2024/07/24 09:40:58 {"timestamp":"2024-07-24T07:40:57.988374052Z","path":"/v2/users/{removed}/vehicles/{removed}/mycar","status":404,"error":"Not Found","message":"No static resource v2/users/{removed}/vehicles/{removed}/mycar.","requestId":"8a1abe18-9282696"}
[cupra ] TRACE 2024/07/24 09:41:27 GET https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/{removed}/vehicles/{removed}/mycar
[cupra ] TRACE 2024/07/24 09:41:27 {"timestamp":"2024-07-24T07:41:27.559217308Z","path":"/v2/users/{removed}/vehicles/{removed}/mycar","status":404,"error":"Not Found","message":"No static resource v2/users/{removed}/vehicles/{removed}/mycar.","requestId":"828b0abc-9264007"}
[cupra ] TRACE 2024/07/24 09:42:27 GET https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/{removed}/vehicles/{removed}/mycar
[cupra ] TRACE 2024/07/24 09:42:27 {"timestamp":"2024-07-24T07:42:27.506778529Z","path":"/v2/users/{removed}/vehicles/{removed}/mycar","status":404,"error":"Not Found","message":"No static resource v2/users/{removed}/vehicles/{removed}/mycar.","requestId":"b6240937-9272867"}
```

### What type of operating system are you running?

Linux

### Version

0.128.4

andig, 2024-07-24:
Looks like api has changed. Does https://github.com/TA2k/ioBroker.vw-connect work for you?

tgold, 2024-07-24:
I don't have ioBroker so I am not able to test.

andig, 2024-07-24:
Unfortunately the git history is not easy to understand: https://github.com/daernsinstantfortress/cupra_we_connect/releases. Do you see where the fix was introduced or which api _is_ used?

andig, 2024-07-24:
Indeed, need to check https://github.com/daernsinstantfortress/WeConnect-Cupra-python which contains the actual code...

sebster6, 2024-07-24:
Could it be [these](https://github.com/daernsinstantfortress/WeConnect-Cupra-python/commit/33e25980597ad73cfc93884301689bb16df3e6db#diff-37c8004d4e3028cdfe20da874de69333e834eb021b5f5e92902501635b726fed) changes?

JBufe, 2024-07-24:
Falls es hilft, der ioBroker Adapter zeigt Verbindungsstatus = Grün also Okay an, aber der SOC stimmt nicht und auch der Timestamp ist alt. 15.06.2024 um genau zu sein.

rainerme, 2024-07-24:
Habe seit heute auch das gleich Problem mit meinem Cupra Born. Interessant war das auch Tronity für ganz kurze Zeit heute morgen 0% angezeigt hat aber nach einer Stunde war dann wieder alles ok. Hat Cupra was verändert? Wäre cool wenn es jemand fixen könnte.

quickmicha, 2024-07-25:
Hi,
same issue here: no vehicle SOC in EVCC.
But in Tronity and in Home Assistant the SOC is available (and correct)
Micha

mafischerhei, 2024-07-25:
Hi.
I’ve got the same issue running into the 404 error …. No soc

backbe, 2024-07-25:
Auch bei mir… kein SOC für meinen CUPRA Born…

digitalonkel, 2024-07-26:
Bei mir das Gleiche. Abhilfemaßnahmen, die früher mal funktionieren haben, wie Ab- und Anmelden in der Cupra App helfen dieses Mal nicht. Fahrzeug gelöscht und neu angelegt auch nicht.

rbeudel, 2024-07-26:
Auch bei mir… kein SOC für meinen CUPRA Born mit 404 Fehler…

MrSmashd, 2024-07-26:
Vielleicht könnte in diesem Zusammenhang auch das Problem der fehlenden KM-Übermittlung an die Load-Session-DB gefixt werden? :)

TazerReloaded, 2024-07-26:
Was mir auch aufgefallen ist: Bei so einem Fehler geht EVCC davon aus, dass der SoC 0% ist. Wenn man eine Minimalladung konfiguriert hat lädt er dann immer das Auto voll, egal was sonst eingestellt ist.
Sinnvoller wäre es, auf Gastfahrzeug zu wechseln. Das habe ich nun auch gemacht, dann verhält er sich bis auf das Zielladen wieder normal.

hazy66, 2024-07-26:
Gleiches Problem hier. Kein soc. Homeassistant und cupra App zeigen korrekte Werte.

pustelbaer, 2024-07-26:
bei mir auch. Der ioB Adapter VW-Connect hingegen zeigt mir den korrekten SOC und Status verbunden an.

Auf das von TazorReloaded beschriebene Problem bin ich auch gestossen. Ich habe in evcc immer 30% Min SOC bei Ankunft eingestellt. Damit hat er mir heute den Born geladen. Ausgerechnet in der Strom-Hochpreisphase. Nicht so toll :-(

vistalba, 2024-07-27:
Ich habe das selbe Problem auch schon zum wiederholten Mal mit EVCC. Wobei es in Homeassistant korrekt funktioniert.

Ich habe mir nun einen Workaround gebaut. Ich übergebe die Werte aus Homeassistant an EVCC. Somit wird das Cupra-API nur von Homeassistant abgefragt. Die Daten danach lokal verteilt. Falls es auch jemandem von euch hilft:

**Wichtig:** Homeassistant URL, Token und die Sensoren sind entsprechend euren Bedürfnissen anzupassen.

```
vehicles:
- type: custom
  name: Cupra
  title: Born
  icon: car
  capacity: 52
  phases: 3
  onIdentify:
    mode: pv
    mincurrent: 6
    maxcurrent: 16
  identifiers:
    - blue
  soc: # battery soc (%)
    source: http
    uri: 'https://homeassistant.local:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
    auth:
      type: bearer
      password: [TOKEN]
    insecure: false
    jq: '.[] | select(.entity_id=="sensor.cupra_born_state_of_charge").state|tonumber'
    timeout: 5s
  range: # optional electric range (km)
    source: http
    uri: 'https://homeassistant.local:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
    auth:
      type: bearer
      password: [TOKEN]
    insecure: false
    jq: '.[] | select(.entity_id=="sensor.cupra_born_range_in_kilometers").state|tonumber'
    timeout: 5s
  odometer:
    source: http
    uri: 'https://homeassistant.local:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
    auth:
      type: bearer
      password: [TOKEN]
    insecure: false
    jq: '.[] | select(.entity_id=="cupra_born_odometer_in_kilometers").state|tonumber'
    timeout: 5s
  limitsoc: # SoC limit configured in vehicle (%) - to be verified
    source: http
    uri: 'https://homeassistant.local:8123/api/states'
    method: GET
    headers:
      - content-type: application/json
    auth:
      type: bearer
      password: [TOKEN]
    insecure: false
    jq: '.[] | select(.entity_id=="sensor.cupra_born_target_state_of_charge").state|tonumber
    timeout: 5s
```

dadaniel, 2024-07-27:
Same issue here.

hazy66, 2024-07-27:
> Ich habe das selbe Problem auch schon zum wiederholten Mal mit EVCC. Wobei es in Homeassistant korrekt funktioniert.
> 
>

sehr cool. Danke. 

Ist der SoC dann auch nutzbar für das Ladelimit dass ich in evcc einstellen kann? Oder eine reine Anzeige?

vistalba, 2024-07-27:
> Ist der SoC dann auch nutzbar für das Ladelimit dass ich in evcc einstellen kann? Oder eine reine Anzeige?

Ja, das funktioniert grundsätzlich, solange Homeassistant über die Cupra API die Daten auch bekommt. (Das Abfrageintervall kenne ich jedoch nicht).
Bei mir ist das nicht der Fall, da der Born an einem Ort ohne Mobilfunkempfang steht und somit nur sehr selten ein update an die Cupra Server schickt.  Somit rechnet EVCC den SoC Anhand des Start-SoC + geladene kWh anhand der konfigurierten Battery Capacity aus.

Aus diesem Grund sind auch Vorklimatisieren und das erkennen, ob der Born am Ladekabel angesteckt ist, nicht konfiguriert.
Kann man aber gemäss Doku auch irgendwie machen:  https://docs.evcc.io/docs/devices/vehicles

Mit dieser Config wird auch der Kilometerstand korrekt geloggt in der EVCC History.

closed 2024-07-27 by andig

backbe, 2024-07-28:
Vielleicht übersehe ich etwas, aber der soc meines Born ist noch immer nicht sichtbar. Habe heute auch ein Update auf die neueste Version von evcc gemacht. Leider noch immer kein soc. Was übersehe ich?

TazerReloaded, 2024-07-28:
Hast Du Version 0.129.0? Die läuft korrekt bei mir:
```
root@evcc:~# evcc vehicle
[main  ] INFO 2024/07/28 13:36:10 evcc 0.129.0
[main  ] INFO 2024/07/28 13:36:10 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/28 13:36:10 using sqlite database: /root/.evcc/evcc.db
Soc:            80%
Capacity:       55.0kWh
Charge status:  A
Range:          372km
Odometer:       43716km
Finish time:    not available
Climate active: false
Position:       **.******,**.******
Limit Soc:      80%
OnIdentified:   Mode:pv
```

backbe, 2024-07-28:
Danke für die Info. Bei mir läuft seit heutigem Update 0.128.4.

Du hast aber sonst keine Änderungen vorgenommen?

hazy66, 2024-07-28:
Das Update auf 0.129.0 ist 3 Stunden alt und enthält den fix der aufgrund dieses Threads entwickelt wurde. Siehe changelog

backbe, 2024-07-28:
Danke für den Hinweis. Da war ich leider 2Stunden zu schnell. Werde heute Abend testen…

martin-zh, 2024-07-29:
Die Abfrage funktioniert via openWB leider noch nicht - der 404er Fehler besteht da weiterhin

HansR-byte, 2024-07-29:
Kann ich bestätigen, während in der CUPRA App nach wie vor alles funktioniert, bekomme ich in openWB weiterhin den 404er Fehler.

hazy66, 2024-07-29:
Das ist hier aber der evcc Thread. Da müsst ihr euch an die openwb Entwickler wenden.

martin-zh, 2024-07-29:
Nein, das müssen wir eher nicht. Das ist wohl kein openWB Problem sondern eines der EVCC Cloud.

hazy66, 2024-07-29:
Ah sorry hab ich falsch verstanden. Ihr  nutzt die SoC Module von evcc in openWb ? 
war mir nicht bewusst dass es da eine Integration gibt. Sorry

vistalba, 2024-07-29:
> Nein, das müssen wir eher nicht. Das ist wohl kein openWB Problem sondern eines der EVCC Cloud.

Was soll denn das sein? Ich vermute einfach Mal du hast eine OpenWB, welche über EVCC gesteuert wird. 
Eine EVCC Cloud gibt es soweit ich weiss nicht.

Aus meiner Sicht gibt es folgende Komponenten:

- EVCC
- Wallbox (bei euch OpenWB, bei mir Go-E)
- Curpa API (Service von Cupra, wo Ihre App drüber läuft)

Das Problem bestand darin, dass es in der Cupra App Änderungen gab und deshalb die Daten von EVCC anders abgefragt werden müssen. Die Cupra Entwickler warnen die EVCC Devs nicht vor. Man merkt dann, dass es nicht mehr geht.
Dies wurde mit dem Change hier korrigiert und es funktioniert nun wieder. 
Wenn die Kombination von OpenWB und EVCC also hier noch nicht funktioniert, würde ich eher einen neuen GitHub Issue eröffnen und genau beschreiben, was noch nicht geht und die Logs anhängen.

martin-zh, 2024-07-29:
Du vermutest falsch - wir fragen mit der openWB den SoC via dem EVCC Cloud Dienst ab. Warten wir doch, bis @andig sich dazu äussert.

andig, 2024-07-29:
Ist erledigt!

backbe, 2024-07-29:
Bei mir, mit Version 0.129.0 und Go-e wird der Soc wieder korrekt angezeigt.

Danke für die schnelle Behebung!

HansR-byte, 2024-07-29:
In openWB wird der SoC wieder angezeigt. Danke für die schnelle Anpassung.

voruti, 2024-07-29:
> wir fragen mit der openWB den SoC via dem EVCC Cloud Dienst ab.

Was ist der EVCC Cloud Dienst? Wo kann ich mehr dazu lesen?

mafischerhei, 2024-07-30:
Läuft auch bei mir wieder. Danke für den schnellen fix