evcc-io/evcc #15027: Zeitlich bereits abgelaufener Ladeplan aus der Vergangenheit wird fortgesetzt

mule1972, 2024-07-24:
### Describe the bug

Das Folgende gilt für den PV-Modus:

Ich weiß nicht, ob es ein Bug oder gewolltes Verhalten ist: Ein Ladevorgang, welcher durch Trennen des Autos von der Ladestation unterbrochen wurde, dann zeitlich eigentlich abgelaufen ist, wird zu einem späteren Zeitpunkt einfach forgesetzt und so auch notfalls Strom aus dem Netz gezogen, um das Ladeziel aus dem veralteten Ladezeitplan zu erreichen.

Ich bin der Meinung, dass es nicht sinnvoll ist zeitlich abgelaufene Ladepläne zu einem späteren Zeitpunkt wieder fortzusetzen und dabei im PV-Modus sogar Strom aus dem Netz gezogen wird. Derzeit lässt sich dieses Verhalten nur verhindern, in dem man den zeitlich abgelaufenen Ladeplan manuell beendet, was derzeit durch den Bug [https://github.com/evcc-io/evcc/issues/14954](url) zudem noch erschwert wird.

### Steps to reproduce

1.) Ladeplan mit Tag x und Abfahrtszeit 9:00Uhr und einer Zielakku-Kapazität von 70% erstellen
2.) Um 8.30Uhr das Auto von der Ladestation trennen
3.) Um 20Uhr das Auto wieder an die Ladestation anschließen

Ergebnis: Der um 9Uhr eigentlich zeitlich abgelaufene Ladeplan wird nach Anschluss des Autos um 20Uhr wieder fortgesetzt und dabei wird auch im PV-Modus Strom aus dem Netz gezogen. 

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.128.4

andig, 2024-07-24:
> Ein Ladevorgang, welcher durch Trennen des Autos von der Ladestation unterbrochen wurde, dann zeitlich eigentlich abgelaufen ist, wird zu einem späteren Zeitpunkt einfach forgesetzt und so auch notfalls Strom aus dem Netz gezogen, um das Ladeziel aus dem veralteten Ladezeitplan zu erreichen.

Tatsächlich ist das by design da wir die Pläne am Fahrzeug speichern sofern sie sich auf einen Ziel-Soc beziehen. Das erscheint auch sinnvoll: wenn Du morgen früh 80% im Auto haben willst ändert sich das nicht dadurch, dass jemand zwischendurch nochmal zum Bäcker fährt.

/cc @naltatis haben wir hier noch ein Dokumentationsthema?

naltatis, 2024-07-24:
>  wenn Du morgen früh 80% im Auto haben willst ändert sich das nicht dadurch, dass jemand zwischendurch nochmal zum Bäcker fährt.

@andig Das ist ja nicht das im Issue beschriebene Verhalten. Bei deinem zwischendurch beim Bäcker Beispiel bin ich bei dir. Wenn ich aber länger unterwegs bin und erst **nach dem Ladezielzeitpunkt** wieder Anstecke würde ich erwarten, dass der Ladeplan nicht mehr aktiv ist.

mule1972, 2024-07-25:
Könnte dieses Issue bitte wieder geöffnet werden, so dass das Thema öffentlich ist und weiter diskutiert werden kann?

mule1972, 2024-07-30:
Heute wieder die Situation erlebt, dass ich für 9:30IUhr eine Zielplanung eingestellt hatte. Musste dann aber kurzfristig schon um 9:15Uhr zur Arbeit fahren und habe den Ladestecker entsprechend entfernt. 
Ich bin dann heute um ca. 20Uhr wieder zurückgekommen und EVCC hat mir die Batterie leer gesaugt und Strom vom Netz bezogen weil der eigentlich seit Stunden abgelaufen Ladeplan wieder reaktiviert wurde.

Aus meiner Sicht macht das überhaupt keinen Sinn oder gibt es dafür wirklich irgendein sinnvolles Szenario?

andig, 2024-07-30:
Vmtl genauso wenig wie Issues ohne Logfile.

mule1972, 2024-07-30:
Du hattest Doch anhand Deines Bäcker-Beispiels zuvor geschrieben, dass es sich um ein bisher so gewolltes Verhalten handelt. Wie soll dabei ein Logfile helfen?

closed 2024-07-31 by andig