evcc-io/evcc #15032: Add Ampere.STORAGE

andig, 2024-07-24:
Ich habe diesen Request gerade gefunden und beantworte den mal so:
Ampere.Storage.Pro ist baugleich mit SAJ-H2.
Siehe auch hier: [Photovoltaik Forum](https://www.photovoltaikforum.com/thread/222147-ekd-speicher-ampere-storage-pro-ein-chinamodell-und-weitere-fragen-dazu/?pageNo=11) 

Ich habe die Batterie so eingebunden und funktioniert einwandfrei.

```
- name: ampere_storage_pro
    type: template
    template: saj-h2
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.xxx # Hostname
    port: 502 # Port

```

Usage: pv und grid funktioniert bei mir nicht, da ich außer der Anlage von EKD Solar noch weitere Wechselrichter und Panels in Betrieb habe.  Da sind die entsprechenden Werte im Wechselrichter systembedingt falsch. Sollte das nicht der Fall sein, werden die Werte für pv und grid vermutlich auch i.O. sein.

Alle PV Leistungen addiere ich in Home Assistant und schicke die per mqtt an evcc. 

Die Ampere.IQ Box und App habe ich zum Glück nicht. Nach meinen Infos kann ich auch nur davon abraten. evcc macht ein viel besseres Management beim Überschussladen als die Ampere.IQ Box. Lieber sollten die Anwender das Geld oder einen Teil davon in [evcc Spnsoring](https://docs.evcc.io/docs/sponsorship) stecken!!

_Originally posted by @UDicke in https://github.com/evcc-io/evcc/discussions/8394#discussioncomment-10137970_

UDicke, 2024-07-24:
Danke! Wie kann ich hier einen kleinen Beitrag leisten? Ich möchte z.B. gerne 
1. versuchen die "Entladesperre beim Schnelladen" für Ampere.Storage.Pro bzw. SAJ-H2 zu implementieren und
2. das Laden der Hausbatterie aus dem Netz mit dynamischen Tarifen. (wenn 1. erfolgreich war)

Wo finde ich die notwendigen Infos mit welchen Parametern evcc versorgt werden möchte? Gibt es eine Möglichkeit z.B.  eine geänderte "templates/definition/meter/saj-h2.yaml" bei mir local zu testen? Bei mir läuft evcc als HA Add-on.

andig, 2024-09-01:
Du kannst das immer ohne Nutzung von Templates lokal implementieren.

UDicke, 2024-09-01:
Danke für die Info! Hast Du ein Beispiel oder den Hinweis auf eine Doku zum Thema für mich?

Die entsprechenden Register habe ich  bei mir getestet und sowohl eine Entladesperre als auch das Laden aus dem Netz kann ich bei mir einstellen.
Die entsprechenden Register schreiben und sehen was passiert kann ich über Home Assistant und/oder qmodmaster, da brauche ich keine "custom meter" für.
Um das zu testen muss ich doch z.B. die Info "Entladesperre ein" von evcc bekommen und dann agieren oder habe ich da was noch nicht verstanden?

andig, 2024-09-01:
Schau mal z.b. im Fronius Template nach batterymode.

UDicke, 2024-09-01:
Sorry, ich bin überfordert!
ich kann weder mit templates/definition/meter/fronius-gen24.yaml noch mit templates/definition/meter/fronius-solarapi-v1.yaml was anfangen.
Wo soll ich denn was passendes finden?

UDicke, 2024-09-01:
WIe wäre es denn wenn ich eine Beschreibung der entsprechenden Register liefere und ein Fachmann implementiert den erweiterten Batteriemode incl. Laden aus dem Netz?

UDicke, 2024-09-02:
Mit diesen Infos sollte es für einen Fachmann ein Kleines sein die aktive Batteriesteuerung incl. Entladesperre und Netzladen zu implementieren.
[Beschreibung_SAJ_aktive_Batteriesteuerung.pdf](https://github.com/user-attachments/files/16839058/Beschreibung_SAJ_aktive_Batteriesteuerung.pdf)
Hier noch eine Auszug "schreibbare Register" aus der Modbus Doku vom SAJ-H2 (ab Seite 12)
[H2-3~6K-S2 single phase communication protocol -2022.12.02-EN_schreibbare_Register.pdf](https://github.com/user-attachments/files/16839094/H2-3.6K-S2.single.phase.communication.protocol.-2022.12.02-EN_schreibbare_Register.pdf)
Falls Bedarf besteht hier die komplette Dokumentation vom SAJ-H2.
[H2-3~6K-S2 single phase communication protocol -2022.12.02-EN.pdf](https://github.com/user-attachments/files/16839114/H2-3.6K-S2.single.phase.communication.protocol.-2022.12.02-EN.pdf)

closed 2024-09-07 by andig

UDicke, 2024-09-07:
Gut, ist denn jetzt das Thema aktive [Batteriesteuerung incl. Entladesperre und Netzladen](https://github.com/evcc-io/evcc/issues/15032#issuecomment-2324951780) auch "closed" und "completed" für SAJ-H2 und Ampere.Storage? @andig muss ich jetzt was tun damit das noch ein "open issue" bleibt?

andig, 2024-09-08:
Batteriesteuerung ist hier OT. Mach gerne ein Issue auf wenn Du die konkreten Register hast die jeweils gesetzt werden müssen.

pthauer, 2024-09-10:
Das klingt alles super. Habe ein Ampere Storage Pro und steuere ihn über die IQBox mit custom template an.
Wenn aber jetzt ein direkter Zugang zum Speicher implementiert ist, macht der Umstieg Sinn. 
Ab wann ist denn mit der Implementierung der Entladesperre zu rechnen? Momentan mache ich das noch händisch, wenn ich mein EV mit den 11kW lade. Es macht keinen Sinn, das aus dem Speicher zu holen. 
Wäre also gut, wenn EVCC vor dem Start des 11kW Ladevorganges den Ladezustand des Speichers abfragt, diesen dann als Entladesperre setzt und mit dem Ladevorgang startet. Nach beenden des Ladevorganges sollte die Entladedperre dann wieder auf 20% oder so gesetzt werden.

UDicke, 2024-09-10:
Der erweiterte Batteriemodus für SAJ-H2 bzw. Ampere.Storage.Pro wird kurzum zur Verfügung stehen nach meiner Einschätzung. Die Erweiterung ist fertig und wartet drauf in einer der nächsten Versionen einzufließen.

Kannst Du ein paar Infos zu Deiner Installation geben?

Kann die IQBox denn die Batterie bzw. das Schnellladen nicht steuern?

Hast Du eine Wallbox von EKD Solar bzw. eine Weidmüller Wallbox? Die sollte doch funktionieren…

 

Ich bin der Meinung, dass die Steuerung über evcc um ein vielfaches besser ist als über die IQBox. Ich habe (zum Glück) keine gekauft und kann das nicht prüfen.

Kannst Du die Funktionalitäten gegenüberstellen?

 

Gruß Ulrich

UDicke, 2024-09-11:
Danke für die Info,
mit der aktiven Batteriesteuerng wird evcc nicht nur eine Entladesperre haben, sondern auch das Laden aus dem Netz mit dynamischen Tarifen unterstützen. Sorry, aber ich muss das nochmal sagen: Ich bin froh, dass ich nicht auch noch so eine IQ.Box bestellt habe...😊