evcc-io/evcc #15033: Cupra Born not found 404

JBufe, 2024-07-24:
Seit ein paar Tagen wird mein Cupra Born nicht erkannt. Fehler 404 .  

evcc vehicle liefert 
[main  ] INFO 2024/07/24 18:43:41 evcc 0.128.4 (70b87715)
[main  ] INFO 2024/07/24 18:43:41 using config file: /root/evcc.yaml
[db    ] INFO 2024/07/24 18:43:41 using sqlite database: /root/.evcc/evcc.db
.Soc:          unexpected status: 404 (Not Found)
Capacity:      58.0kWh
Charge status: unexpected status: 404 (Not Found)
Range:         unexpected status: 404 (Not Found)
Odometer:      unexpected status: 404 (Not Found)
Finish time:   unexpected status: 404 (Not Found)
Climater:      unexpected status: 404 (Not Found)
Limit Soc:     unexpected status: 404 (Not Found)

Cupra connect Software Verbindung existiert. Auch Down Grades auf vorherigen Versionen von EVCC brachten keine Lösung. Ich bin ratlos.

TobiasHuber1980, 2024-07-24:
Siehe https://github.com/evcc-io/evcc/issues/15024

JBufe, 2024-07-24:
OK Danke & dann über ich mich in Geduld. An dieser Stelle einfach mal einen Dank an euch Cracks für die Zeit und Mühe die ihr in die Projekte steckt.

GrimmiMeloni, 2024-07-26:
Dupe of #15024. Closing.

closed 2024-07-26 by GrimmiMeloni