evcc-io/evcc #15034: ABB B23 Gridmeter does not run on template abb-ab using modbus: rs485tcpip

fk8-qvy5, 2024-07-24:
### Describe the bug

**EVCC does not read the gridmeter via tcpip. Ethernetadapter is ET69C02** 
[site ] ERROR 2024/07/24 17:33:42 grid meter: read failed: read tcp 192.168.20.149:40572->192.168.20.189:502: i/o timeout

**gridmeter works with mbpoll, [4144]:         64673 (-863) is the current 3-phase system power**
 mbpoll -m tcp -a 1 -r 4142 -c 10 192.168.20.189
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 4142, count = 10
Communication.........: 192.168.20.189, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[4142]:         65535 (-1)
[4143]:         65535 (-1)
**[4144]:         64673 (-863)**
[4145]:         65535 (-1)
[4146]:         65250 (-286)
[4147]:         65535 (-1)
[4148]:         65209 (-327)
[4149]:         65535 (-1)
[4150]:         65287 (-249)
[4151]:         32767

### Steps to reproduce

1. Connect RS485 between ABB B23 and Ethernet adapter ET69C02 
2. Connect Ethernet adapter to Fritzbox
3. configure ABB B23 to 
    Protoc:    Modbus
    bAUd:    9600
    AddrES:    1
    PArity:    nonE

4. Configure Ethernet Adapter to: 
[image]

5. install template: abb-ab
...
6. run evcc
7. get error message: [site ] ERROR 2024/07/24 17:33:42 grid meter: read failed: read tcp 192.168.20.149:40572->192.168.20.189:502: i/o timeout
[evcc-20240724-172528-trace.log](https://github.com/user-attachments/files/16365477/evcc-20240724-172528-trace.log)
[evcc-20240724-172502-debug.log](https://github.com/user-attachments/files/16365478/evcc-20240724-172502-debug.log)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxxxxxxx

interval: 30s # control cycle interval

meters:
- type: template
  template: abb-ab
  id: 1
  host: 192.168.20.189
  port: 502
  usage: grid
  modbus: rs485tcpip
  name: grid1

chargers:
- type: template
  template: keba
  host: 192.168.20.174
  name: wallbox1

vehicles:
- type: template
  template: nissan 
  title: Leaf Blu  
  icon: car  
  user: xxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxxxxxxxxxxxx
  vin: xxxxxxxxxxxxxxxxxxxxxxx  
  capacity: 39  
  phases: 1  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 32  
  name: ev3

loadpoints:

- title: Sued
  charger: wallbox1
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  
site:
  title: Zuhause
  meters:
    grid: grid1
```

### Log details

```text
[main  ] INFO 2024/07/24 17:22:52 evcc 0.128.4
[main  ] INFO 2024/07/24 17:22:52 using config file: evcc.yaml
[db    ] INFO 2024/07/24 17:22:52 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] INFO 2024/07/24 17:22:52 listening at :7070
[lp-1  ] WARN 2024/07/24 17:22:53 deprecated: mincurrent setting is ignored, please remove
[lp-1  ] WARN 2024/07/24 17:22:53 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/07/24 17:22:53 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/07/24 17:22:53 charge total import: 223.172kWh
[site  ] INFO 2024/07/24 17:22:53 site config:
[site  ] INFO 2024/07/24 17:22:53   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/07/24 17:22:53     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/07/24 17:22:53   vehicles:
[site  ] INFO 2024/07/24 17:22:53     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/07/24 17:22:53 loadpoint 1:
[lp-1  ] INFO 2024/07/24 17:22:53   mode:        off
[lp-1  ] INFO 2024/07/24 17:22:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/24 17:22:53   meters:      charge ✓
[lp-1  ] INFO 2024/07/24 17:22:53     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/07/24 17:22:53 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/24 17:22:53 phase timer inactive
[lp-1  ] DEBUG 2024/07/24 17:22:53 pv timer inactive
[site  ] DEBUG 2024/07/24 17:22:53 ----
[lp-1  ] DEBUG 2024/07/24 17:22:54 charge power: 0W
[lp-1  ] DEBUG 2024/07/24 17:22:54 charge currents: [0 0 0]A
[site  ] ERROR 2024/07/24 17:23:04 grid meter: read failed: read tcp 192.168.20.149:35814->192.168.20.189:502: i/o timeout
[site  ] DEBUG 2024/07/24 17:23:23 ----
[lp-1  ] DEBUG 2024/07/24 17:23:24 charge power: 0W
[lp-1  ] DEBUG 2024/07/24 17:23:24 charge currents: [0 0 0]A
[site  ] ERROR 2024/07/24 17:23:34 grid meter: read failed: read tcp 192.168.20.149:45706->192.168.20.189:502: i/o timeout
[site  ] DEBUG 2024/07/24 17:23:54 ----
[lp-1  ] DEBUG 2024/07/24 17:23:54 charge power: 0W
[lp-1  ] DEBUG 2024/07/24 17:23:54 charge currents: [0 0 0]A
[site  ] ERROR 2024/07/24 17:24:04 grid meter: read failed: read tcp 192.168.20.149:54876->192.168.20.189:502: i/o timeout
[site  ] DEBUG 2024/07/24 17:24:24 ----
[lp-1  ] DEBUG 2024/07/24 17:24:24 charge power: 0W
[lp-1  ] DEBUG 2024/07/24 17:24:24 charge currents: [0 0 0]A
[site  ] ERROR 2024/07/24 17:24:34 grid meter: read failed: read tcp 192.168.20.149:45636->192.168.20.189:502: i/o timeout
[site  ] DEBUG 2024/07/24 17:24:53 ----
[lp-1  ] DEBUG 2024/07/24 17:24:54 charge power: 0W
[lp-1  ] DEBUG 2024/07/24 17:24:54 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

128.4

StefanSchoof, 2024-07-24:
Does `modbus: tcpip` work? You use also tcp in m poll.

fk8-qvy5, 2024-07-24:
What do you mean exactly?T. v. BlumenthalMobil gesendet