evcc-io/evcc #15036: Runtime Error bei refreshToken in mercedes template

phartwig74, 2024-07-24:
### Describe the bug

Bis zur 0.128.2 nightly lief noch alles mit dem mercedes template und gültigen tokens.
Seit 0.128.3 nightly kommt der runtime error nach dem token refresh des mercedes template.
Gestern neuer build auf 0.128.4 nightly und folgender Fehler:
      
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x524ead2]

ohne tokens in der evcc.yaml  kommt kein runtime error.
Ebenso kein runtime error wenn die tokens ungültig werden, bzw. der refreshToken nicht mehr funktioniert.

token Generierung funktioniert fehlerfrei.

Zum Test habe ich mal die Config aus der evcc.yaml in die DB migriert wie in der UI beschrieben mit "evcc migrate".
Vehicle mit neu generierten Tokens versucht anzulegen ... die Prüfung schlägt jedoch fehl und im Log erscheint auch ein runtime error (hab ich hier jedoch nicht dokumentiert, da bisher die config aus der evcc.yaml verwendet wurde).

Wie kann ich zur Problemlösung beitragen ?

### Steps to reproduce

1. mercedes vehicle token erzeugen und in der evcc.yaml eintragen -> evcc Neustart -> runtime error
2. tokens aus der evcc.yaml ggf. aus den evcc.db settings entfernen -> evcc Neustart -> kein runtime error 
3. neue vehicle token erzeugen und in der evcc.yaml eintragen -> evcc Neustart -> runtime error

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error
  ocpp: trace 

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /usr/local/share/evcc/evcc.db

# unique installation id
plant: ***

interval: 15s # control cycle interval

tariffs:
  grid:
    type: fixed
    price: 0.34
  feedin:
    type: fixed
    price: 0.00
  co2:
    type: template
    template: grünstromindex
    zip: 53474

meters:
- type: template
  template: openems 
  usage: grid  
  host: 10.1.90.214  
  password: user  
  name: grid1
- type: template
  template: openems 
[... 367 lines cut ...]
github.com/evcc-io/evcc/core/coordinator.(*Coordinator).identifyVehicleByStatus(0x87180ef00, {0x870910f10?, 0x871454408?, 0x5bac500?})
	github.com/evcc-io/evcc/core/coordinator/coordinator.go:139 +0x12e
github.com/evcc-io/evcc/core/coordinator.(*adapter).IdentifyVehicleByStatus(0x870d8a8e8)
	github.com/evcc-io/evcc/core/coordinator/adapter.go:40 +0x45
github.com/evcc-io/evcc/core.(*Loadpoint).identifyVehicleByStatus(0x871454408)
	github.com/evcc-io/evcc/core/loadpoint_vehicle.go:284 +0x5c
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x871454408, 0x4097640000000000, 0x0, {0x0, 0x0, 0x0}, 0x0, 0x0, 0x0, 0x871f578d8, ...)
	github.com/evcc-io/evcc/core/loadpoint.go:1672 +0x5fb
github.com/evcc-io/evcc/core.(*Site).update(0x871110a80, {0x15c21b0, 0x871454408})
	github.com/evcc-io/evcc/core/site.go:820 +0x70c
github.com/evcc-io/evcc/core.(*Site).Run(0x871110a80, 0x8700fb260, 0x37e11d600)
	github.com/evcc-io/evcc/core/site.go:936 +0x217
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:288 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:287 +0x1e13
```

### What type of operating system are you running?

Linux

### Version

0.128.4 (5e608fb6)

ReneNulschDE, 2024-07-24:
Looks like my last PR to get the MaxSoc does not work in your context. What car model do you have?

phartwig74, 2024-07-24:
> Looks like my last PR to get the MaxSoc does not work in your context. What car model do you have? EQV? Could you add the account `mb.dev.eu@nulsch.de` as a secondary driver of this car in you're MB-Portal? I don't have access to an EQV currently, then I can check this. (tested this with a Hybrid, Diesel, EQB, EQS)

The car is mercedes EQV300 with 90 kWh Batterie... I added the account for you as requested.

phartwig74, 2024-07-25:
#15039 fixed the runtime error in my context.

thank you very much @ReneNulschDE

GrimmiMeloni, 2024-07-26:
@phartwig74 meaning you built locally and confirmed #15039 to be working?

closed 2024-07-27 by andig