evcc-io/evcc #15037: Fahrzeugdaten updaten und per MQTT verschicken

s1r0x, 2024-07-24:
Wäre es möglich, dass 
a.) die Fahrzeugdaten auch upgedatet werden, wenn das Fahrzeug nicht angeschlossen ist?
und
b.) diese Daten per MQTT verschickt werden?

Danke und lg

StefanSchoof, 2024-07-24:
Geht siehe https://docs.evcc.io/docs/reference/configuration/loadpoints#poll

s1r0x, 2024-07-24:
Ah, Danke, damit wäre a.) wohl teilweise (zumindest für ein Auto) erledigt,  aber b.) weiterhin offen

naltatis, 2024-07-25:
Wenn es einem Ladepunkt zugeordnet ist, werden die Daten über den Ladepunkt gepublisht. Ein Publishing rein auf Fahrzeugebene (also wenn man mehr Autos als Wallboxen hat) gibts aktuell nicht.

closed 2024-07-25 by naltatis

s1r0x, 2024-07-25:
Jetzt hab ich's teilweise gefunden, Daten des Autos sind, wie du sagst, unterm loadpoint, aber warum ist das alles nicht unter vehicle?
Und den SOC sehe ich aber nicht gepublisht.

s1r0x, 2024-07-25:
Würdet ihr bitte noch das vehicleOdometer mit dem poll updaten