evcc-io/evcc #15040: 0.128.4 Keba-Modbus: Decoding failed "has invalid keys: id, rtu, uri"

core-tinkerer, 2024-07-24:
### Describe the bug

Hi EVCC,

update to 0.128.4 welcomed me with this log while unchanged config file. Setup is 2 Keba wallboxes.

Happy to help testing, but need some intro into evcc debugging.

### Steps to reproduce

1. Update docker container from 0128.3 to 0.128.4

### Configuration details

```yaml
#Keba w/ sponsor token
chargers:
  - name: wallbox4
    type: template
    template: keba-modbus
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.1.232 # Hostname
    port: 502 # Port 

  - name: wallbox5
    type: template
    template: keba-modbus
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.1.200 # Hostname
    port: 502 # Port
```

### Log details

```text
[main  ] INFO 2024/07/24 21:27:52 evcc 0.128.4
[main  ] INFO 2024/07/24 21:27:52 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/24 21:27:52 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/07/24 21:27:53 connecting evcc-1764477475 at tcp://...:1883
[mqtt  ] DEBUG 2024/07/24 21:27:53 tcp://...:1883 connected
[main  ] INFO 2024/07/24 21:27:53 listening at :7070
[main  ] FATAL 2024/07/24 21:27:53 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'keba-modbus': decoding failed due to the following error(s):
'' has invalid keys: id, rtu, uri
[main  ] FATAL 2024/07/24 21:27:53 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Docker container

### Version

0.128.4

StefanSchoof, 2024-07-24:
See https://github.com/evcc-io/evcc/issues/14969

core-tinkerer, 2024-07-24:
Thanks, learned to look for closed issues. Sorry for bothering.

closed 2024-07-24 by core-tinkerer