evcc-io/evcc #15047: evvc nimmt bei Divergenz von Metering-Angaben fälschlicherweise PV an

madkiss, 2024-07-25:
### Describe the bug

Wie in https://github.com/evcc-io/evcc/discussions/11903 ausgeführt, kann es bei unterschiedlichen Polling-Intervallen für einen Tibber Pulse und eine lokal angeschlossene OCPP-Wallbox zu Situationen kommen, in denen der letzte gepollte Wert für den Grid-Provider bereits einen Wert ohne BEV-Ladung anzeigt, der letzte empfangene Wert der OCPP-Wallbox aber noch eine aktive BEV-Ladung anzeigt. Das passiert besonders dann, wenn EVCC gerade vom Pulse gepollt hat und unmittelbar danach die Wallbox das Ende des Ladevorgangs vermeldet, es also zu einer sehr kurzen Überschneidung kommt.

Dann ergibt sich eine große Differenz zwischen dem vom Grid abgenommenen Strom und dem in der Wallbox verbrauchten Strom. evcc geht in solchen Fällen automatisch davon aus, es müsse sich um Energie aus PV handeln, sogar dann, wenn gar keine PV im Setup vorhanden ist. Das führt zu unplausiblen Anzeigewerten, etwa weil bei Setups ohne PV trotzdem ein Prozentsatz des genutzten Stroms als Sonnenenergie angezeigt wird. 

Ich habe meterInterval für die Wallbox jetzt mal auf 5 Sekunden heruntergesetzt, das wird aber kaum helfen. IMHO wäre es stattdessen sinnvoll, könnte man in einem solchen Fall ein Interval für eine kurze Pause definieren, im Sinne von: Wenn Grid Meter und Wallbox unplausible Werte liefern, warte 10 Sekunden und vergleiche es nochmal.

### Steps to reproduce

1. EVCC mit Tibber Pulse und OCCP-Wallbox aufsetzen und Fahrzeug laden
2. Warten, bis ein Corner Case eintritt, in dem Tibber bereits den geringeren Lastwert anzeigt, der letzte von der Wallbox empfangene Wert aber noch alt ist.
3. Es wird eine unplausible Angabe zur genutzten PV-Energie angezeigt, selbst wenn keine PV Teil des Setups ist.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.lan # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-1: debug
  cache: error
  db: error
  ocpp: trace

# unique installation id
plant: hurz

interval: 15s # control cycle interval

chargers:
- type: template
  template: ocpp
  timeout: 2m
  name: wallbox1
  meterInterval: 5s

vehicles:
- type: template
  template: audi
  title: e-tron S Sportback
  user: hurz
  password: hurz
  vin: hurz
  capacity: 95
  name: ev1
- type: template
  template: audi
  title: RS e-tron GT
  user: hurz
  password: hurz
  vin: hurz
  capacity: 93.4
  name: ev2

loadpoints:
- title: Garage
  charger: wallbox1
  mode: now

site:
  title: Hurz
  meters:

tariffs:
  grid:
    type: tibber
    token: hurz # access token
```

### Log details

```text
Jul 25 09:08:57 evcc evcc[2210]: [site ] DEBUG 2024/07/25 09:08:57 ----
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 charge power: 11040W
Jul 25 09:08:57 evcc evcc[2210]: [site ] DEBUG 2024/07/25 09:08:57 grid meter: 2000W
Jul 25 09:08:57 evcc evcc[2210]: [site ] DEBUG 2024/07/25 09:08:57 grid currents: [0 0 0]A
Jul 25 09:08:57 evcc evcc[2210]: [site ] DEBUG 2024/07/25 09:08:57 pv power: 9040W
Jul 25 09:08:57 evcc evcc[2210]: [site ] DEBUG 2024/07/25 09:08:57 site power: 2000W
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 !! session: chargeRater.chargedEnergy=39.5 - chargedAtStartup=0.0
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 charger status: B
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] INFO 2024/07/25 09:08:57 stop charging <-
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 wake-up timer: start
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 !! session: chargedEnergy=39.5 > chargedEnergy=39.4
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 !! session: start=0.000 stop=0.000 chargedEnergy=39.468
Jul 25 09:08:57 evcc evcc[2210]: [lp-1 ] DEBUG 2024/07/25 09:08:57 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
```

### What type of operating system are you running?

Linux

### Version

0.128.4

naltatis, 2024-07-25:
Die Annahme, dass kein PV vorhanden ist kann man leider nicht so einfach treffen. Es ist ein total valides (und auch praktiziertes) vorgehen, dass man evcc mit einer PV-Anlage betreibt, in evcc aber nur ein Grid-Meter hinterlegt hat. Bspw. weil der eigene Wechselrichter keine Kommunikationsschnittstelle hat. Genau für diesen Fall ist die Logik.

Man könnte überlegen, ob man das Grid-Meter führend macht. Sprich unplausible Mehrverbräuche im Haus rechnerisch deckelt.

madkiss, 2024-07-25:
Denkbar wäre auch ein konfigurierbarer Parameter, der explizit aussagt: Es gibt hier keine PV, alle Energie kommt vom Grid, völlig egal was irgendwelche Divergenzen sagen. Das ist zwar sehr brachial, würde das Problem im konkreten Fall aber lösen ;)

naltatis, 2024-07-26:
Ich hab in #15056 mal nen Lösungsweg formuliert und mache hier zu.

closed 2024-07-26 by naltatis

RenatusRo, 2024-07-26:
Verschwindet das Problem eigentlich, wenn man PV=0W einbaut?

```
  - name: pv
    type: custom
    power:
      source: js
      script: |
        0   
```

naltatis, 2024-07-26:
Ich würde vermuten "nein", aber gerne ausprobieren :D