evcc-io/evcc #15048: Charger Versicharge: Seit FW Update 2.135 -> Energiewerte nur noch 10%

achgut, 2024-07-25:
### Describe the bug

Seit der jetzt ausgerollten FW Version 2.135 ist der vom Modbus Register 1692 ( Real Energy 
Consumed Phase Sum) gelieferte Wert um eine 10er Potenz verschoben (Interner Multiplikationsfaktor von 0,1 auf 1). 
Daher liefert die Funktion TotalEngerie() falsche Werte (nur noch 10% vom korrekten Wert)

Daher muss in der Zeile https://github.com/evcc-io/evcc/blob/54fafd22819184a681bfbff39bac538ddd004770/charger/versicharge.go#L167C1-L167C55
der Divisionswert von **1e4 auf 1e3** geändert werden. Dann kommen wieder richtige Werte.

### Steps to reproduce

Sieht man bei jeder Ladesitzung, dass der Wert jetzt nur 10% der reellen Ladeleistung ist.

### Configuration details

```yaml
kein Einfluss durch die Konfiguration
```

### Log details

```text
kein Log Thema
```

### What type of operating system are you running?

Linux

### Version

alle Versionen

GrimmiMeloni, 2024-07-26:
PR welcome. ;)

tomjschwanke, 2024-07-26:
Can you share the output of `evcc charger --name my_charger` so I can see how the firmware value looks like?

PR #15062 (2024-07-28): Versicharge: fix TotalEnergy() being low by factor 10 on firmware 2.135
fixes #15048

No differentiating between firmware versions, as they update automatically according to [this comment](https://github.com/evcc-io/evcc/pull/9268#issuecomment-1666780948)
closed 2024-07-28 by premultiply