evcc-io/evcc #15053: Polestar API

unikinCK, 2024-07-25:
### Describe the bug

Hello, I am trying to connect my Polestar 2 to evcc. 

After entering email as ID and password I get:
cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot find vehicle, got: []

when entering wrong password I get:
cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: code not found

Vehicle is registered in Polestar account / app.

Any ideas?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: Polestar
    type: template
    template: polestar
    title: Polestar # Wird in der Benutzeroberfläche angezeigt (optional)
    user: xxx@yyy.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxx
    capacity: 75
```

### Log details

```text
cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot find vehicle, got: []
```

### What type of operating system are you running?

Linux

### Version

 evcc version 0.128.3

DAMEK86, 2024-07-26:
no issues with 0.128.4 on my side.
config:
```
    vehicles:
    - name: polestar
      type: template
      template: polestar
      title: Polestar 2
      user: xxx@yyy.de
      password: 'xxx'
      cache: 20m
```
startup logs
```
[main  ] INFO 2024/07/26 06:52:12 evcc 0.128.4                                                                                                                                                                                                                                       │
│ [main  ] INFO 2024/07/26 06:52:12 using config file: /etc/evcc.yaml                                                                                                                                                                                                                  │
│ [db    ] INFO 2024/07/26 06:52:12 using sqlite database: /root/.evcc/evcc.db                                                                                                                                                                                                         │
│ [main  ] INFO 2024/07/26 06:52:12 listening at :7070                                                                                                                                                                                                                                 │
│ [ocpp-1] DEBUG 2024/07/26 06:52:14 waiting for chargepoint: 5m0s                                                                                                                                                                                                                     │
│ [ocpp  ] INFO 2024/07/26 06:52:15 charge point connected, registering: xyz                                                                                                                                                                                                    │
│ [lp-1  ] DEBUG 2024/07/26 06:52:23 charge total import: 201.070kWh                                                                                                                                                                                                                   │
│ [site  ] INFO 2024/07/26 06:52:23 site config:                                                                                                                                                                                                                                       │
│ [site  ] INFO 2024/07/26 06:52:23   meters:      grid ✓ pv ✓ battery ✓                                                                                                                                                                                                               │
│ [site  ] INFO 2024/07/26 06:52:23     grid:      power ✓ energy ✓ currents ✓                                                                                                                                                                                                         │
│ [site  ] INFO 2024/07/26 06:52:23     pv 1:      power ✓ energy ✗ currents ✗                                                                                                                                                                                                         │
│ [site  ] INFO 2024/07/26 06:52:23     pv 2:      power ✓ energy ✗ currents ✗                                                                                                                                                                                                         │
│ [site  ] INFO 2024/07/26 06:52:23     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗                                                                                                                                                                                        │
│ [site  ] INFO 2024/07/26 06:52:23   vehicles:                                                                                                                                                                                                                                        │
│ [site  ] INFO 2024/07/26 06:52:23     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗                                                                                                                                                                                        │
│ [lp-1  ] INFO 2024/07/26 06:52:23 loadpoint 1:                                                                                                                                                                                                                                       │
│ [lp-1  ] INFO 2024/07/26 06:52:23   mode:        minpv                                                                                                                                                                                                                               │
│ [lp-1  ] INFO 2024/07/26 06:52:23   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗                                                                                                                                                                                       │
│ [lp-1  ] INFO 2024/07/26 06:52:23   meters:      charge ✓                                                                                                                                                              [... cut ...]
│ [lp-1  ] DEBUG 2024/07/26 06:52:23 pv timer inactive                                                                                                                                                                                                                                 │
│ [lp-1  ] INFO 2024/07/26 06:52:23 vehicle updated: unknown -> Polestar 2`` 
```

did you checked your yaml config?
on top of that check if you are able to login into https://polestarid.eu.polestar.com/PolestarLogin/login
Seems for some accounts, you need to request the api permissions over support.

unikinCK, 2024-07-26:
Solved: If you buy a Polestar pre-owned you need to register new ownership with Polestar customer service.

closed 2024-07-26 by unikinCK