evcc-io/evcc #15056: Dynamic prices: only show solar % values when PV is configured

naltatis, 2024-07-26:
replaces https://github.com/evcc-io/evcc/issues/15047

**Is your feature request related to a problem? Please describe.**
As described here evcc always assumes a PV is present, even if no PV-meter is configured. This leads to unexpected effects (non-zero solar %) for people that use evcc without PV (dynamic prices).

**Describe the solution you'd like**
We could, for new installations, change our default assumption that a PV is always present. We'll still support solar charging with only a grid-meter but make this opt-in. Config UI could show an "I have solar, but no PV-meter." to enable the existing behavior.

This would also properly solve the issue, that we currently show "PV" and "Min-PV" option (instead of the single "smart" option) even if no PV is installed.

**Additional context**
Since this is currently not a functional issue, I'll not see this implemented before we've fully migrated to web-based configuration.

madkiss, 2024-07-26:
How much will it cost to bump this up in the backlog? For non-pv environments, this issue effectively renders the price display in the "previous charges" list unusable (as demonstrated by the attached screenshot). 

[image]

naltatis, 2024-07-26:
> How much will it cost to bump this up in the backlog?

This is now how this works. 😄

Workarounds:
a) configure a feed-in tariff with the same settings as grid the price calculation. then the calculated price should be correct. just solar % is off.
b) remove the grid meter, then all values should be correct (but you obviously loose the visualizations based on this value). extra benefit: for installations without grid meter, we already have reduced mode UI: off, smart, fast. smart is functionally identical to pv.

madkiss, 2024-07-26:
AIUI, feedin does not support tibber as type, does it?

naltatis, 2024-07-26:
There are no type limitations for feed-in.

andig, 2024-07-28:
> How much will it cost to bump this up in the backlog? 

We appreciate sponsorship in any case :)

joriws, 2025-03-10:
I add my comment as for new evcc user the UI is giving headache. I've configured Dynamic Tariff with Nordpool "template". My car is there with BLE integration, my TWCgen3 is there, my load management is there.

But I don't have solar configured, *why* do I need to select solar for Nordpool-based night time charging? If there is no PV setup take Solar and "min+solar" away. 

Please add button/command for easy cheapest tariff charging if dynamic tariff is configured, display vehicle stats how many kWh is required for set charge level (50% => 80%) for example, with my load config 6A-10A 3-phase select cheapest hours, and just charge.

This is my own dashboard/control in Home Assistant which I am seeking to replace with evcc. It answers my questions about next loading session, it's lenght and dynamic pricing thresholds and rank linits. But my dashboard is missing direct control of the car and dynamic load monitoring.

[image]

andig, 2025-03-11:
> But I don't have solar configured, why do I need to select solar for Nordpool-based night time charging?

At this time, it's the acronym for "auto". Making this better is on the todo list.

joriws, 2025-03-12:
> At this time, it's the acronym for "auto". Making this better is on the todo list.

There is currently problem with this Solar/"auto". If I select "solar"-mode in the evening charging starts **immediately**. Instead of projected charging time (like 30min battery heating + 3h charging) which should set charging to cheapest 4h period (in 1h resolution market). After 11.6.2025 this should be per 15min tariffs optimization in EU where prices are set by Nordpool electricity stock.

Basti77, 2026-10-16:
Nach dem UI-Umbau in 0.317 vermisse ich den "Min+PV"-Modus massiv.

Mein Setup: PV + Hausakku + Garage-Loadpoint. Vorher ein Click auf
"Min+PV" und das Auto lud garantiert mit Min-Strom weiter, bei
Überschuss hochgeregelt. Jetzt:

- "Smart" setzt bei schwachem PV sofort pvAction=disable → Auto aus
- "Schnell" zieht teuren Netzstrom
- Dazwischen gibt es nichts mehr Klickbares

Dass "Smart" das intern können soll ist mir klar, aber praktisch
reicht ein kurzer Wolkenschatten (PV < Min-Ladeleistung) und evcc
schaltet ab. Dann steht das Auto obwohl Hausakku zu 60 % voll ist
und gerne 1,4 kW abgeben würde.

Dazu kommt: der Chevron am "Smart"-Button sieht aus wie ein Dropdown,
aber ein Klick/Tap öffnet weder am Desktop noch am Handy ein Menü —
es passiert einfach gar nichts. Also selbst wenn in dem Dropdown
noch alte Modi versteckt wären, man kommt nicht ran.

Workaround via REST-API (POST .../mode/now + minCurrent/maxCurrent
auf 5A manuell setzen) funktioniert — aber das sollte nicht die
Lösung für die alltägliche "Auto bitte mit Min aus PV laden"-Frage
sein.

+1 für Rückkehr des "Min+PV"-Modus oder zumindest eine Smart-Variante,
die bei PV-Mangel aus dem Hausakku weiterlädt statt abzuschalten.

naltatis, 2026-10-16:
@Basti77 Das Min+PV Verhalten ist nicht weg. Du erhältst genau das gleiche Verhalten im Smart Modus mit "Dauerhaft laden" aktiviert. Siehe https://docs.evcc.io/de/features/modes/#dauerhaft-laden

>  Dazu kommt: der Chevron am "Smart"-Button sieht aus wie ein Dropdown, aber ein Klick/Tap öffnet weder am Desktop noch am Handy ein Menü - es passiert einfach gar nichts. Also selbst wenn in dem Dropdown noch alte Modi versteckt wären, man kommt nicht ran.

Falls sich das wirklich nicht öffnet mach mal ein Issue mit genauer Betriebssystem, Browser und evcc-Version auf. Teste auch vorher einmal obs auf https://demo.evcc.io funktioniert.