evcc-io/evcc #15059: TAPO P110 inop

skysoftgmbh, 2024-07-26:
### Describe the bug

Hi,

seems like TAPO pushed a new firmware today. Now I can't start evcc anymore.

Please have a look.

Thanks,

Peer

### Steps to reproduce

start evcc -> get this error in the log

### Configuration details

```yaml
meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.5.27
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.5.27
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1
  host: 192.168.5.27
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: tapo
  usage: charger
  host: 192.168.5.33
  user: xxxxxxx
  password: xxxxxx
  name: MTapo1
- type: template
  template: tapo
  usage: charger
  host: 192.168.5.37
  user: xxxxxxx
  password: xxxxxxx
  name: MTapo2

chargers:
- type: template
  template: easee
  user: xxxxxx
  password: xxxxxxx
  host: 192.168.5.24
  charger: xxxxxx
  name: wallbox
- type: template
  template: tapo 
  host: 192.168.5.33
  user: xxxxx
  password: xxxxxxx
  standbypower: 0
  integrateddevice: true
  name: Tapo1
- type: template
  template: tapo 
  host: 192.168.5.37
  user: xxxxx
  password: xxxxx
  standbypower: 0
  integrateddevice: true
  name: Tapo2

loadpoints:
- title: Parkplatz
  charger: wallbox
  vehicle: my_car
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 20
- title: Warmwasser 2KW
  charger: Tapo2
  mode: off
  phases: 1
  mincurrent: 9.6 #Verbrauch ca. 2200W
  maxcurrent: 9.6
  enable:
   threshold: 0
   delay: 5m
  disable:
    threshold: 0
    delay: 5m # Laufzeit mindestens 5 Minuten
  guardduration: 15m # Pause mindestens 15 Minunten
  priority: 1
- title: Bosch Charger
  charger: Tapo1
  mode: off
  phases: 1
  mincurrent: 0.4 #Verbrauch ca. 100W
  maxcurrent: 0.4
  enable:
   threshold: 0
   delay: 5m
  disable:
    threshold: 0
    delay: 5m # Laufzeit mindestens 5 Minuten
  guardduration: 15m # Pause mindestens 15 Minunten
```

### Log details

```text
024/07/26 09:15:27,stdout,panic: crypto/cipher: input not full blocks
2024/07/26 09:15:07,stdout,"[tapo  ] DEBUG 2024/07/26 09:15:07 SMART.TAPOPLUG P110 connected (fw:1.3.0 Build 230905 Rel.152200,hw:1.0,mac:1C-61-B4-C3-D8-FD)
"
2024/07/26 09:15:07,stdout,[proxy-5021] DEBUG 2024/07/26 09:15:07 modbus proxy for 192.168.5.27:502 listening at :5021
2024/07/26 09:15:07,stdout,[main  ] INFO 2024/07/26 09:15:07 listening at :7070
2024/07/26 09:15:02,stdout,[db    ] INFO 2024/07/26 09:15:02 using sqlite database: /root/.evcc/evcc.db
2024/07/26 09:15:02,stdout,[main  ] INFO 2024/07/26 09:15:02 using config file: /etc/evcc.yaml
2024/07/26 09:15:02,stdout,[main  ] INFO 2024/07/26 09:15:02 evcc 0.128.4
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/main.go:44 +0x2d
2024/07/26 09:15:01,stdout,main.main()
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/cmd/root.go:107 +0x1a
2024/07/26 09:15:01,stdout,github.com/evcc-io/evcc/cmd.Execute()
2024/07/26 09:15:01,stdout,	github.com/spf13/cobra@v1.8.1/command.go:1041
2024/07/26 09:15:01,stdout,github.com/spf13/cobra.(*Command).Execute(...)
2024/07/26 09:15:01,stdout,	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff
2024/07/26 09:15:01,stdout,github.com/spf13/cobra.(*Command).ExecuteC(0x5db39a0)
2024/07/26 09:15:01,stdout,	github.com/spf13/cobra@v1.8.1/command.go:989 +0xab1
2024/07/26 09:15:01,stdout,"github.com/spf13/cobra.(*Command).execute(0x5db39a0, {0xc0000500a0, 0x0, 0x0})
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/cmd/root.go:182 +0xa8e
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/cmd.runRoot(0x5db39a0, {0x2e1c20d?, 0x7?, 0x2e0f1d1?})
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/cmd/setup.go:900 +0xb8
2024/07/26 09:15:01,stdout,github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x5db50c0)
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/cmd/setup.go:842 +0x36
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/cmd.configureDevices({{{0xc000764c70, 0x4}, {0xc000764c80, 0xa}, 0x1b9e}, {0xc000764ce3, 0x5}, {0xc000c32540, 0xd5}, {0xc000d0c940, ...}, ...})
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/cmd/setup.go:258 +0x217
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/cmd.configureMeters({0xc0003bcc30?, 0x5, 0x1ecf880?}, {0x0, 0x0, 0xc001506d98?})
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/meter/config.go:36 +0x62
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/meter.NewFromConfig({0xc000765788, 0x8}, 0xc0006299e0)
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/meter/template.go:18 +0x48
2024/07/26 09:15:01,stdout,github.com/evcc-io/evcc/meter.NewMeterFromTemplateConfig(0xc000765788?)
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/meter/config.go:36 +0x62
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/meter.NewFromConfig({0xc000bd6834, 0x4}, 0xc00135fad0)
"
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/meter/tapo.go:30 +0x91
2024/07/26 09:15:01,stdout,github.com/evcc-io/evcc/meter.NewTapoFromConfig(0xc00135fad0)
2024/07/26 09:15:01,stdout,	github.com/evcc-io/evcc/meter/tapo/connection.go:51 +0x274
2024/07/26 09:15:01,stdout,"github.com/evcc-io/evcc/meter/tapo.NewConnection({0xc0011254e8, 0x13}, {0xc000bd6880, 0x10}, {0xc000bd6890, 0x9})
"
2024/07/26 09:15:01,stdout,	github.com/insomniacslk/tapo@v1.0.1/plug.go:122 +0x182
2024/07/26 09:15:01,stdout,github.com/insomniacslk/tapo.(*Plug).GetDeviceInfo(0xc001331088)
2024/07/26 09:15:01,stdout,	github.com/insomniacslk/tapo@v1.0.1/klap_protocol.go:190 +0x34
2024/07/26 09:15:01,stdout,"github.com/insomniacslk/tapo.(*KlapSession).Request(0xc000d32000, {0xc0009791c0, 0x3c, 0x40})
"
2024/07/26 09:15:01,stdout,	github.com/insomniacslk/tapo@v1.0.1/klap_protocol.go:242 +0xa45
2024/07/26 09:15:01,stdout,"github.com/insomniacslk/tapo.(*KlapSession).request(0xc000d32000, {0xc0009791c0?, 0x18?, 0xc000580008?})
"
2024/07/26 09:15:01,stdout,	github.com/insomniacslk/tapo@v1.0.1/klap_protocol.go:128 +0x88
2024/07/26 09:15:01,stdout,"github.com/insomniacslk/tapo.(*KlapSession).decrypt(0xc000d32000, {0xc000c40400?, 0xc0016c94e0?, 0x1?})
"
2024/07/26 09:15:01,stdout,	github.com/insomniacslk/tapo@v1.0.1/klap_protocol.go:182 +0x112
2024/07/26 09:15:01,stdout,"github.com/insomniacslk/tapo.decryptCBC({0xc000763600?, 0x3997ee0?, 0x5d7a610?}, {0xc000763640, 0x10, 0x20}, {0xc000c40420, 0x11, 0x1e0})
"
2024/07/26 09:15:01,stdout,	crypto/cipher/cbc.go:145 +0x40b
2024/07/26 09:15:01,stdout,"crypto/cipher.(*cbcDecrypter).CryptBlocks(0x39b3c80?, {0xc000c40420?, 0xc000763640?, 0x10?}, {0xc000c40420?, 0x7f24f37e6108?, 0x200?})
"
2024/07/26 09:15:01,stdout,goroutine 1 [running]:
2024/07/26 09:15:01,stdout,
```

### What type of operating system are you running?

Docker container

### Version

latest

thierolm, 2024-07-26:
My P110 with the same firmware and hardware version is not having any issues:
```ksh
thierolm@LAPTOP-KQMMVN3H:~/git-repos/evcc$ ./evcc -c ./evcc_tapo_test.yaml -l debug charger
[main  ] INFO 2024/07/26 11:20:39 evcc 0.128.4 (5e608fb6)
[main  ] INFO 2024/07/26 11:20:39 using config file: ./evcc_tapo_test.yaml
[db    ] INFO 2024/07/26 11:20:39 using sqlite database: /home/thierolm/.evcc/evcc.db
[tapo  ] DEBUG 2024/07/26 11:20:39 SMART.TAPOPLUG P110 connected (fw:1.3.0 Build 230905 Rel.152200,hw:1.0,mac:54-AF-97-1D-5E-98)
Power:         0W
Charge status: B
Enabled:       false
Charged:       0.0kWh
```

thierolm, 2024-07-26:
Perhaps special characters in your Tapo password and/or user ....

skysoftgmbh, 2024-07-26:
Reboot of the P110 helped :-) it's also just a computer - sorry for ignoring this basic troubleshooting.

Sorry for keeping you busy.

closed 2024-07-26 by skysoftgmbh