evcc-io/evcc #15061: easee: switches down to 1 phase at start up and keeps even "now" is activated

MaTr74, 2024-07-26:
### Describe the bug

I reference to https://github.com/evcc-io/evcc/issues/14387#issuecomment-2243500663

Unlike the described case within the referenced issue in this case it started charging with mode `minpv`. But the expected behaviour would be to keep 3 phases activated at start up and at least switching to 3 phase on mode `now`.

The attached log shows the start of charging at mode `minpv` with 3 phases, scaling down to 6 A (that's correct) and 1 phase (for whatever reason) shortly afterwards, switching to mode `now` at 23:00, switching to 32 A but keeping 1 phase.

### Steps to reproduce

1. Plugin EV
2. Charging starts with 3 phases
3. Phases are scaled down to 1 everytime

### Configuration details

```yaml
site:
  - title: Hotel
    meters:
      grid: grid-meter
      pv:
      - sungrow1
      - sungrow2

loadpoints:
  - title: Wallbox-links
    charger: easee-master
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
      #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Wallbox-rechts
    charger: easee-slave
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
        #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Klimatruhe
    charger: klimatruhe
    mode: off
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 100
      delay: 3m
    guardduration: 5m
      #phases: 1
      #    mincurrent: 4 
      #    maxcurrent: 4 
    priority: 2 
    meter: sonoff-powr2-01

chargers:
  - name: easee-master
[... 345 lines cut ...]
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:48 charge currents: [24.1 0.015 0.009]A
evcc    | [lp-2  ] DEBUG 2024/07/25 23:00:48 charge power: 0W
evcc    | [lp-2  ] DEBUG 2024/07/25 23:00:48 charge currents: [0 0 0]A
evcc    | [site  ] DEBUG 2024/07/25 23:00:48 pv power: 0W
evcc    | [site  ] DEBUG 2024/07/25 23:00:48 grid meter: 7670W
evcc    | [site  ] DEBUG 2024/07/25 23:00:48 grid powers: [6014 412 1243]W
evcc    | [site  ] DEBUG 2024/07/25 23:00:48 grid currents: [25.9 2.12 5.32]A
evcc    | [site  ] DEBUG 2024/07/25 23:00:49 site power: 7670W
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 detected active phases: 1p
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 !! session: chargeRater.chargedEnergy=0.7 - chargedAtStartup=0.0
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 charge total import: 1892.646kWh
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 charger status: C
evcc    | [lp-1  ] WARN 2024/07/25 23:00:49 charger logic error: phases mismatch (got 1, expected 3)
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 switched phases: 3p
evcc    | [lp-1  ] DEBUG 2024/07/25 23:00:49 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
```

### What type of operating system are you running?

Docker container

### Version

v0.128.4

andig, 2024-07-27:
Duplicate of https://github.com/evcc-io/evcc/issues/15063?

closed 2024-07-27 by andig

MaTr74, 2024-07-28:
> Duplicate of #15063?

I don't know. What is the result? EVCC is everything doing right and the problem is lying within the easee api?

andig, 2024-07-28:
Seems phase switching for multiple Easees on a single circuit- your setup- is wrong.

MaTr74, 2024-07-28:
So, in this case it's not a duplicate of #15063, isn't it?

MaTr74, 2024-07-28:
> Seems phase switching for multiple Easees on a single circuit- your setup- is wrong.

How did you come to this conclusion? Can you explain that a little more?

andig, 2024-07-28:
Could you explain your Easee setup? We should probably wait until #15063 is then resolved and test again.

MaTr74, 2024-07-28:
> Could you explain your Easee setup?

What do you mean exactly? 

> We should probably wait until #15063 is then resolved and test again.

Can you please reopen this issue as long as it's not clear what the problem and solution is?

andig, 2024-07-28:
How are your Easees connected? Multiple Circuits? Screenshots from Setup in Easee app or website?

MaTr74, 2024-07-28:
> How are your Easees connected? Multiple Circuits? Screenshots from Setup in Easee app or website?

[image]

andig, 2024-07-28:
Genau. 2 Easee an einem Circuit. Duplikat.

MaTr74, 2024-07-28:
> Genau. 2 Easee an einem Circuit. Duplikat.

I don't understand. The referenced issue is about "Easee-Wallbox (**Single**): Switch phases to 3p fails"

andig, 2024-07-28:
Ahh sorry. I meant https://github.com/evcc-io/evcc/issues/14996!

GrimmiMeloni, 2024-08-08:
I think this is very likely a duplicate of #14387.